GloDroid/glodroid_manifest #89: adb not working

NO-ob, 2020-09-23:
Hi I think adb broken on the 0.5.0 build, I can see the device with adb devices but app sideloading doesn't work 
```
List of devices attached
92c0b5ba2b9d3d87	device
```

```
adb: sideload connection failed: closed
adb: trying pre-KitKat sideload method...
adb: pre-KitKat sideload connection failed: closed

```

I noticed that the popup for authorization doesn't show on the phone when connecting so maybe thats the issue?

rsglobal, 2020-09-24:
@NO-ob ,

Hi, adb works for me in most of the cases.
If only sideload command is broken, please change the description.
Also please describe steps to reproduce in more details.

NO-ob, 2020-09-24:
> @NO-ob ,
> 
> Hi, adb works for me in most of the cases.
> If only sideload command is broken, please change the description.
> Also please describe steps to reproduce in more details.

Hi thanks for responding. Turns out it is just adb sideload, I have updated the issue

rsglobal, 2020-09-24:
Are you sure sideload is a right command?
Is seems you have to use `adb install`:
https://stackoverflow.com/questions/7076240/install-an-apk-file-from-command-prompt

NO-ob, 2020-09-24:
> Are you sure sideload is a right command?
> Is seems you have to use `adb install`:
> https://stackoverflow.com/questions/7076240/install-an-apk-file-from-command-prompt

Oh right i'm an idiot lmao, sorry for wasting your time

closed 2020-09-24 by NO-ob

edbock, 2021-03-27:
Don't feel bad, there are a lot of us idiots out there.  I made the same mistake.  Thank you for this issue.  It helped me.

AntiGamerPlayzz, 2023-02-02:
> > Are you sure sideload is a right command?
> > Is seems you have to use `adb install`:
> > https://stackoverflow.com/questions/7076240/install-an-apk-file-from-command-prompt
> 
> Oh right i'm an idiot lmao, sorry for wasting your time

lmao is tiktok tiktok is cringe

felipegbot, 2023-12-07:
> Don't feel bad, there are a lot of us idiots out there. I made the same mistake. Thank you for this issue. It helped me.

same here, this issue helped me

HeyBroooo, 2025-03-22:
there are 2 adb option both have different function and operation that they perform use it wisely if want to flash some thing then use adb sideload <filename> else the other one