GloDroid/glodroid_manifest #93: Orange PI Plus 2 - deployment issue

BigJacek, 2020-09-25:
Hello,
I'm trying to deploy image v0.5.1 and I have some problem.
It doesn't metfer if I'm trying flash Android on sdcard or on eMMC results are the same:
according to "Deploying GloDroid"
Step 1 - ok
Step 2 - ok
Step 3 - see log:
===========================================
adminek@ESPRIMO-Q5030:~/images$ ./flash-sd.sh 
+ ./fastboot flash gpt deploy-gpt.img
Sending 'gpt' (128 KB)                             OKAY [  0.014s]
Writing 'gpt'                                      OKAY [  0.229s]
Finished. Total time: 0.259s
+ ./fastboot flash bootloader bootloader-sd.img
Sending 'bootloader' (1920 KB)                     OKAY [  0.137s]
Writing 'bootloader'                               OKAY [  0.141s]
Finished. Total time: 0.285s
+ ./fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.024s]
Writing 'uboot-env'                                OKAY [  0.027s]
Finished. Total time: 0.058s
+ ./fastboot reboot
Rebooting                                          OKAY [  0.000s]
Finished. Total time: 0.601s
+ set +x
+ ./fastboot oem format
< waiting for any device >
                                                   OKAY [  0.199s]
Finished. Total time: 0.199s
+ ./fastboot flash bootloader bootloader-sd.img
Sending 'bootloader' (1920 KB)                     OKAY [  0.138s]
Writing 'bootloader'                               OKAY [  0.140s]
Finished. Total time: 0.287s
+ ./fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.023s]
Writing 'uboot-env'                                OKAY [  0.026s]
Finished. Total time: 0.059s
+ ./fastboot flash recovery_boot boot.img
Sending 'recovery_boot' (24576 KB)                 OKAY [  1.683s]
Writing 'recovery_boot'                            OKAY [  1.704s]
Finished. Total time: 3.397s
+ ./fastboot erase misc
Erasing 'misc'                                     OKAY [  0.015s]
Finished. Total time: 0.032s
+ ./fastboot reboot-fastboot
Rebooting into fastboot                            OKAY [  0.007s]
< waiting for any device >
=============================================
and nothing happen next.

On device site I can see fastboot mode (pls. see picture).
[image]
Any suggestions?

Regards,
Jacek

rsglobal, 2020-09-25:
@BigJacek ,

Thank you for reporting this. It' my fault, I forgot to pick OTG commit when migrate from v5.6 to v5.8 kernel.
I'll fix that and update you with preview image soon.

rsglobal, 2020-09-25:
@Quang-vip,

Hello,
Could you check opi-prime on v0.5.1?

BigJacek, 2020-09-25:
ok. Thanks Roman.
Regards,
Jacek

Quang-vip, 2020-09-25:
> 
> 
> @Quang-vip,
> 
> Hello,
> Could you check opi-prime on v0.5.1?

I'll have a look tomorrow ;)

rsglobal, 2020-09-25:
@BigJacek ,

https://drive.google.com/file/d/1JLP6jlo4JxnkmAhTgwDyLQx3-wmdXLZ-/view?usp=sharing

BigJacek, 2020-09-25:
rsglobal

> @BigJacek ,
> 
> https://drive.google.com/file/d/1JLP6jlo4JxnkmAhTgwDyLQx3-wmdXLZ-/view?usp=sharing

ok. I'm getting to work.

BigJacek, 2020-09-25:
Everything works.
The system is easily installed on eMMC.
It seems to me to run even smoother than Android 10.
Now I will test this device in my free time.
Thanks Roman.

Quang-vip, 2020-09-26:
@rsglobal
it seems I'm having some problems with deploying onto the Pi Prime 
It froze after the `./fastboot reboot-fastboot` command.
And on serial console, it stopped at: `[    2.436539] sun50i-h5-pinctrl 1c20800.pinctrl: supply vcc-pc not found, usin`

k02pradeep, 2020-09-30:
> @BigJacek ,
> 
> https://drive.google.com/file/d/1JLP6jlo4JxnkmAhTgwDyLQx3-wmdXLZ-/view?usp=sharing

Hi What are the steps to install this. I tried flash the sd image to sd card and when inserted into opi plus 2, and booted nothing happens. Onboard led doesnot turn up and monitor goes blank. Command ./flash_sd.sh waits for android devices.

Please point me to the right steps to install for OPI plus 2

BigJacek, 2020-09-30:
@k02pradeep
Is there eMMC memory available?
 (if yes) pls. try to install on eMMC.
I'm trying to install image on SD card.

Regards,
Jacek

BigJacek, 2020-09-30:
@rsglobal

I can confirm some problem to deploy image on SD card.
=======================
adminek@ESPRIMO-Q5030:~/images$ export card=/dev/sdb
adminek@ESPRIMO-Q5030:~/images$ sudo dd if=deploy-sd.img of=${card} bs=4096
[sudo] password for adminek: 
65536+0 records in
65536+0 records out
268435456 bytes (268 MB, 256 MiB) copied, 18,5812 s, 14,4 MB/s
adminek@ESPRIMO-Q5030:~/images$ sync
adminek@ESPRIMO-Q5030:~/images$ sync
adminek@ESPRIMO-Q5030:~/images$ sync
adminek@ESPRIMO-Q5030:~/images$ ./flash-sd.sh 
+ ./fastboot flash gpt deploy-gpt.img
< waiting for any device >
Sending 'gpt' (128 KB)                             OKAY [  0.020s]
Writing 'gpt'                                      OKAY [  0.032s]
Finished. Total time: 0.062s
+ ./fastboot flash bootloader bootloader-sd.img
Sending 'bootloader' (1920 KB)                     OKAY [  0.142s]
Writing 'bootloader'                               OKAY [  0.151s]
Finished. Total time: 0.300s
+ ./fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.023s]
Writing 'uboot-env'                                OKAY [  0.027s]
Finished. Total time: 0.059s
+ ./fastboot reboot
Rebooting                                          OKAY [  0.000s]
Finished. Total time: 0.602s
+ set +x
+ ./fastboot oem format
< waiting for any device >
                                                   OKAY [  0.374s]
Finished. Total time: 0.374s
+ ./fastboot flash bootloader bootloader-sd.img
Sending 'bootloader' (1920 KB)                     OKAY [  0.140s]
Writing 'bootloader'                               OKAY [  0.137s]
Finished. Total time: 0.288s
+ ./fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.023s]
Writing 'uboot-env'                                OKAY [  0.162s]
Finished. Total time: 0.195s
+ ./fastboot flash recovery_boot boot.img
Sending 'recovery_boot' (24576 KB)                 OKAY [  1.734s]
Writing 'recovery_boot'                            OKAY [  1.788s]
Finished. Total time: 3.533s
+ ./fastboot erase misc
Erasing 'misc'                                     OKAY [  0.014s]
Finished. Total time: 0.031s
+ ./fastboot reboot-fastboot
Rebooting into fastboot                            OKAY [  0.206s]
< waiting for any device >
Finished. Total time: 12.874s
+ ./fastboot flash boot_a boot.img
Sending 'boot_a' (32768 KB)                        OKAY [  2.426s]
Writing 'boot_a'                                   OKAY [  2.472s]
Finished. Total time: 5.062s
+ ./fastboot flash dtbo_a boot_dtbo.img
Sending 'dtbo_a' (32 KB)                           OKAY [  0.028s]
Writing 'dtbo_a'                                   OKAY [  0.043s]
Finished. Total time: 0.112s
+ ./fastboot flash super super.img
Invalid sparse file format at header magic
Sending sparse 'super' 1/4 (262136 KB)             OKAY [ 19.333s]
Writing 'super'                                    OKAY [ 19.110s]
Sending sparse 'super' 2/4 (242944 KB)             OKAY [ 18.001s]
Writing 'super'                                    OKAY [ 17.988s]
Sending sparse 'super' 3/4 (253360 KB)             OKAY [ 18.729s]
Writing 'super'                                    OKAY [ 18.992s]
Sending sparse 'super' 4/4 (83596 KB)              OKAY [  6.173s]
Writing 'super'                                    OKAY [ 49.786s]
Finished. Total time: 169.155s
+ ./fastboot format:ext4 metadata
Warning: metadata type is raw, but ext4 was requested for formatting.
mke2fs 1.45.4 (23-Sep-2019)
Creating filesystem with 4096 4k blocks and 4096 inodes

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (1024 blocks): done
Writing superblocks and filesystem accounting information: done

Sending 'metadata' (52 KB)                         OKAY [  0.010s]
Writing 'metadata'                                 OKAY [  0.104s]
Finished. Total time: 0.193s
+ ./fastboot format:ext4 userdata
Warning: userdata type is raw, but ext4 was requested for formatting.
mke2fs 1.45.4 (23-Sep-2019)
Creating filesystem with 1526395 4k blocks and 382016 inodes
Filesystem UUID: 1082e1ba-4220-4b24-b56c-432978e7a3af
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (16384 blocks): done
Writing superblocks and filesystem accounting information: done 

Sending 'userdata' (3100 KB)                       OKAY [  0.228s]
Writing 'userdata'                                 OKAY [  0.886s]
Finished. Total time: 1.237s
+ ./fastboot reboot
Rebooting                                          OKAY [  0.000s]
Finished. Total time: 0.051s
adminek@ESPRIMO-Q5030:~/images$
=============================
After the last rebooting (according to the script) nothing happen.
Onboard green led blinks every 10s and that's all.
After the power off/on led no longer blinks and monitor goes blank.

Regards,
Jacek

k02pradeep, 2020-09-30:
> @k02pradeep
> Is there eMMC memory available?
> (if yes) pls. try to install on eMMC.
> I'm trying to install image on SD card.
> 
> Regards,
> Jacek

Default android os is there on emmc. I want to try the glodroid in sd card.

k02pradeep, 2020-09-30:
Tried installing on emmc. Installation was successful. But while booting I see "there's an internal problem with your device. contact your manufacturer for details" and there after os bootup

Also I do not see settings and other options to configure network and other stuff. 
Pressing power button doesnot bring up neither power off dialog not powers it off.

BigJacek, 2020-10-01:
Yesterday, when I tested the installation on the SD card, I also installed the system on eMMC memory 2 times and there were no problems. 
As for the operation of the power button, I honestly did not test it.

BigJacek, 2020-10-01:
Hello,
The message at the start of Android system "There's an internal problem with your device. contact your manufacturer for details" what I know is an old known bug and nobody knows how to fix it
From what I've checked, apart from the message itself, nothing bad is happening.
I've checked power button and I can confirm it doesn't work.
To access the panel with shortcuts to applications and settings just slide the bottom bar up.
In Settings panel select Network & internet and enable Wi-Fi switch (disabled by default).

Regards,
Jacek

Quang-vip, 2020-11-16:
@rsglobal
I just retried to install Android on my Pi Prime with the latest image (`24-09-2020-17h09m_release_v0.5.1_opi_prime-userdebug.tar.gz` on October 1st) and the result is still the same.

Still nothing happened after this command (even on the screen):
`./fastboot reboot-fastboot
Rebooting into fastboot
< waiting for any device >`

I wonder what's happening 🤔

rsglobal, 2023-01-21:
GloDroid team doesn't have this board to confirm. Closing due to inactivity.

closed 2023-01-21 by rsglobal