GloDroid/glodroid_manifest #95: pinephone: Use OPTEE for KeyMaster and GateKeeper

rsglobal, 2020-09-28:
1. Setup OPTEE os:
https://mquaresma.github.io/pine/a64/2020/04/11/Setting-up-OPTEE-and-Linux-for-the-Pine-A64.html

2. Add Linaro HALs:
https://github.com/linaro-swg/kmgk

milkey-mouse, 2020-10-19:
Why not [Trusty](https://source.android.com/security/trusty)?

rsglobal, 2020-10-19:
OPTEE officially supports A64 SOC . Will trusty work on it too?

milkey-mouse, 2020-10-19:
Good point. Looks like it's not officially supported on A64...

rsglobal, 2020-10-19:
I am not an expert, Trusty has generic arm64 device, which gives me a hope that it can be possible on any SOC without using SOC hardware acceleration. But at least true random number generator will be missing, which is bad for security stuff.

AndreySV, 2026-04-01:
New link to built instructions:
https://mquaresma.dev/pine/a64/2020/04/11/Setting-up-OPTEE-and-Linux-for-the-Pine-A64