GloDroid/glodroid_manifest #96: opengl cannot be used, game cannot be run

zcadqewxs, 2020-10-02:
I am using Raspberry Pi 4b
When I run a 3d game, the game has a black screen and cannot run, and the logcat log shows these error messages

`
10-02 06:12:48.021  1360  1360 E libEGL  : validate_display:89 error 3001 (EGL_NOT_INITIALIZED)

10-02 06:12:54.304  1625  1849 E FrameEvents: updateAcquireFence: Did not find frame.
10-02 06:12:54.321  1625  1849 E FrameEvents: updateAcquireFence: Did not find frame.
01-01 00:01:51.884     0     0 E v3d fec00000.v3d: MMU error from client L2T (0) at 0x10f81000, pte invalid
10-02 06:12:56.237  1625  1625 E libprocessgroup: set_timerslack_ns write failed: Operation not permitted
10-02 06:12:57.003  1625  1625 E libprocessgroup: set_timerslack_ns write failed: Operation not permitted
01-01 00:02:14.404     0     0 E audit   : rate limit exceeded
01-01 00:02:17.682     0     0 E         : [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
01-01 00:02:17.849     0     0 E         : [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
01-01 00:02:17.900     0     0 E         : [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
01-01 00:02:17.933     0     0 E         : [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
01-01 00:02:17.960     0     0 E         : [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:

01-01 00:03:31.401     0     0 E         : [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
10-02 06:14:35.604   268   268 E GRALLOC-GBM: failed to create BO, size=1920x1080, fmt=1, usage=5
10-02 06:14:35.605  1175  1209 E GraphicBufferAllocator: Failed to allocate (1920 x 1080) layerCount 1 format 1 usage b00: 5
10-02 06:14:35.605  1175  1209 E BufferQueueProducer: [com.android.launcher3/com.android.launcher3.uioverrides.QuickstepLauncher#0](id:4970000000c,api:1,p:1625,c:1175) dequeueBuffer: createGraphicBuffer failed
01-01 00:03:31.443     0     0 E         : [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
10-02 06:14:35.646   268   268 E GRALLOC-GBM: failed to create BO, size=1920x1080, fmt=1, usage=5
10-02 06:14:35.647  1175  1209 E GraphicBufferAllocator: Failed to allocate (1920 x 1080) layerCount 1 format 1 usage b00: 5
10-02 06:14:35.647  1175  1209 E BufferQueueProducer: [com.android.launcher3/com.android.launcher3.uioverrides.QuickstepLauncher#0](id:4970000000c,api:1,p:1625,c:1175) dequeueBuffer: createGraphicBuffer failed
`

rsglobal, 2020-10-02:
@zcadqewxs ,

Hi,

Please attach full logcat and dmesg logs:
Please describe steps to reproduce the issue.

zcadqewxs, 2020-10-02:
> @zcadqewxs ,
> 
> Hi,
> 
> Please attach full logcat and dmesg logs:
> Please describe steps to reproduce the issue.

@rsglobal

This is the adb log recorded from boot:
[adb.log](https://github.com/GloDroid/glodroid_manifest/files/5316689/adb.log)

dmesg logs:
[dmesg.log](https://github.com/GloDroid/glodroid_manifest/files/5316694/dmesg.log)

game package name: com.dena.a12022874

When I open this game, the screen keeps being black. This is also the case when I open other games

rsglobal, 2020-11-01:
@zcadqewxs ,

Please provide a link to the .apks of those games.

rsglobal, 2021-04-27:
Closing due to inactivity.

closed 2021-04-27 by rsglobal