GloDroid/glodroid_manifest #97: Offer user builds, not only userdebug

FantasyCookie17, 2020-10-03:
This would only need changing a single line or so, I suppose. Rationale is the following: userdebug [enables `ro.debuggable`](https://source.android.com/setup/develop/new-device#build-variants), which opens debugging interfaces and thus [can](https://stackoverflow.com/questions/37143960/androidstudio-what-does-debuggable-do) [decrease](https://www.pentestpartners.com/security-blog/android-debug-mode-and-apps-a-cautionary-tale/) [security](https://madaidans-insecurities.github.io/android.html#lineageos).

rsglobal, 2021-05-24:
@matviizorin 

Assignment removed due to long inactivity. Please update it if you are still working on the issue.

herbsmn, 2022-11-04:
On my PinePhone running GloDroid v0.8.1 I get the message, "Serial console enabled - Performance is impacted".

I'm curious if a user build instead of a userdebug build would turn the serial console off.

rsglobal, 2022-11-04:
> On my PinePhone running GloDroid v0.8.1 I get the message, "Serial console enabled - Performance is impacted".
> 
> I'm curious if a user build instead of a userdebug build would turn the serial console off.

No, I believe it could be disabled in kernel configs.