GloDroid/glodroid_manifest #98: Orange Pi PC2 won't boot, bootloop

DartPower, 2020-10-06:
Orange Pi PC2 won't boot, bootloop

[dmesg_031020_2.log](https://github.com/GloDroid/glodroid_manifest/files/5333243/dmesg_031020_2.log)
[logcat_opipc2_031020.log](https://github.com/GloDroid/glodroid_manifest/files/5333244/logcat_opipc2_031020.log)
[logcat_opipc2_031020_afterreboot2.log](https://github.com/GloDroid/glodroid_manifest/files/5333245/logcat_opipc2_031020_afterreboot2.log)
[dmesg_031020_1.log](https://github.com/GloDroid/glodroid_manifest/files/5333246/dmesg_031020_1.log)

d4rth-v4d3r, 2020-11-06:
Were you able to run Android in Orange PI PC 2?

DartPower, 2020-11-06:
Me don't have a much free disk space, because im cant compile it fully and deploy.
Can you make the release/CI/buildbot?

DartPower, 2020-11-06:
Soryy, don't seen the Releases. Will check soon

DartPower, 2020-11-06:
This build (in rel) im already tested (on 2nd October), and this is not working.

DartPower, 2020-11-07:
Booted. But UI is very laggy...

rsglobal, 2022-11-20:
closing as stale

closed 2022-11-20 by rsglobal