GloDroid/glodroid_manifest #99: [request] support old A10/13/20

johndoe71rus, 2020-10-06:
Any chance support old A10/13/20 SoC? Almost all hardware support in kernel. 
Build Android Go version?

rsglobal, 2020-10-06:
There is no such plans. 

Maintaining big device matrix is expensive for us (in terms of testing time, support, etc.)
Instead we would like to focus on devices which have some reasonable benefits (price / features).
For low-price devices we have 16$ opi-pc support.
For avg. price / general purpose usage we have rpi4, opi3.
We are planning to add Khadas VIM3 here to cover high performance boards.

Are there any rational reason to support old A10/13/20 SoC?

Is Android-GO available in FOSS domain? I've googled for a 5 minutes and haven't found anything but a brochure.

johndoe71rus, 2020-10-06:
A13 renamed to "actual" R8. A20 renamed to T2. 
A13 is the A10 without hdmi. Theoreticaly A10 is R8 with hdmi.

I just heard that Android go appeared in version 8. For devices with poor hardware. I do not know if there is a Go version for version 10 and 11. I think the code base is the same, the differences can only be in the build configuration.

closed 2021-05-25 by rsglobal