GloDroid/glodroid_manifest #101: [Raspberry Pi4] flash-sd.sh fail

dchjung, 2020-10-29:
I need help.
I built with rpi4-user or userdebug on Ubuntu 18.
I flashed the deploy-sd.img file on the SD card through etcher for Ubuntu.
And after inserting the SD card into the device and connecting it to Ubuntu, the ./flash-sd.sh script is executed, but the following error occurs.

rpi4$ ./flash-sd.sh 
+ ./fastboot flash gpt deploy-gpt.img
< waiting for any device >
Sending 'gpt' (128 KB)                             OKAY [  0.010s]
Writing 'gpt'                                      OKAY [  0.036s]
Finished. Total time: 0.084s
+ ./fastboot flash bootloader bootloader-sd.img
Invalid sparse file format at header magic
Sending sparse 'bootloader' 1/1 (7036 KB)          OKAY [  0.361s]
Writing 'bootloader'                               OKAY [ 11.124s]
Finished. Total time: 11.570s
+ ./fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.016s]
Writing 'uboot-env'                                OKAY [  0.037s]
Finished. Total time: 0.088s
+ ./fastboot reboot
Rebooting                                          OKAY [  0.000s]
Finished. Total time: 0.101s
+ set +x
+ ./fastboot oem format
< waiting for any device >
                                                   OKAY [  0.033s]
Finished. Total time: 0.034s
+ ./fastboot flash bootloader bootloader-sd.img
Invalid sparse file format at header magic
Sending sparse 'bootloader' 1/1 (7036 KB)          OKAY [  0.364s]
Writing 'bootloader'                               OKAY [  8.568s]
Finished. Total time: 9.028s
+ ./fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.017s]
Writing 'uboot-env'                                OKAY [  0.031s]
Finished. Total time: 0.112s
+ ./fastboot flash recovery_boot boot.img
Invalid sparse file format at header magic
Sending sparse 'recovery_boot' 1/2 (16380 KB)      OKAY [  0.829s]
Writing 'recovery_boot'                            OKAY [  0.778s]
Sending sparse 'recovery_boot' 2/2 (5388 KB)       OKAY [  0.266s]
Writing 'recovery_boot'                            OKAY [  0.443s]
Finished. Total time: 2.402s
+ ./fastboot erase misc
Erasing 'misc'                                     OKAY [  0.032s]
Finished. Total time: 0.132s
+ ./fastboot reboot-fastboot
Rebooting into fastboot                            OKAY [  0.037s]
< waiting for any device >
Finished. Total time: 17.536s
+ ./fastboot flash boot_a boot.img
Sending 'boot_a' (32768 KB)                        OKAY [  1.247s]
Writing 'boot_a'                                   OKAY [  1.697s]
Finished. Total time: 3.012s
+ ./fastboot flash dtbo_a boot_dtbo.img
Sending 'dtbo_a' (46 KB)                           OKAY [  0.023s]
Writing 'dtbo_a'                                   FAILED (remote: 'No such file or directory')
fastboot: error: Command failed

When this happens, a menu screen for selecting the boot mode appears on the device screen.

Can you see why this is happening?

rsglobal, 2020-10-29:
Hi,
Please run `make images`  and upload the archive here.

dchjung, 2020-10-30:
This image was created in rpi4-use mode.

[https://drive.google.com/file/d/17txg26vGkgeT5jxWE2Tsime15jLlRu_p/view?usp=sharing](https://drive.google.com/file/d/17txg26vGkgeT5jxWE2Tsime15jLlRu_p/view?usp=sharing)

rsglobal, 2020-11-01:
@dchjung ,

I have the same error with your images. 
It seems you've turned off permissive mode.
From UART console:
```
[    9.704540] audit: type=1400 audit(9.699:8): avc:  denied  { write } for  pid=126 comm="fastbootd" name="mmcblk0p6" dev="tmpfs" ino=12828 scontext=u:r:fastbootd:s0 tcontext=u:object_r:block_device:s0 tc
lass=blk_file permissive=0
[    9.724891] audit: type=1400 audit(9.707:9): avc:  denied  { write } for  pid=126 comm="fastbootd" name="mmcblk0p6" dev="tmpfs" ino=12828 scontext=u:r:fastbootd:s0 tcontext=u:object_r:block_device:s0 tc
lass=blk_file permissive=0
```

GloDroid is not ready now to run in enforced selinux mode.

dchjung, 2020-11-02:
@rsglobal ,
Thanks for checking.
I don't understand your answer very well.
Anyway my purpose is to remove the /system/xbin/su file.
So i tried adb remount but android won't boot after that.
So is there any way to remove the su file when building android or after build?
I want to create an unrooted build image.

rsglobal, 2020-11-02:
@dchjung ,

By default selinux is enforced for user builds.
To do the things in a right manner, you need to create/fix all necessary selinux rules, which is not trivial task.

Instead you can try to set this option to 1 (modify AOSP's Android.bp) , build and test again:
https://cs.android.com/android/platform/superproject/+/master:system/core/init/Android.bp;l=89;drc=aca0beaf77247675400d9771c6269bd899fa379f

dchjung, 2020-11-04:
@rsglobal ,
I solved the problem with your help .
Thank you sincerely.
Personally, Glodroid looks better than LineageOS.
Display cracking happens occasionally, but it doesn't matter to me.

rsglobal, 2020-11-05:
Thanks. Hope someday we could combine out efforts with @KonstaT

Can this issue be closed?

closed 2020-11-09 by dchjung