GloDroid/glodroid_manifest #102: Don`t download repository

cyber-bat, 2020-11-01:
Don`t download repository. Write:

`repo init -u https://github.com/glodroid/glodroid_manifest
Downloading Repo source from https://gerrit.googlesource.com/git-repo
Downloading manifest from https://github.com/glodroid/glodroid_manifest
remote: Enumerating objects: 25, done.        
remote: Counting objects: 100% (25/25), done.        
remote: Compressing objects: 100% (18/18), done.        
remote: Total 245 (delta 11), reused 18 (delta 7), pack-reused 220        
Receiving objects: 100% (245/245), 65.26 KiB | 777.00 KiB/s, done.
Resolving deltas: 100% (146/146), done.

Traceback (most recent call last):
  File "/home/outer/GloDroid/.repo/repo/main.py", line 628, in <module>
    _Main(sys.argv[1:])
  File "/home/outer/GloDroid/.repo/repo/main.py", line 602, in _Main
    result = run()
  File "/home/outer/GloDroid/.repo/repo/main.py", line 595, in <lambda>
    run = lambda: repo._Run(name, gopts, argv) or 0
  File "/home/outer/GloDroid/.repo/repo/main.py", line 264, in _Run
    result = cmd.Execute(copts, cargs)
  File "/home/outer/GloDroid/.repo/repo/subcmds/init.py", line 532, in Execute
    self._ConfigureUser(opt)
  File "/home/outer/GloDroid/.repo/repo/subcmds/init.py", line 388, in _ConfigureUser
    name = self._Prompt('Your Name', mp.UserName)
  File "/home/outer/GloDroid/.repo/repo/project.py", line 1026, in UserName
    self._LoadUserIdentity()
  File "/home/outer/GloDroid/.repo/repo/project.py", line 1039, in _LoadUserIdentity
    u = self.bare_git.var('GIT_COMMITTER_IDENT')
  File "/home/outer/GloDroid/.repo/repo/project.py", line 3411, in runner
    (self._project.name, name, p.stderr))
error.GitError: manifests var: 
*** Please tell me who you are.

Run

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

to set your account's default identity.
Omit --global to set the identity only in this repository.

fatal: unable to auto-detect email address (got 'outer@ubuntu18.(none)')
`
I don`t understand why?

rsglobal, 2020-11-01:
Hi

Have you tried to configure name and e-mail?
```
git config --global user.email "you@example.com"
git config --global user.name "Your Name"
```

cyber-bat, 2020-11-01:
Yes

cyber-bat, 2020-11-01:
reinstalled repo from the https://source.android.com/setup/develop#installing-repo, still not successful

cyber-bat, 2020-11-02:
Enter commands
gpg --recv-key 8BB9AD793E8E6153AF0F9A4416530D5E920F5C65
curl https://storage.googleapis.com/git-repo-downloads/repo.asc | gpg --verify - ~/bin/repo
All worked.

closed 2020-11-02 by cyber-bat