GloDroid/glodroid_manifest #103: Add support for Amlogic based SBCs

superna9999, 2020-11-04:
This is the issue tracking for adding support for the Amlogic based SBCs like:

- AML-S905X-CC (Le Potato)
- AML-S805X-CC (La Frite)
- Khadas VIM1
- Khadas VIM2
- Khadas VIM3
- Khadas VIM3L
- Odroid N2
- Odroid N2+
- Odroid C4

Initial support is done into :
- https://github.com/superna9999/glodroid_manifest
- https://github.com/superna9999/glodroid_device/tree/amlogic
- https://github.com/superna9999/glodroid_forks/tree/uboot-next
- https://github.com/superna9999/glodroid_forks/tree/kernel-sunxi-next
- https://github.com/superna9999/glodroid_forks/tree/mesa3d-next

Actually implemented:
- AML-S905X-CC (Le Potato) but Not Tested
- Khadas VIM2 but Not Tested
- Khadas VIM3 but Not Tested
- Khadas VIM3L 
- Odroid N2 but Not Tested

Missing:
- AML-S805X-CC (La Frite): Needs to chainload from embedded u-boot
- Khadas VIM1: Nothing specific
- Odroid N2+: Not supported yes in mainline U-boot
- Odroid C4: U-boot too old

Issues:
- [ ] Device goes in OOM after opening an app
- [ ] Bootrapping needs to be finalized & documented
- [ ] The Amlogic bootloader doesn't support beeing on a GPT disk, so we need to flash the bootloader on the eMMC boot0/boot0 or the eventual SPI flash, u-boot's fastboot doesn't support specifying custom bootloader partitions
- [ ] `adb reboot-bootloader` writes correctly the BCB but doesn't actually reboots the board...
- [ ] Switch to a more recent Mesa3D
- [ ] Switch to a more recent U-Boot to support more boards
- [ ] Testing
- [ ] More Testing

rsglobal, 2020-11-04:
@superna9999 ,

Thanks a lot for your great work.

Could you send MR into mesa3d with your fix?
https://github.com/superna9999/glodroid_forks/commit/583512f069e05a192e603ac1d6952073038474d4

rsglobal, 2020-11-06:
I updated mesa3d to latest mainline:
https://github.com/GloDroid/glodroid_forks/commits/mesa3d-next

rsglobal, 2020-11-06:
u-boot rebased onto trini/master: https://github.com/GloDroid/glodroid_forks/commits/uboot-next

rsglobal, 2021-01-23:
We will probably go with patching u-boot to shift main GPT behind boot-loader.
Kernel can be started with `gpt` argument, which should force using backup GPT even if main is broken.

This will allow to run GloDroid using same deploy procedure for both SD + EMMC and give us some time to look for another solution.

@superna9999 ,

We was talking that sniffing SDCARD signals may expose new inputs about amlogic boot procedure. Do you have any updates on this?

rsglobal, 2021-10-19:
Ok, so let's assume all those boards has emmc, where SPL can be located on the second hardware partition.
My only concerns which prevents this from merging is what to do in case SPL is broken and board is bricked.
You've told me that user can short some emmc pin to continue loading from sdcard. But all this sound too scary for many users (for me as well). To go with this approach I would like to have well-described procedure with the photos for every board.
Unfortunately I didn't have a chance to google for such kind of manuals. So if you can share the links, it will speed-up merging process significantly.

rsglobal, 2021-10-19:
Rockchip hardware has similar boot flow (emmc->sd).

But it has smart SPL, which after starting from EMMC, will probe sd-card, and in case bootable media found there, it starts SPL (or some of its part) from sdcard.

It really would be nice to have single immutable simple SPL which will be located on EMMC hard and just load other SPLs from sdcard/emmc from sectors 128+. In this case "bricking" the board would be much less probably.

rsglobal, 2021-10-20:
Let;s do this. But it would be nice to apply this change only for boot media and keep other medias with standard GPT offsets.

superna9999, 2021-10-20:
Yep sure, I added some tweaks, linux needs no change except the "gpt" kernel param, U-Boot would need some change to only write the backup GPT.

rsglobal, 2021-11-23:
Flashing manuals:  
https://gitlab.com/baylibre/amlogic/atv/aosp/device/amlogic/yukawa/-/wikis/Khadas_VIM3#flashing-yukawa-images

rsglobal, 2021-11-23:
@superna9999 ,

Hello Neil,

Could you share your work for disabling main GPT in u-boot and kernel?
I would like to try VIM3 in next our release.

superna9999, 2021-11-23:
@rsglobal here you go:
- U-boot: https://github.com/superna9999/u-boot/commit/2dc9e4b6fa134403571fb7c478b578ee2ae90d95

- Linux
https://github.com/superna9999/linux/commit/de90dda65748b7d58ab5ac2683fb1095f9620484
This one would require passing `alternate-gpt-okay` on the kernel cmdline

leksmax, 2022-08-23:
hello i  cant  build  firmware
have  this  mesage 
[100% 24858/24858] writing packaging rules ...
Starting ninja...
FAILED: ninja: 'kernel/glodroid-sunxi/tools/testing/selftests/powerpc/copyloops/memcpy_mcsafe_64.S', needed by 'out/target/product/lepotato/obj/KERNEL_OBJ/arch/arm64/boot/Image', missing and no known rule to make it
10:44:30 ninja failed with: exit status 1

rsglobal, 2023-05-01:
Initial Amlogic support added:
https://github.com/GloDroidCommunity/amlogic-series