GloDroid/glodroid_manifest #104: pinephone: Implement GNSS HAL

rsglobal, 2020-11-11:
Task 0:
Using this code enable GNSS in the `modem.rc` file.

```
adb shell
$ su
# echo 1 > /sys/class/modem-power/modem-power/device/powered
# cat /dev/ttyUSB1 &
# echo AT+QGPS=1,30,50,0,2 > /dev/ttyUSB2

You should see GPS data on the screen.

To disable GPS:
# echo AT+QGPSEND > /dev/ttyUSB2
```

Task 1. Try to use GNSS@1.0 with existing library:
https://github.com/GloDroid/glodroid_device/blob/1fe99ca9dd51119b07ce1e162b040b673f7c22fe/common/fstab#L6

Task 2: Use AOSP GNSS@2 mock implementation and integrate library's logic from Task 1 into it.

picoworm, 2023-01-21:
Now as we have working mm-radio HAL (https://github.com/GloDroid/mm-radio), same approach can be used to bring-up modem-manager based GNSS HAL, (mm-gnss).

rsglobal, 2023-01-21:
Now as we have working mm-radio HAL (https://github.com/GloDroid/mm-radio), same approach can be used to bring-up modem-manager based GNSS HAL, (mm-gnss).

rsglobal, 2025-02-16:
won't do

closed 2025-02-16 by rsglobal