GloDroid/glodroid_manifest #105: Compatibility issue

joek85, 2020-11-14:
hello,
i downloaded the version v0.4.0, compiled successfully, and booted, but stuck at home launching like there is no Home package although Launcher3 does exists.
seems like there is a problem with Compatibility Matrix but don't know where to solve it.
regarding kernel versions compatibility, the current version is 5.8.5, i remember when i worked with v0.2.0 the kernel version was 5+ if i remember correctly and booted successfully.
would you please point me where to solve this problem?

Thanks

` W/VintfObject: VintfObject.verifyWithoutAvb() returns 1: Runtime info and framework compatibility matrix are incompatible: Framework is incompatible with kernel version 5.8.5, compatible kernel versions are 4.9.165 4.9.165 4.9.165 4.9.165 4.9.165 4.9.165 4.9.165 4.9.165 4.9.165 4.9.165 4.14.105 4.14.105 4.14.105 4.14.105 4.14.105 4.14.105 4.14.105 4.14.105 4.14.105 4.19.42 4.19.42 4.19.42 4.19.42 4.19.42 4.19.42 4.19.42 4.19.42 4.19.42`

`2020-11-14 18:19:50.083 345-345/system_process E/Build: Vendor interface is incompatible, error=1
2020-11-14 18:19:50.083 345-345/system_process E/ActivityTaskManager: Build fingerprint is not consistent, warning user
2020-11-14 18:19:50.086 345-345/system_process E/ActivityTaskManager: No home screen found for Intent { act=android.intent.action.MAIN cat=[android.intent.category.HOME] flg=0x100 }
    java.lang.Throwable`

joek85, 2020-11-15:
problem solved:
in build/target/product/handheld_product.mk

`$(call inherit-product, $(SRC_TARGET_DIR)/product/media_product.mk)
#Launcher3QuickStep \
# /product packages
#    QuickSearchBox \
PRODUCT_PACKAGES += \
    Browser2 \
    Calendar \
    Camera2 \
    Contacts \
    DeskClock \
    Email \
    Gallery2 \
    LatinIME \
    Music \
    OneTimeInitializer \
    Provision \
    Settings \
    SettingsIntelligence \
    StorageManager \
    SystemUI \
    WallpaperCropper \
    frameworks-base-overlays`

`#Launcher3QuickStep \
# /product packages
#    QuickSearchBox \`

are malformed comments, which the build didn't include packages below, either remove or correct them like this

`# Launcher3QuickStep
# product packages
# QuickSearchBox`

now those packages are built and booted successfully.
i will keep it open for now, maybe someone has faced this issue.
many thanks.

marynasergiyenko, 2020-11-18:
@joek85 , thank you for your feedback. I propose you to close this issue. If someone faces such problems, he can find it in closed issues.

closed 2020-11-18 by joek85