GloDroid/glodroid_manifest #107: lima, panfrost: pixelwheels game doesn't start

rsglobal, 2020-11-22:
pixelwheels workks on rpi4, but doesn't work on opi3, pinephone.

logcat crash:

```
11-21 21:36:54.900  2297  2325 W GL2JNIView: creating OpenGL ES 2.0 context
11-21 21:36:54.903  2297  2325 E AndroidRuntime: FATAL EXCEPTION: GLThread 212
11-21 21:36:54.903  2297  2325 E AndroidRuntime: Process: com.agateau.tinywheels.android, PID: 2297
11-21 21:36:54.903  2297  2325 E AndroidRuntime: java.lang.IllegalArgumentException
11-21 21:36:54.903  2297  2325 E AndroidRuntime: 	at com.google.android.gles_jni.EGLImpl._eglCreateContext(Native Method)
11-21 21:36:54.903  2297  2325 E AndroidRuntime: 	at com.google.android.gles_jni.EGLImpl.eglCreateContext(EGLImpl.java:63)
11-21 21:36:54.903  2297  2325 E AndroidRuntime: 	at com.badlogic.gdx.backends.android.surfaceview.GLSurfaceView20$ContextFactory.createContext(GLSurfaceView20.java:144)
11-21 21:36:54.903  2297  2325 E AndroidRuntime: 	at android.opengl.GLSurfaceView$EglHelper.start(GLSurfaceView.java:1060)
11-21 21:36:54.903  2297  2325 E AndroidRuntime: 	at android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1431)
11-21 21:36:54.903  2297  2325 E AndroidRuntime: 	at android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1272)
```
Crashed at:
https://cs.android.com/android/platform/superproject/+/master:frameworks/base/core/jni/com_google_android_gles_jni_EGLImpl.cpp;l=223;drc=master

roman-kovalivskyi, 2021-01-09:
After some investigation, I've found that call to EGLConfigChooser::chooseConfig at [GLSurfaceView.java:1054](https://cs.android.com/android/platform/superproject/+/master:frameworks/base/opengl/java/android/opengl/GLSurfaceView.java;drc=53ada2ab282c1b6b72365bc1c6b7aaa29e170eca;l=1054) returns null instead of valid config. [GdxEglConfigChooser](https://github.com/libgdx/libgdx/blob/bc314368a18e0c916437f3dd7b4172714e6abf8b/backends/gdx-backend-android/src/com/badlogic/gdx/backends/android/surfaceview/GdxEglConfigChooser.java#L82) from [libgdx](https://github.com/libgdx/libgdx/tree/bc314368a18e0c916437f3dd7b4172714e6abf8b) is used as an implementation for config chooser for Pixel Wheels.

Seems like libgdx [expects there to be config which would support RGB565 for a fallback option](https://github.com/libgdx/libgdx/blob/bc314368a18e0c916437f3dd7b4172714e6abf8b/backends/gdx-backend-android/src/com/badlogic/gdx/backends/android/surfaceview/GdxEglConfigChooser.java#L101) while trying to find better one. For Orange Pi 3 it fails, since for some reason there is no exact match for desired config, neither available RGB565 config, so it returns null and causes further errors for context creation.

Replacing GdxEglConfigChooser with default [SimpleEGLConfigChooser](https://cs.android.com/android/platform/superproject/+/master:frameworks/base/opengl/java/android/opengl/GLSurfaceView.java;drc=53ada2ab282c1b6b72365bc1c6b7aaa29e170eca;l=1005) by simply removing implementation of [setEGLConfigChooser](https://cs.android.com/android/platform/superproject/+/master:frameworks/base/opengl/java/android/opengl/GLSurfaceView.java;drc=53ada2ab282c1b6b72365bc1c6b7aaa29e170eca;l=405) kinda fixes the issue, SimpleEGLConfigChooser picks first available config and game launches properly, I don't observe any issues, but it couldn't be used as a proper solution to this issue, of course. 

Unfortunately, I wasn't able to check which config libgdx expects in best case, since it's third-party library that comes outside of AOSP, so I think it's better to further investigate onto why there's no EGL config for RGB565 and if it's possible to add such config in one way or another. 

As for now, first call to eglChooseConfig returns 8 configs, each of them has red, green and blue components size equals 8.

rsglobal, 2021-01-09:
Thanks for digging into it.
Just grepping mesa3d sources I can find presence of RGB565 in lima formats, which should be exposed via EGL configs.
Can you share list of those 8 configs?

roman-kovalivskyi, 2021-01-11:
Based on logs from mesa, [`droid_add_configs_for_visuals`](https://elixir.bootlin.com/mesa/latest/source/src/egl/drivers/dri2/platform_android.c#L1184) fails to create configs for `HAL_PIXEL_FORMAT_RGB_565` and `HAL_PIXEL_FORMAT_BGRA_8888`, though latter, as explained in the comment below, is created only if there's no other rgba format available. Configs for `HAL_PIXEL_FORMAT_RGBA_8888` and `HAL_PIXEL_FORMAT_RGBX_8888` is created successfully.

Seems like `dri2_add_config` fails to create config for `HAL_PIXEL_FORMAT_RGB_565` [because of rgba_shifts mismatch](https://elixir.bootlin.com/mesa/latest/source/src/egl/drivers/dri2/egl_dri2.c#L565), for this format it expects shifts to be `{ 11, 5, 0, -1 }`, but it reads only `{ 0, 8, 16, -1 }` or `{ 16, 8, 0, -1 }`. I haven't found yet how does those shifts are filled and why there's no correct shift.

---------------------------------------------

@rsglobal 
> Thanks for digging into it.
> Just by grepping mesa3d sources I can find presence of RGB565 in lima formats, which should be exposed via EGL configs.
> Can you share list of those 8 configs?

Here's my debug logs with rgba, depth and stencil components printed:
```
##> [ComponentSizeChooser::chooseConfig]: choosing between 8 configs 
##> [chooseConfig]: configs[0]:
##>	 depth size = 24
##>	 stencil size = 0
##>	 red size = 8
##>	 green size = 8
##>	 blue size = 8
##>	 alpha size = 0
##> [chooseConfig]: configs[1]:
##>	 depth size = 24
##>	 stencil size = 8
##>	 red size = 8
##>	 green size = 8
##>	 blue size = 8
##>	 alpha size = 0
##> [chooseConfig]: configs[2]:
##>	 depth size = 24
##>	 stencil size = 0
##>	 red size = 8
##>	 green size = 8
##>	 blue size = 8
##>	 alpha size = 0
##> [chooseConfig]: configs[3]:
##>	 depth size = 24
##>	 stencil size = 8
##>	 red size = 8
##>	 green size = 8
##>	 blue size = 8
##>	 alpha size = 0
##> [chooseConfig]: configs[4]:
##>	 depth size = 24
##>	 stencil size = 0
##>	 red size = 8
##>	 green size = 8
##>	 blue size = 8
##>	 alpha size = 8
##> [chooseConfig]: configs[5]:
##>	 depth size = 24
##>	 stencil size = 8
##>	 red size = 8
##>	 green size = 8
##>	 blue size = 8
##>	 alpha size = 8
##> [chooseConfig]: configs[6]:
##>	 depth size = 24
##>	 stencil size = 0
##>	 red size = 8
##>	 green size = 8
##>	 blue size = 8
##>	 alpha size = 8
##> [chooseConfig]: configs[7]:
##>	 depth size = 24
##>	 stencil size = 8
##>	 red size = 8
##>	 green size = 8
##>	 blue size = 8
##>	 alpha size = 8
```
Sorry for not printing exact pixel formats, I'll check which formats exactly are available now via EGL configs.

rsglobal, 2021-01-11:
HAL_PIXEL_FORMAT_RGB_565 added successfully on lima,
HAL_PIXEL_FORMAT_BGRA_8888 is skipped due to https://gitlab.freedesktop.org/mesa/mesa/-/blob/faaba0d6afe0c5f6985345c7c6226435658d196a/src/egl/drivers/dri2/platform_android.c#L1240

Even with both formats pixelwheels still fail.

rsglobal, 2021-05-24:
@roman-kovalivskyi , unassigned due to long inactivity. Please re-assign if you are still working on it.

rsglobal, 2023-03-30:
The game works on recent builds for some reason.

closed 2023-03-30 by rsglobal