GloDroid/glodroid_manifest #108: Support p-boot bootloader and megi's kernel on PinePhone

darkdragon-001, 2020-11-24:
It would be awesome, if this could be integrated with https://xnux.eu/p-boot-demo/

I checked the structure:
- all distros use the [same kernel](https://megous.com/git/linux/tree/README.md?h=private-5.10)
- [p-boot](https://megous.com/git/p-boot/tree/README) bootloader is used
- rootfs is btrfs subvolume

I was wondering, how GloDroid/Android is structured. Is it possible to just flash an img file to sd-card without fastboot etc.?
Which image contains the rootfs?
Are any special kernel parameters needed for boot?

It would be really awesome if one could multi-boot GloDroid alongside Linux distros :blush:

rsglobal, 2020-11-25:
It is possible, but not as easy as on linux distros.
The easiest way is to put glodroid partitions alongside other, and then make p-boot to start u-boot and let it to remaining work.

rsglobal, 2020-11-25:
Here is a partition list which is used by Android:
https://github.com/GloDroid/glodroid_device/blob/1fe99ca9dd51119b07ce1e162b040b673f7c22fe/platform/uboot/bootscript.cpp#L29

darkdragon-001, 2020-11-25:
Is there a way to merge all partitions into one or is it required to have them separately?

rsglobal, 2020-11-25:
fstab is here: https://github.com/GloDroid/glodroid_device/blob/1fe99ca9dd51119b07ce1e162b040b673f7c22fe/common/fstab#L6

I can't tell anything regarding merging partitions into one. Linux can allow to use device mapper to create another block device using single partition as storage, but I've never done this before.

darkdragon-001, 2020-11-25:
And for the underlying file systems: are there any limitations from the android side or can anything supported by the Linux kernel be used (e.g. btrfs)?

Does the kernel need any specific features or could [megi's kernel](https://megous.com/git/linux/tree/?h=orange-pi-5.10&id=3bb6f8ffd1d973e2704445039f6c57d8a62d2268) be used for the PinePhone?

rsglobal, 2020-11-26:
You can change filesystem type. Not sure about btrfs, but likely yes.
We are using megi's kernel with little modifications which is required to run Android.

darkdragon-001, 2020-11-26:
Awesome! 

Are your changes to the kernel documented somewhere? Can regular Linux be used with these changes as well? Would be awesome to use exactly the _same_ kernel (only on filesystem once) for both 😊

rsglobal, 2020-11-26:
Here is kernel sources: https://github.com/GloDroid/glodroid_forks/tree/kernel-sunxi-next
The situation with defconfig is more complex. GloDroid composes it at build time using a lot of fragments.
I still think the best way to include GloDroid into multiimage is to append all partitions AS IS and jump to u-boot to do remaining work.
Also this would consume 6 or more GB of extra space on the final image.

darkdragon-001, 2020-11-26:
Hmmm 🤔 

Including as is doesn't make sense. The cool thing of the multi-boot image is the small size (one kernel, btrfs with zfs compression). 17 distributions are all together about 8GB...

rsglobal, 2020-11-26:
Kernel+BL part is pretty small. Not more than 64MB.
framework+hal part takes about 1.5GB. I believe it may be compressed to 700 or so.
But Android needs some space for usredata partition to store any external information (apks, downloads, etc.).

rsglobal, 2021-05-06:
Android partition layout is too different from linux distros. Looks like nobody wants to start doing this. Closing for now.

closed 2021-05-06 by rsglobal