GloDroid/glodroid_manifest #111: Pi PC Plus

Bauuuuu, 2020-11-28:
Tried on Ornage pi pc plus out of curiosity
Booted fine 
Although
No audio (hdmi or headphone jack)
No wifi 
No Bt

https://gist.github.com/Bauuuuu/689f3eafeefb0fefeb78977441814e47
grabbed log if anyone wants to have a look. I dont have time to look in log but can do some tests in future for this device if anyone wants to fix this
Cheers

rsglobal, 2021-05-25:
PC plus were added by https://github.com/GloDroid/glodroid_device/pull/22 
Wifi is working, jack output / built-in mic is working.
Bluetooth is probably not, untested.

Images will be available in the next release.

closed 2021-05-25 by rsglobal

rsglobal, 2021-05-25:
Images: https://drive.google.com/file/d/1iDvE97-VVUvFCHFrxrxubEmpivVcSCXF/view?usp=sharing