GloDroid/glodroid_manifest #114: fastboot not recognising pinephone

squidink7, 2020-12-12:
I've successfully flashed deploy-sd.img but it boots to a black screen with the LED on. When I run flash-sd.sh it says waiting for device and stops there. I've also tried flashing the old v0.2 iso and then running adb reboot bootloader, which worked but fastboot still doesn't recognise the device. It's appearing in Plasma as pinephone however, and adb commands all work just not fastboot.

squidink7, 2020-12-12:
Fixed it by running as root

closed 2020-12-12 by squidink7

EnfrigFalco, 2021-05-22:
Same issue. Using Fedora 34 and android-tools. Phone has red dot. Fedora shows "waiting for any device". Tried with and without root. Tried re-flashing sd-card as GPT and MBR, tried deploy-sd, deploy-sd-for-emmch, and deploy-gpt. Using latest 'balena etcher' for flashing.