GloDroid/glodroid_manifest #116: *** buffer overflow detected ***: terminated

maximst, 2020-12-18:
I try build GloDroid for Orange Pi PC on Ubuntu 20.04 and get the error:
`
make[1]: вход в каталог «~/build/GloDroid/out/target/product/opi_pc/obj/UBOOT_OBJ»
  GEN     ./Makefile
scripts/kconfig/conf  --syncconfig Kconfig
  CFG     u-boot.cfg
  GEN     include/autoconf.mk.dep
  CFG     spl/u-boot.cfg
  GEN     include/autoconf.mk
  GEN     spl/include/autoconf.mk
  GEN     ./Makefile
  UPD     include/generated/timestamp_autogenerated.h
  PYMOD   scripts/dtc/pylibfdt/_libfdt.so
  Using ~/build/GloDroid/external/u-boot as source for U-Boot
  HOSTCC  tools/mkenvimage.o
  HOSTCC  tools/fit_image.o
  HOSTCC  tools/image-host.o
  HOSTCC  tools/dumpimage.o
  HOSTCC  tools/mkimage.o
  HOSTCC  tools/sunxi-spl-image-builder.o
  HOSTLD  tools/mkenvimage
  HOSTLD  tools/dumpimage
  HOSTLD  tools/mkimage
  HOSTLD  tools/sunxi-spl-image-builder
  CC      cmd/version.o
  CC      common/main.o
  LD      cmd/built-in.o
  LD      common/built-in.o
  CC      drivers/fastboot/fb_getvar.o
  LD      drivers/fastboot/built-in.o
  LD      drivers/built-in.o
  CC      lib/smbios.o
  CC      lib/display_options.o
  LD      lib/built-in.o
  LD      u-boot
*** buffer overflow detected ***: terminated
Aborted (core dumped)
make[1]: *** [~/build/GloDroid/external/u-boot/Makefile:1755: u-boot] Error 134
`
This happens on any tags

elvoblin, 2020-12-21:
+1
I have same problem

rsglobal, 2021-01-10:
Hi,

What is your host OS?

elvoblin, 2021-01-10:
> What is your host OS?

Ubuntu 20.04

rsglobal, 2021-01-10:
We are also using ubuntu 20.04, 20.10 an there was no such issues.
Are you using standard terminal app?

elvoblin, 2021-01-10:
yes.
**Терминал GNOME
3.36.2
Эмулятор терминала для рабочего стола GNOME
Используется VTE версии 0.60.3 +BIDI +GNUTLS +ICU +SYSTEMD**

```
$echo $BASH_VERSION

5.0.17(1)-release
```

```
$cat /etc/lsb-release

DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.1 LTS"
```

maximst, 2021-02-09:
Hello!
I decided to collect GloDroid on ubuntu 18.04 in virtualbox VM with  11Gb of RAM. 
And the process starts but fails with an error:
```
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=11
TARGET_PRODUCT=opi_pc
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm
TARGET_ARCH_VARIANT=armv7-a-neon
TARGET_CPU_VARIANT=generic
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-4.15.0-132-generic-x86_64-Ubuntu-18.04.5-LTS
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=RP1A.200720.011
OUT_DIR=out
============================================
[ 16% 707/4252] //frameworks/base:api-stubs-docs-non-updatable metalava merged
FAILED: out/soong/.intermediates/frameworks/base/api-stubs-docs-non-updatable/android_common/api-stubs-docs-non-updatable-stubs.srcjar out/soong/.intermediates/frameworks/base/api-stubs-docs-non-updatable/android_common/api-stubs-docs-non-updatable-violations.txt out/soong/.intermediates/frameworks/base/api-stubs-docs-non-updatable/android_common/api-stubs-docs-non-updatable_annotations.zip out/soong/.intermediates/frameworks/base/api-stubs-docs-non-updatable/android_common/api-stubs-docs-non-updatable_api.txt out/soong/.intermediates/frameworks/base/api-stubs-docs-non-updatable/android_common/api-stubs-docs-non-updatable_removed.txt out/soong/.intermediates/frameworks/base/api-stubs-docs-non-updatable/android_common/api_lint.timestamp out/soong/.intermediates/frameworks/base/api-stubs-docs-non-updatable/android_common/api_lint_report.txt
rm -rf out/soong/.intermediates/frameworks/base/api-stubs-docs-non-updatable/android_common/stubsDir && mkdir -p out/soong/.intermediates/frameworks/base/api-stubs-docs-non-updatable/android_common/stubsDir && rm -rf out/soong/.intermediates/frameworks/base/api-stubs-docs-non-updatable/android_common/srcjars && mkdir -p out/soong/.intermediates/frameworks/base/api-stubs-docs-non-updatable/android_common/srcjars && out/soong/host/linux-x86/bin/zipsync -d out/soong/.intermediates/frameworks/base/api-stubs-docs-non-updatable/android_common/srcjars -l out/soong/.intermediates/frameworks/base/api-stubs-docs-non-updatable/android_common/srcjars/list -f "*.java" out/soong/.intermediates/frameworks/base/core/res/framework-res/android_common/gen/android/R.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/proto/src/ipconnectivity.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/privacy.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/section.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/activitymanager.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/alarmmanager.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/appexit_enums.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/appexitinfo.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/enums.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/media_output_enum.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/notification.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/notification_channel.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/notification_channel_group.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/notificationmanager.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/pendingintent.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/profilerinfo.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/settings_enums.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/statusbarmanager.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/tvsettings_enums.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/window_configuration.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/job/enums.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/bluetooth/enums.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/bluetooth/a2dp/enums.srcjar out/soong/.intermediates/ [... cut ...]
        at com.intellij.psi.impl.source.tree.SharedImplUtil.acceptChildren(SharedImplUtil.java:200)
        at com.intellij.psi.impl.source.JavaStubPsiElement.acceptChildren(JavaStubPsiElement.java:117)
        at com.intellij.psi.JavaRecursiveElementVisitor.visitElement(JavaRecursiveElementVisitor.java:43)
        at com.intellij.psi.JavaElementVisitor.visitClass(JavaElementVisitor.java:40)
16:51:10 ninja failed with: exit status 1
```

I try set **_JAVA_OPTIONS** variable to **-Xmx8g**, but it doesn't help

maximst, 2021-03-03:
@MaksymPrymierov, @matviizorin
Гебята, земляки(Харьков лучший город земли :-)), прошу вас помогите собрать билд под orangepi pc c параметром DEVICE_TYPE=tv (я думаю если вы добавите в релизы сборки андроид тв, то это будет гораздо нужнее чем простые билды. Потому, что в чистом линуксе в одноплатниках хоть и работет ускорение видео в плеерах, но в браузерах это невозможно, поэтому например приложение ютуб не создать похожим на то что юзается в комерческих смарт тв. А tizen напрример хрен соберешь для платки отличной от малины.) Вобщем андроид тв сборки под китайские одноплатники были-б просто прорывом :-)

rsglobal, 2021-03-04:
Hi,
Please use English in the public domain. Feel free to contact us privately on Discord in case you prefer using Russian or Ukrainian.

ATV builds won't give you any performance advantages. It has just a different UI and settings menu. Nothing more.
We have already enabled accelerated video scaling support for the Android, so you can watch 360p videos on 4k displays without any performance degradation. Unfortunately video decoder still uses CPU. We hope things will change after Android-12 release from Google, and we will be able to play 1080p+ videos in good performance.

joek85, 2021-03-22:
try to decrease the make jobs, for example make -j8, depends on the number of cores your CPU has, double that number by 2, if you have 4 cores CPU use make -j8 since modern CPUs have 2 threads per core, also consider lower that number because if it's high you may overload the memory bus speed.

elvoblin, 2021-03-22:
Unfortunately it didn't help:
```
$ make -j2 images
......

$ cat out/error.1.log | head
FAILED: Building U-Boot: 
Outputs: out/target/product/opi_plus2e/obj/UBOOT_OBJ/u-boot-sunxi-with-spl.bin
Error: exited with code: 1
Command: /bin/bash -c "(echo \"TARGET_PRODUCT = \" opi_plus2e: ) && (mkdir -p out/target/product/opi_plus2e/obj/UBOOT_OBJ ) && (PATH=/usr/bin:/bin:\$PATH ARCH=arm CROSS_COMPILE=\$(readlink -f prebuilts/gcc/linux-x86/arm/gcc-linaro-arm-linux-gnueabihf/bin/arm-linux-gnueabihf-)  make -j4 -C external/u-boot O=\$(readlink -f out/target/product/opi_plus2e/obj/UBOOT_OBJ) orangepi_plus2e_defconfig ) && (PATH=/usr/bin:/bin external/u-boot/scripts/kconfig/merge_config.sh -m -O out/target/product/opi_plus2e/obj/UBOOT_OBJ/ out/target/product/opi_plus2e/obj/UBOOT_OBJ/.config device/glodroid/platform/common/sunxi/uboot.config device/glodroid/platform/common/uboot.config device/glodroid/platform/common/sunxi/uboot.config device/glodroid/platform/common/uboot-sd.config ) && (PATH=/usr/bin:/bin:\$PATH ARCH=arm CROSS_COMPILE=\$(readlink -f prebuilts/gcc/linux-x86/arm/gcc-linaro-arm-linux-gnueabihf/bin/arm-linux-gnueabihf-)  make -j4 -C external/u-boot O=\$(readlink -f out/target/product/opi_plus2e/obj/UBOOT_OBJ) olddefconfig ) && (PATH=/usr/bin:/bin:\$PATH ARCH=arm CROSS_COMPILE=\$(readlink -f prebuilts/gcc/linux-x86/arm/gcc-linaro-arm-linux-gnueabihf/bin/arm-linux-gnueabihf-)  make -j4 -C external/u-boot O=\$(readlink -f out/target/product/opi_plus2e/obj/UBOOT_OBJ) KCFLAGS=\"-fgnu89-inline \" ) && (cp out/target/product/opi_plus2e/obj/UBOOT_OBJ/u-boot-sunxi-with-spl.bin out/target/product/opi_plus2e/obj/UBOOT_OBJ/u-boot-sunxi-with-spl.bin.sd ) && (PATH=/usr/bin:/bin:\$PATH ARCH=arm CROSS_COMPILE=\$(readlink -f prebuilts/gcc/linux-x86/arm/gcc-linaro-arm-linux-gnueabihf/bin/arm-linux-gnueabihf-)  make -j4 -C external/u-boot O=\$(readlink -f out/target/product/opi_plus2e/obj/UBOOT_OBJ) orangepi_plus2e_defconfig ) && (PATH=/usr/bin:/bin external/u-boot/scripts/kconfig/merge_config.sh -m -O out/target/product/opi_plus2e/obj/UBOOT_OBJ/ out/target/product/opi_plus2e/obj/UBOOT_OBJ/.config device/glodroid/platform/common/sunxi/uboot.config device/glodroid/platform/common/uboot.config device/glodroid/platform/common/sunxi/uboot.config device/glodroid/platform/common/uboot-emmc.config ) && (PATH=/usr/bin:/bin:\$PATH ARCH=arm CROSS_COMPILE=\$(readlink -f prebuilts/gcc/linux-x86/arm/gcc-linaro-arm-linux-gnueabihf/bin/arm-linux-gnueabihf-)  make -j4 -C external/u-boot O=\$(readlink -f out/target/product/opi_plus2e/obj/UBOOT_OBJ) olddefconfig ) && (PATH=/usr/bin:/bin:\$PATH ARCH=arm CROSS_COMPILE=\$(readlink -f prebuilts/gcc/linux-x86/arm/gcc-linaro-arm-linux-gnueabihf/bin/arm-linux-gnueabihf-)  make -j4 -C external/u-boot O=\$(readlink -f out/target/product/opi_plus2e/obj/UBOOT_OBJ) KCFLAGS=\"-fgnu89-inline \" ) && (cp out/target/product/opi_plus2e/obj/UBOOT_OBJ/u-boot-sunxi-with-spl.bin out/target/product/opi_plus2e/obj/UBOOT_OBJ/u-boot-sunxi-with-spl.bin.emmc )"
Output:
TARGET_PRODUCT =  opi_plus2e:
make: вход в каталог «/home/voblin/work/orange/GloDroid/external/u-boot»
make[1]: вход в каталог «/home/voblin/work/orange/GloDroid/out/target/product/opi_plus2e/obj/UBOOT_OBJ»
  GEN     ./Makefile
  HOSTCC  scripts/basic/fixdep
```

elvoblin, 2021-05-25:
build using docker container was succesfull

closed 2021-05-26 by rsglobal