GloDroid/glodroid_manifest #119: Password options missing

squidink7, 2020-12-19:
In settings on my pinephone there is only none and swipe available. I have tried searching for password and pin which lead me back to there as if those options are there but just hidden. This along with sleep mode (which I got working with an app on fdroid that triggers it through a quick setting tile) are the only things keeping me from using this as a daily driver. Is screen locking with a pin/pattern/etc. intentionally disabled on your part? Or is there something I'm missing.

lm41, 2021-02-01:
You can get the sleep mode to work, if you revert this Commit https://github.com/GloDroid/glodroid_device/commit/90994b291ca55a4d747ecf21ab1b8bf2a5fc3136

lm41, 2021-02-01:
"I have a hunch that the reason the screen lock option isn't available is because these are userdebug builds, as the user ones aren't working, but I don't know for sure if that's why." 
https://discord.com/channels/753603904406683670/753608918504177765/792876320332841010

lm41, 2021-02-01:
I had the same issue, and glodroid is for me nice to have, but not my daily driver (yet).

rsglobal, 2021-05-12:
Lock screen is enabled on v0.6.1 . It looks like the password prompt require special application. I can't find anything similar in AOSP yet.

closed 2021-11-20 by rsglobal