GloDroid/glodroid_manifest #120: Increase super partition size pinephone

squidink7, 2020-12-20:
I want to try out some GSI images on my pinephone but I keep getting out of space errors. Would it be possible to increase the size of the super partition? It doesn't need to be by much just enough to fit about a 1GB system img.

rsglobal, 2020-12-20:
Sure. Are you using images only or you can also build from sources?

squidink7, 2020-12-20:
I've tried to build from sources but couldn't seem to get it to work so if you could push an image with it that would be awesome, thank you so much.

rsglobal, 2021-05-12:
I tried run GSI on 0.6.1 without any success:

To increase super-partition you need the following changes:
```diff
diff --git a/common/boardconfig-common.mk b/common/boardconfig-common.mk
index 36b0ca3..db79dfe 100644
--- a/common/boardconfig-common.mk
+++ b/common/boardconfig-common.mk
@@ -71,6 +71,6 @@ BOARD_EXT4_SHARE_DUP_BLOCKS := true
 # Dynamic partition
-BOARD_SUPER_PARTITION_SIZE := 1468006400
+BOARD_SUPER_PARTITION_SIZE := 3355443200
 BOARD_BUILD_SUPER_IMAGE_BY_DEFAULT := true
 BOARD_SUPER_PARTITION_GROUPS := glodroid_dynamic_partitions
-BOARD_GLODROID_DYNAMIC_PARTITIONS_SIZE := 1466957824
+BOARD_GLODROID_DYNAMIC_PARTITIONS_SIZE := 3355377664
 BOARD_GLODROID_DYNAMIC_PARTITIONS_PARTITION_LIST := system system_ext vendor product
diff --git a/platform/uboot/bootscript.cpp b/platform/uboot/bootscript.cpp
index ecc334b..d8b533a 100644
--- a/platform/uboot/bootscript.cpp
+++ b/platform/uboot/bootscript.cpp
@@ -45,3 +45,3 @@ EXTENV(partitions, ";name=dtbo_a,size=8M,uuid=\${uuid_gpt_dtbo_a}")
 EXTENV(partitions, ";name=vbmeta_a,size=512K,uuid=\${uuid_gpt_vbmeta_a}")
-EXTENV(partitions, ";name=super,size=1400M,uuid=\${uuid_gpt_super}")
+EXTENV(partitions, ";name=super,size=3200M,uuid=\${uuid_gpt_super}")
 EXTENV(partitions, ";name=metadata,size=16M,uuid=\${uuid_gpt_metadata}")
```

Images: https://mega.nz/#!5X5SzTSB!IJm6KKZt40gWdb-dxtiDxuw_0ZQKGsCjqmjod9Vq_FI

closed 2021-05-12 by rsglobal

rsglobal, 2021-05-12:
I suspect GSI was build for `user` variant, but GD has some selinux rules missing to run it