GloDroid/glodroid_manifest #121: Issue with PINE64's Listing of GloDroid

lapidary-master, 2020-12-27:
There is a slight issue with the PINE64 listing of GloDroid and I feel as though, the owner of the project should at the very least attempt to address it

When you visit the PINE64 website and navigate to the "PinePhone Software Releases" and scroll down the page or use the hyperlink at the top of the page to GloDroid, PINE64 has marked GloDroid with the following warning;

> **Warning:** The development of the OS appears to be halted, available images are outdated.

Now I don't know what would cause this to be flagged as such but I just wanted to bring this to the primary developer's attention.

rakor, 2021-01-10:
Seens as if they are right.

rsglobal, 2021-01-10:
Hi Guys.

1. Development of GloDroid isn't halted.
2. GloDroid in not an OS. We just allow to run AOSP on various hardware using opensource stack that previously wasn't possible. Android is actively developed by Google and other communities. 
3. We are not responsible for any information on non-our sites.
4. Currently we have almost no relationship with the pine64 team.
5. In case pine64 team tracks our releases, `Warning:` label will will be removed soon after we will publish next release.

closed 2021-01-10 by rsglobal