GloDroid/glodroid_manifest #122: How to add a platform based on GloDroid

10cl, 2021-01-04:
This project is awesome!!!
in order to learn this project, I bought a pi3 then make images, flashing....
last, I found it was too slow. so, I want know how to add pi4 based on GloDroid.

rsglobal, 2021-01-04:
Hi. 
Support for OPI4 will be added soon. Please wait for a 2-3 weeks.

10cl, 2021-03-21:
Hi，
Great job! 
But the prebuilt image cannot be displayed by HDMI,   `adb shell screencap`  is passed.

rsglobal, 2021-03-23:
We had similar issues on some displays. This needs more investigation.

rsglobal, 2021-05-19:
It turned out that Orangepi-4 doesn't have mainline community support.
This is bad news for us, since GloDroid is relying mostly on it.
Currently we are using orangepi rk3399 board device tree files, which can cause problems with hdmi and other components.