GloDroid/glodroid_manifest #123: App with Osmdroid does not load map

jpmeijers, 2021-01-17:
I have a simple Android app that I am trying to run on GloDroid on the Pinephone. The app uses osmdroid to show a map on the screen in a fragment. When I run the app the map does not load and I see the following error in logcat:

```
2020-09-24 22:36:10.448 2652-2652/org.ttnmapper.phonesurveyor I/RenderThread: type=1400 audit(0.0:1601): avc: denied { read } for name="dri" dev="tmpfs" ino=675 scontext=u:r:untrusted_app:s0:c97,c256,c512,c768 tcontext=u:object_r:device:s0 tclass=dir permissive=1 app=com.mydomain.myapp
2020-09-24 22:36:10.448 2652-2652/org.ttnmapper.phonesurveyor I/RenderThread: type=1400 audit(0.0:1602): avc: denied { open } for path="/dev/dri" dev="tmpfs" ino=675 scontext=u:r:untrusted_app:s0:c97,c256,c512,c768 tcontext=u:object_r:device:s0 tclass=dir permissive=1 app=com.mydomain.myapp
2020-09-24 22:36:10.452 2652-2652/org.ttnmapper.phonesurveyor I/RenderThread: type=1400 audit(0.0:1604): avc: denied { open } for path="/sys/devices/platform/soc/1c40000.gpu/uevent" dev="sysfs" ino=3355 scontext=u:r:untrusted_app:s0:c97,c256,c512,c768 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1 app=com.mydomain.myapp
```

I also reported this issue on osmdroid, but I believe it is an issue with selinux permissions in GloDroid, as the app works fine on my other Android phones.

https://github.com/osmdroid/osmdroid/issues/1679

jpmeijers, 2021-01-17:
The map actually loaded after I realised the wifi was not connected. It is however still strange that I get these errors printed in logcat.

rsglobal, 2021-01-18:
Hi.

selinux rules is on our todo list. Issue isn't top priority since it only produces warnings.

hamiltonrn, 2021-01-18:
**Signature Spoofing** is a very desirable feature, there's some LineageOS with it, no need for Xposed.

closed 2021-11-26 by rsglobal