GloDroid/glodroid_manifest #126: Install µG (microg)

jpmeijers, 2021-01-17:
I have a Pinephone running GloDroid and I am trying to install microg as a replacement for Google's play services. I need this for apps that require location services.

https://microg.org/download.html

Currently I am struggling because microg requires "signature spoofing", provided by "Xposed". I tried installing Xposed Installer from f-droid, but it tells me "No matching ZIPs were found".

I do not know what to do next. Does anyone have recommendations?

hamiltonrn, 2021-01-18:
Native **Signature Spoofing** is a very desirable feature, some LineageOS has it.

nas00, 2021-01-21:
Someone suggested magisk in another issue - and it [worked](https://forum.pine64.org/showthread.php?tid=10613&pid=74279#pid74279) .

jpmeijers, 2021-01-22:
I'm a little confused by the steps provided:

`1) patch boot.img with magisk app` - The app is on the phone, so do I need to copy boot.img from the artifact I downloaded from Github to the phone?
`2) adb reboot fastboot` - ADB is running on my computer, so do I need to copy the patched boot.img back to my computer?
`3) fastboot flash boot_a magisk_patched.img`
`4) fastboot reboot`

nas00, 2021-01-22:
Yes and yes - there's a link to more detailed instructions in the same post.

jpmeijers, 2021-01-22:
I scanned through the rest of the thread and can not find any instruction about this. Because the Pinephone does not want to mount as a USB storage device on my computer, I used another Android phone to do the patching of boot.img. I then followed the instructions to flash it to the Pinephone, but after that it did not want to boot. Screen is on but completely black.

I then decided to modify `flash-emmc.sh` to use `magisk_patched.img` rather than `boot.img` for the `boot_a` step. After this I see the same result as before.

My assumption is that pathing boot.img failed. Could it be because I used another android phone to do this?

nas00, 2021-01-22:
Click the word 'magisk' in the linked post. I haven't tried it for a while so will get round to checking it still works at some point.

jpmeijers, 2021-01-22:
As recommended there I used the Canary build of Magisk. This time running and patching the boot.img on the Pinephone. Getting the boot.img image onto and off of the Pinephone is still very difficult. I ended up using a simple ftp server app, which seems to work if one use the SAF option.

Boot_a partition was again flashed with the patched boot.img, but again only a black screen on startup.

beckerichmika, 2021-01-23:
So is signature spoofing still looked into or are you trying to get Magisk to install microg? There are apps that don't require the original Gapps, but they don't work if they find any trace of magisk on the system. I tgink both options should be possible.

nas00, 2021-01-23:
magisk offers signature spoofing and an option to hide itself but will probably need a user build of glodroid (https://github.com/GloDroid/glodroid_manifest/issues/97) to test those apps out.

jpmeijers, 2021-01-23:
Main goal is to have working location services. To get that I want to install microg.

nas00, 2021-01-23:
So I gave it a go and confirm the magisk patched boot file doesn't work on the latest release. Seeing that people are reporting similar issues on other devices (https://github.com/topjohnwu/Magisk/issues/3722), it's probably not glodroid specific but attached a log anyway.
 
[glodroidv0.5.1_magisk-canary-app-debug-apk-f5593e05.log](https://github.com/GloDroid/glodroid_manifest/files/5860961/glodroidv0.5.1_magisk-canary-app-debug-apk-f5593e05.log)

genofire, 2022-04-01:
Maybe add preinstalled F-Droid either: https://github.com/GloDroid/glodroid_external_apks/pull/3/files

cyclone018, 2022-04-21:
I've tried to get microG to work on Android 12 but the Signature Spoofing with Edsposed and Riru with Magisk didn't work for me... Like at all. But I was able to get the original Gapps to work. With a little tweaking, a downgrade, and a registration, it works but not at a desirable state. Like almost 90% of the apps I use on a daily basis say "Not Supported" but anyway. 

[image]

rsglobal, 2023-03-23:
LineageOS+GMS(microg) is available. See https://github.com/GloDroidCommunity/pine64-pinephone.
No rebuild images for now, but they can be built from sources.