GloDroid/glodroid_manifest #127: orange pi 3 usb otg failed

0lxb, 2021-01-24:
```
U-Boot SPL 2020.07-rc4-gc046ee6a3-dirty (Jan 23 2021 - 06:06:19 +0000)
DRAM: 2048 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.3(debug):de9d0d7c7
NOTICE:  BL31: Built : 04:28:33, Jan 23 2021
NOTICE:  BL31: Detected Allwinner H6 SoC (1728)
NOTICE:  BL31: Found U-Boot DTB at 0xc090328, model: OrangePi 3
INFO:    ARM GICv2 driver initialized
INFO:    PMIC: Probing AXP805 on I2C
INFO:    PMIC: aldo1 voltage: 3.300V
INFO:    PMIC: aldo2 voltage: 3.300V
INFO:    PMIC: aldo3 voltage: 3.300V
INFO:    PMIC: bldo1 voltage: 1.800V
INFO:    PMIC: bldo2 voltage: 1.800V
INFO:    PMIC: cldo1 voltage: 3.300V
INFO:    PMIC: dcdcd voltage: 0.960V
INFO:    PMIC: dcdce voltage: 1.200V
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
NOTICE:  PSCI: System suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9

U-Boot 2020.07-rc4-gc046ee6a3-dirty (Jan 23 2021 - 06:06:19 +0000) Allwinner Technology

CPU:   Allwinner H6 (SUN50I)
Model: OrangePi 3
DRAM:  2 GiB
MMC:   Device 'mmc@4022000': seq 1 is in use by 'mmc@4021000'
mmc@4020000: 0, mmc@4021000: 2, mmc@4022000: 1
Loading Environment from FAT... Unable to use mmc 1:1... In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Allwinner Sun4I USB PHY driver loaded
sun4i_usb_phy_xlate: phy_id = 0
Allwinner mUSB OTG (Peripheral)
Net:
Warning: usb_ether MAC addresses don't match:
Address in ROM is               de:ad:be:ef:00:01
Address in environment is       02:07:ab:2f:93:2a
eth0: usb_ether
Hit any key to stop autoboot:  0
3655 bytes read in 2 ms (1.7 MiB/s)
## Executing script at 53000000
Could not find "misc" partition
** Bad device specification mmc 0#misc **
Couldn't find partition mmc 0#misc
starting USB...
Bus usb@5101000: USB EHCI 1.00
Bus usb@5101400: USB OHCI 1.0
Bus usb@5311000: sun4i_usb_phy_xlate: phy_id = 3
USB EHCI 1.00
Bus usb@5311400: sun4i_usb_phy_xlate: phy_id = 3
USB OHCI 1.0
scanning bus usb@5101000 for devices... 1 USB Device(s) found
scanning bus usb@5101400 for devices... 1 USB Device(s) found
scanning bus usb@5311000 for devices... 1 USB Device(s) found
scanning bus usb@5311400 for devices... 1 USB Device(s) found
sun4i_usb_phy_xlate: phy_id = 0
sun4i_usb_phy_vbus_detect: id_det = -22
musb-hdrc: peripheral reset irq lost!
```

rsglobal, 2021-01-24:
Hi,

Looks like normal u-boot log.
Ensure you've installed `sudo apt install adb` and continue flashing by flash-sd.sh

0lxb, 2021-01-25:
> Hi,
> 
> Looks like normal u-boot log.
> Ensure you've installed `sudo apt install adb` and continue flashing by flash-sd.sh

Thank you for your reply！
The flash-sd.sh error log is as follows.

`➜  orangepi3 ./flash-sd.sh    
+ ./fastboot flash gpt deploy-gpt.img
Sending 'gpt' (128 KB)                             OKAY [  0.018s]
Writing 'gpt'                                      OKAY [  0.017s]
Finished. Total time: 0.036s
+ ./fastboot flash bootloader bootloader-sd.img
Sending 'bootloader' (1920 KB)                     OKAY [  0.181s]
Writing 'bootloader'                               OKAY [  0.175s]
Finished. Total time: 0.367s
+ ./fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.029s]
Writing 'uboot-env'                                OKAY [  0.033s]
Finished. Total time: 0.076s
+ ./fastboot reboot
Rebooting                                          OKAY [  0.000s]
Finished. Total time: 0.000s
+ set +x
+ ./fastboot oem format
ERROR: could not clear input pipe; result e00002eb, ignoring...
ERROR: could not clear output pipe; result e0004061, ignoring....
ERROR: usb_write failed with status e00002c0
FAILED (Write to device failed (No such file or directory))
fastboot: error: Command failed`

joek85, 2021-03-22:
Is there any USB hub connected between your computer and the board?

0lxb, 2021-10-16:
Sorry for not replying in time.
It's OK now.
May be a problem with the USB cable.

closed 2021-11-20 by rsglobal