GloDroid/glodroid_manifest #134: Build failed, DD can not open file: out/target/product/opi_plus2e/sdcard.img : Directory or file does not exists

mimoccc, 2021-03-18:
FAILED: out/target/product/opi_plus2e/sdcard.img
/bin/bash -c "PATH=/sbin:/bin:/usr/bin:\$PATH device/glodroid/platform/tools/gensdimg.sh -C=out/target/product/opi_plus2e -T=SD -P=s
unxi out/target/product/opi_plus2e/sdcard.img"
===> Create raw disk image
dd: nepodarilo sa otvoriť 'out/target/product/opi_plus2e/sdcard.img': Adresár alebo súbor neexistuje

mimoccc, 2021-03-18:
Got it.
First call :
source ./build/envsetup.sh
lunch
make images
Then:
make sdcard

mimoccc, 2021-03-18:
Got it.

It need to build first images.

source ./build/envsetup.sh
lunch opi_plus2e-userdebug
make images
make sdcard

closed 2021-03-18 by mimoccc

rsglobal, 2021-03-18:
`make sdcard` is an old rule, and we are not using it anymore.
Please refer to manual in the README.md for latest guidelines.