GloDroid/glodroid_manifest #137: App crash - logs provided

Robert-L-Turner, 2021-04-30:
Hi Folks,

Pretty inexperienced at android debugging and linux phones but I'd welcome any feedback on the below.  Fresh SD install of glodroid onto a Pinephone CE.  

Installed the com.anki.cozmo 3.4.0 apk and keep getting a download failure with the attached logs.  Two errors jump out:

_ILicensingService not found_ and an exception around the _unity3d.plugin.downloader._  Could anyone point me in the right direction?

```
Hi Folks,

Pretty inexperienced at android debugging and linux phones but I'd welcome any feedback on the below.  Fresh SD install of glodroid onto a Pinephone CE.  

Installed the com.anki.cozmo 3.4.0 apk and keep getting a download failure with the attached logs.  Two errors jump out:

ILicensingService not found and an exception around the unity3d.plugin.downloader.  Could anyone point me in the right direction?04-30 05:12:59.553   472   789 I ActivityTaskManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=com.anki.cozmo/.CozmoActivity bnds=[562,171][698,379]} from uid 10085
04-30 05:12:59.557   472   492 D EventSequenceValidator: Transition from ACTIVITY_FINISHED to INTENT_STARTED
04-30 05:12:59.558   472   487 I system_server: oneway function results will be dropped but finished with status OK and parcel size 4
04-30 05:12:59.564   472   487 I chatty  : uid=1000(system) Binder:472_2 identical 1 line
04-30 05:12:59.574   472   487 I system_server: oneway function results will be dropped but finished with status OK and parcel size 4
04-30 05:12:59.578   378   378 I perfetto: ing_service_impl.cc:758 Configured tracing session 14, #sources:1, duration:5000 ms, #buffers:1, total buffer size:4096 KB, total sessions:1, uid:1071 session name: ""
04-30 05:12:59.579   377   377 I perfetto: probes_producer.cc:230  Ftrace setup (target_buf=14)
04-30 05:12:59.607     0     0 W perfetto: enabled ftrace
04-30 05:12:59.594   472   492 D CompatibilityChangeReporter: Compat change id reported: 135634846; UID 10097; state: DISABLED
04-30 05:12:59.595   472   499 D CompatibilityChangeReporter: Compat change id reported: 143937733; UID 10097; state: DISABLED
04-30 05:12:59.611   472   492 D EventSequenceValidator: Transition from INTENT_STARTED to ACTIVITY_LAUNCHED
04-30 05:12:59.615   377   377 I perfetto: ftrace_procfs.cc:176    enabled ftrace
04-30 05:12:59.625   310   310 D Zygote  : Forked child process 2088
04-30 05:12:59.632   472   499 I ActivityManager: Start proc 2088:com.anki.cozmo/u0a97 for pre-top-activity {com.anki.cozmo/com.anki.cozmo.CozmoActivity}
04-30 05:12:59.637  2088  2088 I Zygote  : seccomp disabled by setenforce 0
04-30 05:12:59.656   472   493 I GRALLOC-GBM: lock bo 0xb400f7bf774f5b10, cnt=0, usage=33
04-30 05:12:59.696   375   414 I adbd    : jdwp connection from 2088
04-30 05:12:59.739   472   771 I system_server: oneway function results will be dropped but finished with status OK and parcel size 4
04-30 05:12:59.749  2088  2088 D ApplicationLoaders: Returning zygote-cached class loader: /system/framework/android.test.base.jar
04-30 05:12:59.759  2088  2088 I com.anki.cozmo: The ClassLoaderContext is a special shared library.
04-30 05:12:59.826  2088  2088 D NetworkSecurityConfig: No Network Security Config specified, using platform default
04-30 05:12:59.827  2088  2088 D NetworkSecurityConfig: No Network Security Config specified, using platform default
04-30 05:12:59.924  2088  2088 I com.anki.cozmo: Thread[1,tid=2088,Native,Thread*=0xe5804610,peer=0x71c02598,"main"] recursive attempt to load library "/data/app/~~L-e-twQQoMtA3e67jV2nyw==/com.anki.cozmo-M6KH28kpg01eiHKFQKD0qQ==/lib/arm/libDAS.so"
04-30 05:13:00.202   627   627 D KeyguardClockSwitch: Updating clock: 513
04-30 05:13:00.228     0     0 W healthd : battery l=46 v=3804 t=0.0 h=2 st=3 c=85000 chg=u
04-30 05:13:00.239     0     0 W audit   : audit_lost=24 audit_rate_limit=5 audit_backlog_limit=64
04-30 05:13:00.239     0     0 E audit   : rate limit exceeded
04-30 05:13:00.221   320   320 I health@2.0-serv: type=1400 audit(0.0:236): avc: denied { read } for name="present" dev="sysfs" ino=20361 scontext=u:r:hal_health_default:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
04-30 05:13:00.282   472   492 W ActivityTaskManager: Current config: {1.0 ?mcc?mnc [en_US] ldltr sw428dp w428dp h784dp 269dpi nrml long port -touch -keyb/v/h -nav/h winConfig={ mBounds=Rect(0, 0 - 720, 1440) mAppBounds=Rect(0, 0 - 720, 1359) mWindowingMode=fullscreen mDisplayWindowingMode=fullscreen mActivityType=undefined mAlwaysOnTop=undefined mRotation=ROTATION_0} s.182} unchanged for IME proc com.android.inputmethod.latin
04-30 05:13:00.221   320   320 I health@2.0-serv: type=1400 audit(0.0:237): avc: denied { open } for path="/sys/devices/platform/soc/1f03400.rsb/sunxi-rsb-3a3/axp20x-battery-power-supply/power_supply/axp20x-battery/present" dev="sysfs" ino=20361 scontext=u:r:hal_health_default:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
04-30 05:13:00.286   472   492 I ActivityTaskManager: Config changes=20000480 {1.0 ?mcc?mnc [en_US] ldltr sw428dp w808dp h404dp 269dpi nrml long land -touch -keyb/v/h -nav/h winConfig={ mBounds=Rect(0, 0 - 1440, 720) mAppBounds=Rect(0, 0 - 1359, 720) mWindowingMode=fullscreen mDisplayWindowingMode=fullscreen mActivityType=undefined mAlwaysOnTop=undefined mRotation=ROTATION_90} s.209}
04-30 05:13:00.314   472   492 W ActivityTaskManager: Current config: {1.0 ?mcc?mnc [en_US] ldltr sw428dp w428dp h784dp 269dpi nrml long port -touch -keyb/v/h -nav/h winConfig={ mBounds=Rect(0, 0 - 720, 1440) mAppBounds=Rect(0, 0 - 720, 1359) mWindowingMode=fullscreen mDisplayWindowingMode=fullscreen mActivityType=undefined mAlwaysOnTop=undefined mRotation=ROTATION_0} s.182} unchanged for IME proc com.android.inputmethod.latin
04-30 05:13:00.323   472   492 I WindowManager: Override config changes=20000480 {1.0 ?mcc?mnc [en_US] ldltr sw428dp w808dp h404dp 269dpi nrml long land -touch -keyb/v/h -nav/h winConfig={ mBounds=Rect(0, 0 - 1440, 720) mAppBounds=Rect(0, 0 - 1359, 720) mWindowingMode=fullscreen mDisplayWindowingMode=fullscreen mActivityType=undefined mAlwaysOnTop=undefined mRotation=ROTATION_90} s.209} for displayId=0
04-30 05:13:00.327   472   492 V ActivityTaskManager: Sending to IME proc com.android.inputmethod.latin new config {1.0 ?mcc?mnc [en_US] ldltr sw428dp w808dp h404dp 269dpi nrml long land -touch -keyb/v/h -nav/h winConfig={ mBounds=Rect(0, 0 - 1440, 720) mAppBounds=Rect(0, 0 - 1359, 720) mWindowingMode=fullscreen mDisplayWindowingMode=fullscreen mActivityType=undefined mAlwaysOnTop=undefined mRotation=ROTATION_90} s.209}
04-30 05:13:00.329   472   492 W ActivityTaskManager: Current config: {1.0 ?mcc?mnc [en_US] ldltr sw428dp w808dp h404dp 269dpi nrml long land -touch -keyb/v/h -nav/h winConfig={ mBounds=Rect(0, 0 - 1440, 720) mAppBounds=Rect(0, 0 - 1359, 720) mWindowingMode=fullscreen mDisplayWindowingMode=fullscreen mActivityType=undefined mAlwaysOnTop=undefined mRotation=ROTATION_90} s.209} unchanged for IME proc com.android.inputmethod.latin
04-30 05:13:00.345   225   225 I hwservicemanager: getTransport: Cannot find entry android.hardware.configstore@1.0::ISurfaceFlingerConfigs/default in either framework or device manifest.
04-30 05:13:00.345   472   492 V WindowManager: Orientation start waiting for draw, mDrawState=DRAW_PENDING in Window{ec98f56 u0 com.android.systemui.ImageWallpaper}, surfaceController Surface(name=com.android.systemui.ImageWallpaper)/@0xb83b595
04-30 05:13:00.347   330   330 E Layer   : [Surface(name=Task=1)/@0xa46827d - animation-leash#0] No local sync point found
04-30 05:13:00.347   330   330 E Layer   : [Surface(name=Task=1)/@0xa46827d - animation-leash#0] No local sync point found
04-30 05:13:00.347   330   330 E Layer   : [Surface(name=Task=50)/@0xb98182b - animation-leash#0] No local sync point found
04-30 05:13:00.347   330   330 E Layer   : [Surface(name=Task=50)/@0xb98182b - animation-leash#0] No local sync point found
04-30 05:13:00.221   320   320 I health@2.0-serv: type=1400 audit(0.0:238): avc: denied { getattr } for path="/sys/devices/platform/soc/1f03400.rsb/sunxi-rsb-3a3/axp20x-battery-power-supply/power_supply/axp20x-battery/present" dev="sysfs" ino=20361 scontext=u:r:hal_health_default:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
04-30 05:13:00.391   472   492 I InputManager-JNI: Viewport [0] to add: local:0, isActive: true
[... cut ...]
04-30 05:13:52.326   378   378 E libc    : The property "ro.build.fingerprint" has a value with length 92 that is too large for __system_property_get()/__system_property_read(); use __system_property_read_callback() instead.
04-30 05:13:52.327     0     0 W perfetto: disabled ftrace
04-30 05:13:52.330   377   377 I perfetto: ftrace_procfs.cc:183    disabled ftrace
04-30 05:13:52.334   378   378 I perfetto: ng_service_impl.cc:1948 Tracing session 17 ended, total sessions:0
04-30 05:13:54.727   472   472 W WindowManager: removeWindowToken: Attempted to remove non-existing token: android.os.Binder@bf95692

```

rsglobal, 2021-04-30:
Hi @Robert-L-Turner ,

Can you share simple steps how to reproduce your issue?

Something like:
1. Download https://xxx/application.apk
2. Install `adb install application.apk`
3. Run, got fail

Robert-L-Turner, 2021-04-30:
Of course, but its not much more complex than that:

1. Fresh install of glodroid to a Pinephone Mobian Community Edition
    -> Have installed to a 128GB SD card w/o issue
    -> Then also repeated these steps but with a successful eMMC install as some of the errors might have been storage related?
2. Downloaded multiple com.anki.cozmo apks: 3.4, 3.2, 2.3.1
3. Sideloaded (correct term?) the apk using `adb install com.anki.cozmo_xxxxx`
4. Launch the application directly on the phone
    -> Anki welcome screen pops up
    -> Android permissions dialog prompts to allow access to files.
    -> Dialog _'Developer warning for package 'com.anki.cozmo' Failed to post notification on channel null please review log for details'_
    -> Main window has a download progress bar halted at 0% _'Download was unable to proceed - please close & reopen the app.  If the problem persists, please reinstall.'_
    -> If I use an apk prior to 3.4, specifically 2.3.1, the error below the progress bar is _'The cozmo app could not read files: error 3'_
5. Restart app with `adb logcat` with results as attached
6. Others
    ->  Confirm internet access and DNS resolution
    ->  Review app access which seemingly confirms file access etc
    ->  As mentioned I moved to eMMC thinking maybe there was a distinction between external storage permissions and local file permissions.

rsglobal, 2021-04-30:
> Downloaded multiple com.anki.cozmo apks: 3.4, 3.2, 2.3.1

Could you provide a links to apks please?

Robert-L-Turner, 2021-04-30:
https://apk.tools/details-cozmo-apk/

If it helps I can throw them up on my github?

The version there and on the google play store are both 3.4.0.

rsglobal, 2021-04-30:
> If it helps I can throw them up on my github?

Please do

Robert-L-Turner, 2021-04-30:
Setup a private repo and invited you as a contributor.  Let me know if that works.

rsglobal, 2021-04-30:
I have the access. But I will be able to have a look into this only at the end of next week.

Robert-L-Turner, 2021-04-30:
No problem at all.  Really appreciate the quick engagement and the work to support the Pinephone community.

Robert-L-Turner, 2021-05-23:
Hi @rsglobal .  Not a chaser :) But I'm still interested in debugging this if you have time.

rsglobal, 2021-05-24:
@Robert-L-Turner ,

It looks like this app require Google Play service installed to download additional content.
Google Mobile Services isn't free therefore GloDroid does not use them.

closed 2021-11-11 by rsglobal