GloDroid/glodroid_manifest #138: audio jack switching not working on pinephone

squidink7, 2021-05-08:
the pinephone has had some issues with switching to the headphones automatically when plugged in, but using the latest 5.12 kernel on Manjaro this has been fixed. I noticed in the 0.6.1 builds this isn't working yet it's still running linux 5.12. Would you know why that may be?

rsglobal, 2021-05-08:
As far as I know plug/unplug events are successfully translated to Android framework. What is missing is configuration sets for routing audio in this file https://github.com/GloDroid/glodroid_device/blob/13504605aedb8bebcbbe9b95dba5c292686888db/pinephone/audio.pinephone.xml

rsglobal, 2021-05-08:
From Linux side it usually implemented by alsa-ucm-conf: Example for different hardware: https://github.com/Srinivas-Kandagatla/alsa-ucm-conf/commit/c84edaf5afaaea45d0347863d7f868b3a95de915

rsglobal, 2021-05-24:
#143 should address this issue.

madriemx, 2022-02-01:
Should this issue be reopened now, while #143 wasn't sufficient?

QushyQushy, 2022-06-26:
This issue should be reopened, it's the only thing not working apart from the camera.

IMSOASIAN, 2022-09-01:
I second this, auto jack switching is not working on my PinePhone either.

herbsmn, 2022-11-03:
I just tested this on v0.8.1 with the pinephone and it is still an issue. Please reopen this issue.

herbsmn, 2022-11-03:
Would updating the glodroid tinyhal fork potentially help? It doesn't seem to have been updated since September 2020: https://github.com/GloDroid/glodroid_forks/tree/tinyhal-next

It looks like it is regularly updated here: https://github.com/CirrusLogic/tinyhal

rsglobal, 2023-04-23:
Fixed in 2023w16 release: https://github.com/GloDroidCommunity/pine64-pinephone/releases/tag/2023w16

closed 2023-04-23 by rsglobal