GloDroid/glodroid_manifest #141: 'kenerl/glodroid-megous' not found when trying to build Pinetab imgs

tophneal, 2021-05-14:
I'm trying to build imgs for the PineTab, but shortly into building I get the error

`device/glodroid/platform/kernel/kernel.Mk:70: error: FindEmulator: find: 'kernel/glodroid-megous': No such file or directory`

`ls glodroid/kernel` shows: configs, firmware, glodroid-broadcom, glodroid-modules, glodroid-sunxi, prebuilts, tests

rsglobal, 2021-05-14:
Hi,

Please run:

```
repo init -u https://github.com/glodroid/glodroid_manifest -b next
repo sync -c
```

tophneal, 2021-05-14:
Thanks, that resolved that issue. Unfortunately, a new one arose, but no explicit error is given. I'll look into it more.

rsglobal, 2021-05-25:
Good. Please create another issue if it still relevant. Also please check new [README.md](https://github.com/GloDroid/glodroid_manifest#you-should-install-additional-packages-in-order-to-build-glodroid) page for new ubuntu packages dependencies. 
There were several force-pushes to the v0.6.1 TAG before release, so you might need repo init/repo sync from scratch again.

closed 2021-05-25 by rsglobal