GloDroid/glodroid_manifest #142: Why is green screen after power on for opi3

haidong-yao, 2021-05-24:
**1.  source build/envsetup.sh
2 . lunch opi3-userdebug
3. make images
4. sudo dd if=sdcard.img of=/dev/sdb bs=4096   

The display screen is a display with HDMI interface， Green screen after power on（Not power button） 。
adb shell is ok，One of the LED is red**

rsglobal, 2021-05-24:
Hi,

> 4. sudo dd if=sdcard.img of=/dev/sdb bs=4096

Please refer to the latest manual available [HERE](https://github.com/GloDroid/glodroid_manifest#step-1)

haidong-yao, 2021-05-24:
Hi rsglobal,

**Step 1**

sudo ln -s ~/glodroid/out/target/product/opi3/fastboot /usr/bin/fastboot
sudo ln -s ~/glodroid/out/target/product/opi3/adb      /usr/bin/adb
sudo ln -s ~/glodroid/out/target/product/opi3/mke2fs   /usr/sbin/mke2fs

**Step 2**

sudo dd if=deploy-sd.img of=/dev/sdb bs=4096

Insert sdcard to board and power up.
fastboot devices
82c00007a49e6ba2	fastboot

**Step 3**

Modify flash-sd.sh  remove "./"

#!/bin/sh -xe

fastboot flash gpt deploy-gpt.img
fastboot flash bootloader bootloader-sd.img
fastboot flash uboot-env  env.img
fastboot reboot

set +x
if [ "-sd" = "-emmc" ]; then
  echo "Power-down, remove recovery SD-CARD, and power-up the board."
  echo "Press enter to continue."
  read key
fi
set -x

fastboot oem format
fastboot flash bootloader      bootloader-sd.img
fastboot flash uboot-env       env.img
fastboot flash recovery_boot   boot.img
fastboot erase misc

fastboot reboot-fastboot

fastboot flash boot_a boot.img
fastboot flash dtbo_a boot_dtbo.img
fastboot flash super  super.img
fastboot format:ext4  metadata
fastboot format:ext4  userdata
fastboot reboot

**Step 4**

run flash-sd.sh
 
:~/glodroid/out/target/product/opi3$ sudo ./flash-sd.sh
+ fastboot flash gpt deploy-gpt.img
Sending 'gpt' (128 KB)                             OKAY [  0.018s]
Writing 'gpt'                                      OKAY [  0.043s]
Finished. Total time: 0.079s
+ fastboot flash bootloader bootloader-sd.img
Sending 'bootloader' (1920 KB)                     OKAY [  0.191s]
Writing 'bootloader'                               OKAY [  0.178s]
Finished. Total time: 0.380s
+ fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.030s]
Writing 'uboot-env'                                OKAY [  0.035s]
Finished. Total time: 0.081s
+ fastboot reboot
Rebooting                                          OKAY [  0.000s]
Finished. Total time: 0.553s
+ set +x
+ fastboot oem format
< waiting for any device >
                                                   OKAY [  0.023s]
Finished. Total time: 0.023s
+ fastboot flash bootloader bootloader-sd.img
Sending 'bootloader' (1920 KB)                     OKAY [  0.187s]
Writing 'bootloader'                               OKAY [  0.177s]
Finished. Total time: 0.377s
+ fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.029s]
Writing 'uboot-env'                                OKAY [  0.036s]
Finished. Total time: 0.080s
+ fastboot flash recovery_boot boot.img
Sending 'recovery_boot' (24576 KB)                 OKAY [  2.337s]
Writing 'recovery_boot'                            OKAY [  2.403s]
Finished. Total time: 4.755s
+ fastboot erase misc
Erasing 'misc'                                     OKAY [  0.022s]
Finished. Total time: 0.047s
+ fastboot reboot-fastboot
Rebooting into fastboot                            OKAY [  0.009s]
< waiting for any device >
Finished. Total time: 14.760s
+ fastboot flash boot_a boot.img
Sending 'boot_a' (32768 KB)                        OKAY [  3.214s]
Writing 'boot_a'                                   OKAY [  1.913s]
Finished. Total time: 5.189s
+ fastboot flash dtbo_a boot_dtbo.img
Sending 'dtbo_a' (32 KB)                           OKAY [  0.029s]
Writing 'dtbo_a'                                   OKAY [  0.088s]
Finished. Total time: 0.188s
+ fastboot flash super super.img
Invalid sparse file format at header magic
Sending sparse 'super' 1/5 (262124 KB)             OKAY [ 25.772s]
Writing 'super'                                    OKAY [ 14.594s]
Sending sparse 'super' 2/5 (262128 KB)             OKAY [ 25.587s]
Writing 'super'                                    OKAY [ 14.579s]
Sending sparse 'super' 3/5 (262140 KB)             OKAY [ 25.771s]
Writing 'super'                                    OKAY [ 14.694s]
Sending sparse 'super' 4/5 (249532 KB)             OKAY [ 24.518s]
Writing 'super'                                    OKAY [ 14.208s]
Sending sparse 'super' 5/5 (206484 KB)             OKAY [ 20.343s]
Writing 'super'                                    OKAY [ 21.732s]
Finished. Total time: 210.081s
+ fastboot format:ext4 metadata
Warning: metadata type is raw, but ext4 was requested for formatting.
mke2fs 1.45.4 (23-Sep-2019)
Creating filesystem with 4096 4k blocks and 4096 inodes

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (1024 blocks): done
Writing superblocks and filesystem accounting information: done

Sending 'metadata' (52 KB)                         OKAY [  0.011s]
Writing 'metadata'                                 OKAY [  0.085s]
Finished. Total time: 0.207s
+ fastboot format:ext4 userdata
Warning: userdata type is raw, but ext4 was requested for formatting.
mke2fs 1.45.4 (23-Sep-2019)
Creating filesystem with 3475451 4k blocks and 869696 inodes
Filesystem UUID: a08293c5-b1f1-40a4-945b-215e67c47a51
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (16384 blocks): done
Writing superblocks and filesystem accounting information: done   

Sending 'userdata' (4288 KB)                       OKAY [  0.419s]
Writing 'userdata'                                 OKAY [  0.336s]
Finished. Total time: 0.809s
+ fastboot reboot
Rebooting                                          OKAY [  0.000s]
Finished. Total time: 0.051s

**It's still a green screen!  where it went wrong? What's this deploy-gpt.img?**

rsglobal, 2021-05-24:
Hm. It shouldn't be so.  

Does `adb shell` work for you? If yes, please attach output of:

```
adb shell dmesg 2>&1 | tee dmesg.log
adb shell logcat 2>&1 | tee logcat.log
```

Please check your monitor is working. (by installing armbian with the same hardware).

rsglobal, 2021-05-24:
> What's this deploy-gpt.img?

It's just initial partition table. (First several sectors of deploy-XX.img)

haidong-yao, 2021-05-24:
monitor is ok， because emmc is image of orangePi ，It can show luncher，so hardware is ok。

[dmesg.log](https://github.com/GloDroid/glodroid_manifest/files/6532103/dmesg.log)
[logcat.log](https://github.com/GloDroid/glodroid_manifest/files/6532106/logcat.log)

rsglobal, 2021-05-24:
Installing Android on sd, while Android is present on emmc can cause some issues, but it looks like this is not the case:

```
01-01 00:00:24.298   325   325 E HWComposer: loadLayerMetadataSupport: getLayerGenericMetadataKeys failed: UNSUPPORTED (8)
01-01 00:00:24.299   325   325 I HWComposer: Switching to legacy multi-display mode
01-01 00:00:24.299   325   325 E HWComposer: isConnected failed for display 0: Invalid display
01-01 00:00:24.329   305   305 I ServiceManager: Waiting for service 'statscompanion' on '/dev/binder'...
01-01 00:00:24.329   377   377 I ServiceManager: Waiting for service 'package_native' on '/dev/binder'...
01-01 00:00:24.346   389   389 I ServiceManager: Waiting for service 'package_native' on '/dev/binder'...
01-01 00:00:24.413   325   325 E HWC2    : getDisplayAttribute(0, 1, Unknown) failed: BadConfig (1)
01-01 00:00:24.414   325   325 E HWC2    : getDisplayAttribute(0, 3, Unknown) failed: BadConfig (1)
01-01 00:00:24.414   325   325 E HWC2    : getDisplayAttribute(0, 7, Unknown) failed: BadConfig (1)
01-01 00:00:24.415   325   325 E HWC2    : getDisplayAttribute(0, 9, Unknown) failed: BadConfig (1)
01-01 00:00:24.415   325   325 E HWC2    : getDisplayAttribute(0, 10, Unknown) failed: BadConfig (1)
01-01 00:00:24.416   325   325 E HWC2    : getDisplayAttribute(0, 11, Unknown) failed: BadConfig (1)
01-01 00:00:24.416   325   325 E HWC2    : getDisplayAttribute(0, 12, Unknown) failed: BadConfig (1)
01-01 00:00:24.417   325   325 E HWC2    : getDisplayAttribute(0, 13, Unknown) failed: BadConfig (1)
01-01 00:00:24.417   325   325 E HWC2    : getDisplayAttribute(0, 14, Unknown) failed: BadConfig (1)
01-01 00:00:24.418   325   325 E HWC2    : getDisplayAttribute(0, 15, Unknown) failed: BadConfig (1)
01-01 00:00:24.419   325   325 E HWC2    : getDisplayAttribute(0, 16, Unknown) failed: BadConfig (1)
01-01 00:00:24.419   325   325 E HWC2    : getDisplayAttribute(0, 17, Unknown) failed: BadConfig (1)
01-01 00:00:24.420   325   325 E HWC2    : getDisplayAttribute(0, 18, Unknown) failed: BadConfig (1)
01-01 00:00:24.420   325   325 E HWC2    : getDisplayAttribute(0, 19, Unknown) failed: BadConfig (1)
01-01 00:00:24.421   325   325 E HWC2    : getDisplayAttribute(0, 20, Unknown) failed: BadConfig (1)
01-01 00:00:24.421   325   325 E HWC2    : getDisplayAttribute(0, 21, Unknown) failed: BadConfig (1)
01-01 00:00:24.422   325   325 E HWC2    : getDisplayAttribute(0, 22, Unknown) failed: BadConfig (1)
01-01 00:00:24.422   325   325 E HWC2    : getDisplayAttribute(0, 23, Unknown) failed: BadConfig (1)
01-01 00:00:24.423   325   325 E HWC2    : getDisplayAttribute(0, 24, Unknown) failed: BadConfig (1)
01-01 00:00:24.423   325   325 E HWC2    : getDisplayAttribute(0, 26, Unknown) failed: BadConfig (1)
01-01 00:00:24.424   325   325 E HWC2    : getDisplayAttribute(0, 27, Unknown) failed: BadConfig (1)
01-01 00:00:24.424   325   325 E HWC2    : getDisplayAttribute(0, 28, Unknown) failed: BadConfig (1)
01-01 00:00:24.425   325   325 E HWC2    : getDisplayAttribute(0, 30, Unknown) failed: BadConfig (1)
01-01 00:00:24.425   325   325 E HWC2    : getDisplayAttribute(0, 31, Unknown) failed: BadConfig (1)
01-01 00:00:24.426   325   325 E HWC2    : getDisplayAttribute(0, 32, Unknown) failed: BadConfig (1)
01-01 00:00:24.426   325   325 E HWC2    : getDisplayAttribute(0, 33, Unknown) failed: BadConfig (1)
01-01 00:00:24.427   325   325 E HWC2    : getDisplayAttribute(0, 35, Unknown) failed: BadConfig (1)
01-01 00:00:24.427   325   325 E HWC2    : getDisplayAttribute(0, 37, Unknown) failed: BadConfig (1)
01-01 00:00:24.428   325   325 E HWC2    : getDisplayAttribute(0, 38, Unknown) failed: BadConfig (1)
01-01 00:00:24.428   325   325 E HWC2    : getDisplayAttribute(0, 39, Unknown) failed: BadConfig (1)
01-01 00:00:24.429   325   325 E HWC2    : getDisplayAttribute(0, 40, Unknown) failed: BadConfig (1)
01-01 00:00:24.429   325   325 E HWC2    : getDisplayAttribute(0, 41, Unknown) failed: BadConfig (1)
01-01 00:00:24.430   325   325 E HWC2    : getDisplayAttribute(0, 42, Unknown) failed: BadConfig (1)
01-01 00:00:24.430   305   305 I ServiceManager: Waiting for service 'statscompanion' on '/dev/binder'...
01-01 00:00:24.430   377   377 I ServiceManager: Waiting for service 'package_native' on '/dev/binder'...
01-01 00:00:24.430   325   325 E HWC2    : getDisplayAttribute(0, 44, Unknown) failed: BadConfig (1)
```

I can see this on your logcat output.

Please try with different display.

rsglobal, 2021-05-24:
Please also check with v0.5.1 and v0.6.1 release images.

haidong-yao, 2021-05-24:
Maybe the monitor doesn't support it, What kind of monitor do you use to test？or LCD？
thanks
[image]
[image]

haidong-yao, 2021-05-25:
I changed to a high-resolution monitor and can enter the glodroid luncher ，the testing is very good！

Thanks  rsglobal

closed 2021-05-25 by haidong-yao

rsglobal, 2021-05-25:
Can you specify your monitor type to help us handle such kind of issues?

haidong-yao, 2021-05-25:
The first monitor is ViewSonic VA2462H.
 The second monitor is Samsung  S27E360HSIXF.

haidong-yao, 2021-05-25:
**Which device of hardware supports mail-G52?**

haidong-yao, 2021-05-25:
Which device of hardware supports mali-G52?

rsglobal, 2021-05-25:
> Which device of hardware supports mali-G52?

Could you rephrase your question please

rsglobal, 2021-05-25:
Which SOC / board? I don't know. Try googling.

haidong-yao, 2021-05-26:
oh, Thanks rsglobal

The  opi3 is mali-t720, I want to add log, Learning GPU，How to compile kernel?  

I don't know where to define ”make images“，it's all compiled。

for example：
1.  source build/envsetup.sh
2. lunch opi3-userdebug
**3.  make kerne ?**

Write boot.img through fastboot
**fastboot flash boot_a boot.img？**

rsglobal, 2021-05-26:
3. 
`make out/target/product/opi3/boot.img`
or just
`m out/target/product/opi3/boot.img`

`m` is working from any directory inside the tree.

haidong-yao, 2021-05-26:
Write boot.img failed！ 

:~/glodroid/out/target/product/opi3$ **fastboot flash boot_a boot.img**
**fastboot: error: boot partition is smaller than boot image**

24M	boot.img

rsglobal, 2021-05-26:
24M should fit. Anyway try to change partition layout: 
https://github.com/GloDroid/glodroid_device/blob/618cd78a15f4d320ea48ebc0d18128aa76f1b754/platform/uboot/bootscript.cpp#L43

haidong-yao, 2021-05-28:
Must execute fastboot reboot fastboot