GloDroid/glodroid_manifest #143: pinephone: Align ALSA UCM configuration with PostMarketOS

rsglobal, 2021-05-24:
Benefits:
 - Get free & tested updates from linux community,
 - Ability to report issue / find the solution together.

GloDroid configuration:
https://github.com/GloDroid/glodroid_device/blob/master/pinephone/audio.pinephone.xml

PostMarketOS configuration:
https://gitlab.com/postmarketOS/pmaports/-/tree/master/device/main/device-pine64-pinephone/ucm

squidink7, 2021-12-15:
I noticed recent changes to the audio.pinephone.xml file, is this issue at all related to that or no? Also is that the only file that needs adjustment to solve this? and if so could information be gathered from the example tinyhal implementation (https://github.com/CirrusLogic/tinyhal/blob/master/audio.example.xml) which seems to already have a device section for both loudspeaker and headphone port? I understand that it cannot be used without modification, just wondering about how this all works. I apologise for my lack of knowledge on this topic although am eager to learn!

rsglobal, 2021-12-16:
In the attempt to complete this issue it was discovered that tinyhal lacks a lot of required functionality.
Therefore it can be used to provide basic features only. We have to search for alternative solution or implement our own audio HAL.

closed 2021-12-16 by rsglobal

MarijnS95, 2021-12-16:
For someone planning to implement an alternative/new solution, consider purely using UCM as configuration format. It is after all already in use to feed PipeWire and PulseAudio for that exact same reason, and should allow a single config to be written and reused across every platform. 

@rsglobal and me talked about this a long while ago and I recall creating an issue for it on the CirrusHAL repo - but it doesn't seem to be there. Roman, if you remember this might be the right spot to link it just in case.