GloDroid/glodroid_manifest #148: making error

cyber-bat, 2021-06-20:
Please help. What is error?
[  0% 3/6312] build out/target/product/opi3/obj_arm/MESON_MESA3D/.build.timestamp
FAILED: out/target/product/opi3/obj_arm/MESON_MESA3D/.build.timestamp
/bin/bash -c "(rm -rf out/target/product/opi3/obj_arm/MESON_MESA3D/ ) && (mkdir -p out/target/product/opi3/obj_arm/MESON_MESA3D/ ) && (mkdir -p out/target/product/opi3/obj_arm/MESON_MESA3D//build ) && (cp -r external/mesa3d//* out/target/product/opi3/obj_arm/MESON_MESA3D/ ) && (cd out/target/product/opi3/obj_arm/MESON_MESA3D && PATH=/usr/bin:/usr/local/bin:\$PATH meson ./build --cross-file /home/outer/GloDroid/out/target/product/opi3/obj_arm/MESON_MESA3D_GEN/aosp_cross --buildtype=release -Ddri-search-path=/vendor/lib/dri -Dplatforms=android -Dplatform-sdk-version=30 -Ddri-drivers= -Dgallium-drivers=panfrost,kmsro -Dvulkan-drivers= -Dgbm=enabled -Degl=enabled -Dfreedreno-kgsl=true -Dandroid-use-mapper4-metadata=true ) && (PATH=/usr/bin:/bin:/sbin:\$PATH ninja -C out/target/product/opi3/obj_arm/MESON_MESA3D/build ) && (touch out/target/product/opi3/obj_arm/MESON_MESA3D/.build.timestamp )"
The Meson build system
Version: 0.45.1
Source dir: /home/outer/GloDroid/out/target/product/opi3/obj_arm/MESON_MESA3D
Build dir: /home/outer/GloDroid/out/target/product/opi3/obj_arm/MESON_MESA3D/build
Build type: cross build
Program python found: YES (/usr/bin/python)

meson_options.txt:50:0: ERROR: Options "" are not in allowed choices: "auto, i915, i965, r100, r200, nouveau"

A full log can be found at /home/outer/GloDroid/out/target/product/opi3/obj_arm/MESON_MESA3D/build/meson-logs/meson-log.txt
[  0% 4/6312] build out/target/product/opi3/obj/MESON_MESA3D/.build.timestamp
FAILED: out/target/product/opi3/obj/MESON_MESA3D/.build.timestamp
/bin/bash -c "(rm -rf out/target/product/opi3/obj/MESON_MESA3D/ ) && (mkdir -p out/target/product/opi3/obj/MESON_MESA3D/ ) && (mkdir -p out/target/product/opi3/obj/MESON_MESA3D//build ) && (cp -r external/mesa3d//* out/target/product/opi3/obj/MESON_MESA3D/ ) && (cd out/target/product/opi3/obj/MESON_MESA3D && PATH=/usr/bin:/usr/local/bin:\$PATH meson ./build --cross-file /home/outer/GloDroid/out/target/product/opi3/obj/MESON_MESA3D_GEN/aosp_cross --buildtype=release -Ddri-search-path=/vendor/lib64/dri -Dplatforms=android -Dplatform-sdk-version=30 -Ddri-drivers= -Dgallium-drivers=panfrost,kmsro -Dvulkan-drivers= -Dgbm=enabled -Degl=enabled -Dfreedreno-kgsl=true -Dandroid-use-mapper4-metadata=true ) && (PATH=/usr/bin:/bin:/sbin:\$PATH ninja -C out/target/product/opi3/obj/MESON_MESA3D/build ) && (touch out/target/product/opi3/obj/MESON_MESA3D/.build.timestamp )"
The Meson build system
Version: 0.45.1
Source dir: /home/outer/GloDroid/out/target/product/opi3/obj/MESON_MESA3D
Build dir: /home/outer/GloDroid/out/target/product/opi3/obj/MESON_MESA3D/build
Build type: cross build
Program python found: YES (/usr/bin/python)

meson_options.txt:50:0: ERROR: Options "" are not in allowed choices: "auto, i915, i965, r100, r200, nouveau"

A full log can be found at /home/outer/GloDroid/out/target/product/opi3/obj/MESON_MESA3D/build/meson-logs/meson-log.txt
15:41:48 ninja failed with: exit status 1

rsglobal, 2021-06-22:
Hi,

```
The Meson build system
Version: 0.45.1
```

Looks like you are using Ubuntu < 20.04.
Please use docker to build under your OS.
You can find manual here: https://github.com/GloDroid/glodroid_manifest/pull/136#issue-620893052

nas00, 2021-06-22:
The below option should work too:

sudo pip3 install meson

closed 2021-06-25 by cyber-bat