GloDroid/glodroid_manifest #149: making error Glodroid

cyber-bat, 2021-06-24:
Please help me to fix it.
test github.com/google/blueprint/pathtools
FAILED: out/soong/.bootstrap/blueprint-pathtools/test/test.passed
out/soong/.bootstrap/bin/gotestrunner -p ./build/blueprint/pathtools -f out/soong/.bootstrap/blueprint-pathtools/test/test.passed -- out/soong/.bootstrap/blueprint-pathtools/test/test -test.short
--- FAIL: TestGlobEscapes (0.00s)
    --- FAIL: TestGlobEscapes/**/* (0.00s)
        glob_test.go:571: incorrect matches list:
        glob_test.go:571:  pattern: "**/*"
        glob_test.go:571:      got: []string{"**/", "a/", "b", "**/a", "**/b/", "**/b/b", "a/a"}
        glob_test.go:571: expected: []string{"*", "**/", "?", "a/", "b", "**/*", "**/a", "**/b/", "**/b/b", "a/a"}
    --- FAIL: TestGlobEscapes/**/\* (0.00s)
        glob_test.go:571: incorrect matches list:
        glob_test.go:571:  pattern: "**/\\*"
        glob_test.go:571:      got: []string(nil)
        glob_test.go:571: expected: []string{"*", "**/*"}
    --- FAIL: TestGlobEscapes/\*\*/* (0.00s)
        glob_test.go:571: incorrect matches list:
        glob_test.go:571:  pattern: "\\*\\*/*"
        glob_test.go:571:      got: []string{"**/a", "**/b/"}
        glob_test.go:571: expected: []string{"**/*", "**/a", "**/b/"}
    --- FAIL: TestGlobEscapes/\*\*/**/* (0.00s)
        glob_test.go:571: incorrect matches list:
        glob_test.go:571:  pattern: "\\*\\*/**/*"
        glob_test.go:571:      got: []string{"**/a", "**/b/", "**/b/b"}
        glob_test.go:571: expected: []string{"**/*", "**/a", "**/b/", "**/b/b"}
FAIL

nas00, 2021-06-25:
Try the repo sync command again according to [this](https://groups.google.com/g/android-building/c/iwBWkldqqIU). 

If you're building on an ancient pc, it's fun getting over the obstacles but not really worth the effort/time.

cyber-bat, 2021-06-25:
Used commands repo -c --sync-force --sync-remove-dirty and repo -c --fast-fail don't help. Make for opi3-userdebug

nas00, 2021-06-26:
Then my guess is that your golang-go package is outdated - probably as a result of running an older version of ubuntu.

> go version

cyber-bat, 2021-06-26:
<repo not installed>
repo launcher version 2.15
       (from /home/outer/bin/repo)
using repo from GloDroid

cyber-bat, 2021-07-23:
go version go1.13.8 linux/amd64

nas00, 2021-07-23:
I think you will need to upgrade to ubuntu 20.04 or the try the docker option as the simplest solution.

cyber-bat, 2021-07-23:
Yes i make on ubuntu 20.04

cyber-bat, 2021-07-23:
Used any variants of options repo sync. Think what problem in goland.

nas00, 2021-07-23:
You could try skippng the test as suggested [here](https://blog.csdn.net/whatareyou123/article/details/109328410).

Comment out glob_test.go in /GloDroid/build/blueprint/Blueprints and try the build again.

closed 2021-08-20 by cyber-bat