GloDroid/glodroid_manifest #150: Glodroid will plan to update os to android12?

haidong-yao, 2021-07-05:


rsglobal, 2021-07-05:
yes

haidong-yao, 2021-07-06:
> yes

which branches are for android12 ? Or just a plan at present?
Can I get the time schedule，
I have strong interest in run the android12 on opi3,
Can I join in this work？

rsglobal, 2021-07-06:
Android 12 hasn't been released yet. Usually, a new Android is released in early September.
Usually it takes 1-3 weeks to migrate GloDroid to a new version of AOSP.

Android-beta from Google usually doesn't works out of the box due to some conflicts between projects. In case you're able to build beta for emulator, you can easily run GloDroid on it.
Just a few changes required. I can guide you through the process if you need.

haidong-yao, 2021-07-07:
[glodroid.zip](https://github.com/GloDroid/glodroid_manifest/files/6774018/glodroid.zip)
Attach files is details.

I used a preview version of Android 12，and  gbm_gralloc，drm_hwcomposer，mesa3d of GloDroid。 

log error：
============================================
[  0% 1/117005] Gen Header: libfreedreno_ir3_32 <= ir3_lexer.c
FAILED: out/target/product/ums512/gen/STATIC_LIBRARIES/libfreedreno_ir3_intermediates/ir3/ir3_lexer.c
/bin/bash -c "M4=out/soong/.intermediates/prebuilts/build-tools/m4/linux_glibc_x86_64/m4 out/soong/.intermediates/prebuilts/build-tools/flex/linux_glibc_x86_64/flex -o out/target/product/ums512/gen/STATIC_LIBRARIES/libfreedreno_ir3_intermediates/ir3/ir3_lexer.c external/mesa3d/src/freedreno/ir3/ir3_lexer.l"
/bin/bash: out/soong/.intermediates/prebuilts/build-tools/flex/linux_glibc_x86_64/flex: No such file or directory
[  0% 2/117005] Gen Header: libfreedreno_ir3_32 <= ir3_parser.c
FAILED: out/target/product/ums512/gen/STATIC_LIBRARIES/libfreedreno_ir3_intermediates/ir3/ir3_parser.c
/bin/bash -c "out/soong/.intermediates/prebuilts/build-tools/bison/linux_glibc_x86_64/bison external/mesa3d/src/freedreno/ir3/ir3_parser.y --name-prefix=ir3_yy --output=out/target/product/ums512/gen/STATIC_LIBRARIES/libfreedreno_ir3_intermediates/ir3/ir3_parser.c"
/bin/bash: out/soong/.intermediates/prebuilts/build-tools/bison/linux_glibc_x86_64/bison: No such file or directory
[  0% 3/117005] Gen Header: libfreedreno_ir3_32 <= ir3_parser.h
FAILED: out/target/product/ums512/gen/STATIC_LIBRARIES/libfreedreno_ir3_intermediates/ir3/ir3_parser.h
/bin/bash -c "out/soong/.intermediates/prebuilts/build-tools/bison/linux_glibc_x86_64/bison external/mesa3d/src/freedreno/ir3/ir3_parser.y --name-prefix=ir3_yy --defines=out/target/product/ums512/gen/STATIC_LIBRARIES/libfreedreno_ir3_intermediates/ir3/ir3_parser.h"
/bin/bash: out/soong/.intermediates/prebuilts/build-tools/bison/linux_glibc_x86_64/bison: No such file or directory
[  0% 4/117005] Mesa Lex: libmesa_glsl_32 <= external/mesa3d/src/compiler/glsl/glcpp/glcpp-lex.l
FAILED: out/target/product/ums512/gen/STATIC_LIBRARIES/libmesa_glsl_intermediates/glsl/glcpp/glcpp-lex.c
....
....
....

rsglobal, 2021-07-07:
Could you make pull-request with your changes?

Also I recommend not to use any kind preview or betas. If it's fine for you to use just master branch-lets use it, since it 100% in working shape.

haidong-yao, 2021-08-16:
****oh.
now , I was faced with a new problem, It is failed to compilte mesa3d. however, "make images " can compile successfully.**
base on glodroid_manifest-0.5.1.**

:~/Glodroid/external/mesa3d$ mm
//============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=11
TARGET_PRODUCT=opi3
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a53
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a15
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-4.15.0-151-generic-x86_64-Ubuntu-18.04-LTS
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=RP1A.200720.011
OUT_DIR=out
//============================================
[  0% 1/2875] target  C: libfreedreno_common <= external/mesa3d/src/freedreno/common/freedreno_uuid.c
FAILED: out/target/product/opi3/obj/STATIC_LIBRARIES/libfreedreno_common_intermediates/common/freedreno_uuid.o
/bin/bash -c "PWD=/proc/self/cwd  prebuilts/clang/host/linux-x86/clang-r383902b/bin/clang -I external/mesa3d/src/gallium/include -I external/mesa3d/src/gallium/auxiliary -I external/mesa3d/src -I external/mesa3d/include -I external/mesa3d/src/freedreno -I out/target/product/opi3/obj/STATIC_LIBRARIES/libfreedreno_common_intermediates -I out/target/product/opi3/gen/STATIC_LIBRARIES/libfreedreno_common_intermediates  -Iexternal/libdrm/include/drm -Iexternal/libdrm/android -Iexternal/libdrm/include/drm -Iexternal/libdrm/android -isystem external/libdrm -isystem external/libdrm -Iexternal/libcxx/include -Iexternal/libcxxabi/include -isystem out/soong/.intermediates/bionic/libc/libc.llndk/android_vendor.30_arm64_armv8-a_cortex-a53_shared/gen/include -isystem bionic/libc/kernel/android/scsi -isystem bionic/libc/kernel/android/uapi -isystem bionic/libc/kernel/uapi -isystem bionic/libc/kernel/uapi/asm-arm64   -Iexternal/libcxxabi/include   -isystem out/target/product/opi3/obj/include -c  -Werror=implicit-function-declaration -DANDROID -fmessage-length=0 -W -Wall -Wno-unused -Winit-self -Wpointer-arith -no-canonical-prefixes -DNDEBUG -UDEBUG -fno-exceptions -Wno-multichar -O2 -g -fno-strict-aliasing -Werror=date-time -Werror=pragma-pack -Werror=pragma-pack-suspicious-include -fdebug-prefix-map=/proc/self/cwd= -D__compiler_offsetof=__builtin_offsetof -faddrsig -Werror=int-conversion -fexperimental-new-pass-manager -Wno-reserved-id-macro -Wno-unused-command-line-argument -fcolor-diagnostics -Wno-sign-compare -Wno-defaulted-function-deleted -Wno-inconsistent-missing-override -Wno-c99-designator -ftrivial-auto-var-init=zero -enable-trivial-auto-var-init-zero-knowing-it-will-be-removed-from-clang -ffunction-sections -fdata-sections -fno-short-enums -funwind-tables -fstack-protector-strong -Wa,--noexecstack -D_FORTIFY_SOURCE=2 -Wstrict-aliasing=2 -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Werror=format-security -nostdlibinc -march=armv8-a -mcpu=cortex-a53 -Bprebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/aarch64-linux-android/bin   -std=gnu99    -Wno-error -Werror=incompatible-pointer-types -Wno-unused-parameter -Wno-pointer-arith -Wno-missing-field-initializers -Wno-initializer-overrides -Wno-mismatched-tags -DPACKAGE_VERSION=\\\"20.3.0-devel\\\" -DPACKAGE_BUGREPORT=\\\"https://gitlab.freedesktop.org/mesa/mesa/-/issues\\\" -DANDROID_API_LEVEL=30 -DENABLE_SHADER_CACHE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DHAVE___BUILTIN_EXPECT -DHAVE___BUILTIN_FFS -DHAVE___BUILTIN_FFSLL -DHAVE_DLFCN_H -DHAVE_FUNC_ATTRIBUTE_FLATTEN -DHAVE_FUNC_ATTRIBUTE_UNUSED -DHAVE_FUNC_ATTRIBUTE_FORMAT -DHAVE_FUNC_ATTRIBUTE_PACKED -DHAVE_FUNC_ATTRIBUTE_ALIAS -DHAVE_FUNC_ATTRIBUTE_NORETURN -DHAVE_FUNC_ATTRIBUTE_RETURNS_NONNULL -DHAVE_FUNC_ATTRIBUTE_WARN_UNUSED_RESULT -DHAVE___BUILTIN_CTZ -DHAVE___BUILTIN_POPCOUNT -DHAVE___BUILTIN_POPCOUNTLL -DHAVE___BUILTIN_CLZ -DHAVE___BUILTIN_CLZLL -DHAVE___BUILTIN_UNREACHABLE -DHAVE_PTHREAD=1 -DHAVE_DLADDR -DHAVE_DL_ITERATE_PHDR -DHAVE_LINUX_FUTEX_H -DHAVE_ENDIAN_H -DHAVE_ZLIB -DMAJOR_IN_SYSMACROS -DVK_USE_PLATFORM_ANDROID_KHR -fvisibility=hidden -fno-math-errno -fno-trapping-math -Wno-sign-compare -DHAVE_TIMESPEC_GET -DHAVE_SYS_SHM_H -DHAVE_LIBDRM -D__ANDROID_VNDK__ -fPIC -DUSE_AARCH64_ASM -DDEFAULT_DRIVER_DIR=\\\"/vendor/lib64/dri\\\" -Wno-enum-compare -Wno-enum-compare-switch -Wno-null-pointer-arithmetic -Wno-null-dereference -Wno-pointer-compare -Wno-xor-used-as-pow -Wno-final-dtor-non-final-class -target aarch64-linux-android30 -std=c99  -Werror=int-to-pointer-cast -Werror=pointer-to-int-cast -Werror=fortify-source -Werror=address-of-temporary -Werror=return-type -Wno-tautological-constant-compare -Wno-tautological-type-limit-compare -Wno-reorder-init-list -Wno-implicit-int-float-conversion -Wno-int-in-bool-context -Wno-sizeof-array-div -Wno-tautological-overlap-compare -Wno-deprecated-copy -Wno-range-loop-construct -Wno-misleading-indentation -Wno-zero-as-null-pointer-constant -Wno-deprecated-anon-enum-enum-conversion -Wno-deprecated-enum-enum-conversion -Wno-string-compare -Wno-enum-enum-conversion -Wno-enum-float-conversion -Wno-pessimizing-move -MD -MF out/target/product/opi3/obj/STATIC_LIBRARIES/libfreedreno_common_intermediates/common/freedreno_uuid.d -o out/target/product/opi3/obj/STATIC_LIBRARIES/libfreedreno_common_intermediates/common/freedreno_uuid.o external/mesa3d/src/freedreno/common/freedreno_uuid.c"
external/mesa3d/src/freedreno/common/freedreno_uuid.c:30:10: fatal error: 'git_sha1.h' file not found
#include "git_sha1.h"
         ^~~~~~~~~~~~
1 error generated.
18:48:25 ninja failed with: exit status 1

#### failed to build some targets (5 seconds) ####

rsglobal, 2021-08-20:
Please use the `next` branch:
https://github.com/GloDroid/glodroid_manifest/tree/next

 It has latest updates and able to run master branch without visible issues for now (at least on PinePhone).

rsglobal, 2021-08-20:
Images for PinePhone is here: https://mega.nz/file/5HxBTQBB#niV4lODN_iUGFtZGkJXdoul1CN9pMAjmMJ-8eRXInDY