GloDroid/glodroid_manifest #152: Hardware acceleration support

joek85, 2021-07-25:
Hello, 
I am testing the latest release on Opi Plus2e sdcard version.
i played a 1080p video using the default NuPlayer, it was very laggy and skips lots of frames, sometimes it freezes.
on Skytube and NewPipe videos will play smoothly when reducing the resolution to 360p and below, hovering the mouse over the video will show some random flickering.
it seems it is still not hardware accelerated?
could you please point me to the right direction where to solve the problem?
i would like to help.
here are some logs:
``
2021-07-26 00:12:17.424 363-14527/? V/C2Store: in init
2021-07-26 00:12:17.424 363-14527/? V/C2Store: loading dll
2021-07-26 00:12:17.436 14633-16413/? I/CCodec: Created component [c2.android.avc.decoder]
2021-07-26 00:12:17.437 14633-16413/? D/CCodecConfig: read media type: video/avc
2021-07-26 00:12:17.448 14633-16413/? D/ReflectedParamUpdater: extent() != 1 for single value type: algo.buffers.max-count.values
2021-07-26 00:12:17.449 14633-16413/? D/ReflectedParamUpdater: extent() != 1 for single value type: output.subscribed-indices.values
2021-07-26 00:12:17.449 14633-16413/? D/ReflectedParamUpdater: extent() != 1 for single value type: input.buffers.allocator-ids.values
2021-07-26 00:12:17.450 14633-16413/? D/ReflectedParamUpdater: extent() != 1 for single value type: output.buffers.allocator-ids.values
2021-07-26 00:12:17.450 14633-16413/? D/ReflectedParamUpdater: extent() != 1 for single value type: algo.buffers.allocator-ids.values
2021-07-26 00:12:17.451 14633-16413/? D/ReflectedParamUpdater: extent() != 1 for single value type: output.buffers.pool-ids.values
2021-07-26 00:12:17.451 14633-16413/? D/ReflectedParamUpdater: extent() != 1 for single value type: algo.buffers.pool-ids.values
2021-07-26 00:12:17.462 14633-16413/? D/ReflectedParamUpdater: ignored struct field coded.color-format.locations
2021-07-26 00:12:17.467 14633-16413/? D/CCodecConfig: ignoring local param raw.size (0xd2001800) as it is already supported
2021-07-26 00:12:17.467 14633-16413/? D/CCodecConfig: ignoring local param raw.color (0xd2001809) as it is already supported
2021-07-26 00:12:17.468 14633-16413/? D/ReflectedParamUpdater: ignored struct field raw.hdr-static-info.mastering
2021-07-26 00:12:17.472 14633-16413/? I/CCodecConfig: query failed after returning 12 values (BAD_INDEX)
2021-07-26 00:12:17.473 14633-16413/? D/CCodecConfig: c2 config diff is Dict {
      c2::u32 coded.pl.level = 20496
      c2::u32 coded.pl.profile = 20481
      c2::u32 coded.vui.color.matrix = 0
      c2::u32 coded.vui.color.primaries = 0
      c2::u32 coded.vui.color.range = 2
      c2::u32 coded.vui.color.transfer = 0
      c2::u32 default.color.matrix = 0
      c2::u32 default.color.primaries = 0
      c2::u32 default.color.range = 0
      c2::u32 default.color.transfer = 0
      c2::u32 input.buffers.max-size.value = 2097152
      c2::u32 input.delay.value = 0
      string input.media-type.value = "video/avc"
      c2::u32 output.delay.value = 8
      string output.media-type.value = "video/raw"
      c2::u32 raw.color.matrix = 0
      c2::u32 raw.color.primaries = 0
      c2::u32 raw.color.range = 2
      c2::u32 raw.color.transfer = 0
      c2::u32 raw.max-size.height = 240
      c2::u32 raw.max-size.width = 320
      c2::u32 raw.pixel-format.value = 35
      c2::i32 raw.rotation.flip = 0
      c2::i32 raw.rotation.value = 0
      c2::u32 raw.sar.height = 1
      c2::u32 raw.sar.width = 1
      c2::u32 raw.size.height = 240
      c2::u32 raw.size.width = 320
      c2:
2021-07-26 00:12:17.475 14633-16413/? W/ColorUtils: expected specified color aspects (2:0:0:0)
2021-07-26 00:12:17.478 14633-16411/? D/SurfaceUtils: disconnecting from surface 0xac341948, reason onConfigure
2021-07-26 00:12:17.479 14633-16412/? D/SurfaceUtils: connecting to surface 0xac341948, reason connectToSurface
2021-07-26 00:12:17.480 14633-16412/? I/MediaCodec: [c2.android.avc.decoder] setting surface generation to 14984194
2021-07-26 00:12:17.480 14633-16412/? D/SurfaceUtils: disconnecting from surface 0xac341948, reason connectToSurface(reconnect)
2021-07-26 00:12:17.481 14633-16412/? D/SurfaceUtils: connecting to surface 0xac341948, reason connectToSurface(reconnect)
2021-07-26 00:12:17.482 14633-16413/? D/CCodec: [c2.android.avc.decoder] buffers are bound to CCodec for this session
2021-07-26 00:12:17.482 14633-16413/? D/CCodecConfig: no c2 equivalents for durationUs
2021-07-26 00:12:17.482 14633-16413/? D/CCodecConfig: no c2 equivalents for frame-count
2021-07-26 00:12:17.482 14633-16413/? D/CCodecConfig: no c2 equivalents for track-id
2021-07-26 00:12:17.482 14633-16413/? D/CCodecConfig: no c2 equivalents for language
2021-07-26 00:12:17.482 14633-16413/? D/CCodecConfig: no c2 equivalents for display-width
2021-07-26 00:12:17.482 14633-16413/? D/CCodecConfig: no c2 equivalents for display-height
2021-07-26 00:12:17.482 14633-16413/? D/CCodecConfig: no c2 equivalents for color-standard
2021-07-26 00:12:17.483 14633-16413/? D/CCodecConfig: no c2 equivalents for csd-1
[... 187 lines cut ...]
2021-07-26 00:12:23.723 14708-14775/system_process W/InputDispatcher: Still no focused window. Will drop the event in 4968ms
2021-07-26 00:12:23.739 14708-14775/system_process W/InputDispatcher: Still no focused window. Will drop the event in 4951ms
2021-07-26 00:12:23.741 14631-14647/? W/APM::AudioPolicyEngine: getDevicesForStrategy() unknown strategy: -1
2021-07-26 00:12:23.743 14708-16106/system_process I/system_server: oneway function results will be dropped but finished with status OK and parcel size 4
2021-07-26 00:12:23.757 15980-15980/com.android.documentsui D/AbstractActionHandler: Loader has finished for: content://com.android.providers.media.documents/document/videos_bucket%3A-1730634595
2021-07-26 00:12:23.758 15980-15980/com.android.documentsui I/Model: Updating model with new result set.
2021-07-26 00:12:23.759 14708-14775/system_process W/InputDispatcher: Still no focused window. Will drop the event in 4931ms
2021-07-26 00:12:23.759 15980-15980/com.android.documentsui D/DirectoryFragment: Received model update. Loading=false
2021-07-26 00:12:23.770 14708-14775/system_process W/InputDispatcher: Still no focused window. Will drop the event in 4921ms
2021-07-26 00:12:23.786 14708-14775/system_process W/InputDispatcher: Still no focused window. Will drop the event in 4905ms
2021-07-26 00:12:23.793 15980-16544/com.android.documentsui D/ProvidersAccess: Matched roots: [Root{userId=0, authority=null, rootId=null, title=Recent, isUsb=false, isSd=false, isMtp=false} @ content:/root/null, Root{userId=0, authority=com.android.externalstorage.documents, rootId=primary, title=opi_plus2e, isUsb=false, isSd=false, isMtp=false} @ content://com.android.externalstorage.documents/root/primary, Root{userId=0, authority=com.android.providers.downloads.documents, rootId=downloads, title=Downloads, isUsb=false, isSd=false, isMtp=false} @ content://com.android.providers.downloads.documents/root/downloads, Root{userId=0, authority=com.android.providers.media.documents, rootId=images_root, title=Images, isUsb=false, isSd=false, isMtp=false} @ content://com.android.providers.media.documents/root/images_root, Root{userId=0, authority=com.android.providers.media.documents, rootId=videos_root, title=Videos, isUsb=false, isSd=false, isMtp=false} @ content://com.android.providers.media.documents/root/videos_root, Root{userId=0, authority=com.android.providers.media.documents, rootId=audio_root, title=Audio, isUsb=false, isSd=false, isMtp=false} @ content://com.android.providers.media.documents/root/audio_root, Root{userId=0, authority=com.android.providers.media.documents, rootId=documents_root, title=Documents, isUsb=false, isSd=false, isMtp=false} @ content://com.android.providers.media.documents/root/documents_root, Root{userId=0, authority=com.android.shell.documents, rootId=bugreport, title=Bug reports, isUsb=false, isSd=false, isMtp=false} @ content://com.android.shell.documents/root/bugreport, Root{userId=0, authority=com.android.traceur.documents, rootId=traces, title=System traces, isUsb=false, isSd=false, isMtp=false} @ content://com.android.traceur.documents/root/traces]
2021-07-26 00:12:23.803 14708-14775/system_process W/InputDispatcher: Still no focused window. Will drop the event in 4888ms
2021-07-26 00:12:23.824 14708-14775/system_process W/InputDispatcher: Still no focused window. Will drop the event in 4867ms
[... cut ...]
2021-07-26 00:12:24.106 14708-14775/system_process W/InputDispatcher: Still no focused window. Will drop the event in 4585ms
2021-07-26 00:12:24.106 14708-14775/system_process D/InputDispatcher: Waiting to send key to Window{13c04fc u0 com.android.documentsui/com.android.documentsui.files.FilesActivity} because there are unprocessed events that may cause focus to change
2021-07-26 00:12:24.133 14559-14679/? E/hwc-platform-drm-generic: Failed to close gem handle 1, errno: 22
2021-07-26 00:12:24.133 14559-14679/? E/hwc-platform-drm-generic: Failed to close gem handle 1, errno: 22

thank you.
``

picoworm, 2021-07-26:
Hi,

To do this you have to implement codec2 components that will support v4l2 request API.
Or use this: https://android.googlesource.com/platform/external/v4l2_codec2/

Both are not trivial and require a lot of knowledge, time and paitiance for harware-decoder bring-up.

rsglobal, 2021-07-26:
Hi,

To do this you have to implement codec2 components that will support v4l2 request API.
Or somehow use this: https://android.googlesource.com/platform/external/v4l2_codec2/

Both are not trivial and require a lot of knowledge, time and paitiance for harware-decoder bring-up.

joek85, 2021-07-26:
Hello, 

I've enabled

`CONFIG_MEDIA_SUPPORT`
`CONFIG_MEDIA_CONTROLLER`
`CONFIG_MEDIA_CONTROLLER_REQUEST_API`
`CONFIG_V4L_MEM2MEM_DRIVERS`
`CONFIG_VIDEO_SUNXI_CEDRUS`

and successfully got 
`/dev/video0`
`/dev/media0`

now, looking at android v4l2_codec2 make file, it seems that it can only be built for 
`device/google/cheets2 `
which is not available in the AOSP source tree.
all other dependencies that are required by v4l2_codec2 are available but only 
`/device/google/cheets2/codec2/vdastore/include` 
which is not.

however i will try to install VLC and see what happens, just to test if the kernel driver is working as expected.

rsglobal, 2021-07-26:
> device/google/cheets2

This was required on Android-10,
Looks like Android-11 and master doesn't need it.

joek85, 2021-07-26:
> This was required on Android-10,

you're right, my bad, didn't catch that.
i am currently working on Android10, i don't have enough bandwidth to download Android11.
i will try to port it to my source and hope it will compile with no problems.

acrolink, 2021-09-30:
Any updates on this? Does this build support `video/avc` (H264) hardware encoding and decoding?

volkertb, 2022-04-24:
Quoting [this reply in the duplicate topic](https://github.com/GloDroid/glodroid_manifest/issues/175#issuecomment-995630016):

> Duplicate of #152 .
> 
> Currently only software codecs are supported. There is a way to integrate stateful codecs support which we will do for rpi4 near future, but allwinner has only stateless codec APIs which require a lot of work to enable on mainline Android.

Have you considered [vicodec](https://www.collabora.com/news-and-blog/blog/2019/10/09/adding-stateless-support-to-vicodec/)? Apparently, with Dafna's improvements, it can provide stateful codec emulation for stateless hardware codecs, such as the one in the Pinephone's SoC.

rsglobal, 2022-04-24:
@volkertb ,

Unfortunately all I can find in regard to vicodec in the kernel code and Dafna's blog is an stateless API for software encoding/decoding based on FWHT algorithm.

volkertb, 2022-04-24:
In her blog she says that the `vicodec` driver should expose three interfaces under `/dev`:

- One for a stateful encoder (not applicable for the Allwinner SoC)
- One for a stateless decoder (which basically passes through to the new stateless "Request API")
- One for a stateful decoder (which should provide stateful decoder emulation and talks to stateless codec drivers such as Cedrus on the backend through the Request API)

So as far as I understand it, the chain goes like this:

V4L2 -> `vicodec` -> Cedrus (the driver for the Allwinner video engine)

This seems to be the relevant commit:
- https://github.com/torvalds/linux/commit/997deb811bf55731a7c0bd972d8d0a266ef8d7e5

There is also this PDF document from Hans Verkuil at Cisco Systems: https://elinux.org/images/7/76/Elce2019v4l2.pdf

In that document, he specifically refers to `vicodec` as a "virtual driver" that "emulates a stateful encoder, a stateful decoder and a stateless decoder".

Would this work for GloDroid?

Perhaps we could email Dafna and/or Hans for questions if there are still any missing puzzle pieces. :slightly_smiling_face:

rsglobal, 2022-11-26:
ffmpeg-based codec2 service is a good candidate to add stateless (and stateful) v4l2-based codec support:

 - ffmpeg + request API patches: https://github.com/jernejsk/FFmpeg
 - codec2 to ffmpeg layer (dirty at the moment): https://github.com/goffioul/external-stagefright-plugins

KonstaT, 2023-01-15:
I managed to get FFmpeg H.264/H.265/VP8/VP4 software decoders working on Pi 4 with slightly modified
https://github.com/waydroid/android_external_ffmpeg
https://github.com/waydroid/android_external_stagefright-plugins
and relevant patches to libstagefright on frameworks/av.

Tried to use FFmpeg's h264_v4l2m2m for H.264 hardware decoding but not getting any buffers out of it.
```
01-15 17:19:18.641   335  1920 I GenericSource: start
01-15 17:19:18.662   335  1931 D CCodec  : allocate(c2.android.aac.decoder)
01-15 17:19:18.663   335  1932 D CCodec  : allocate(c2.ffmpeg.h264.decoder)
01-15 17:19:18.674   335  1932 I CCodec  : setting up 'default' as default (vendor) store
01-15 17:19:18.674   282   695 D android.hardware.media.c2@1.2-service: createComponent: c2.ffmpeg.h264.decoder
01-15 17:19:18.676   282   695 D C2FFMPEGVideoDecodeComponent: C2FFMPEGVideoDecodeComponent: mediaType = video/avc
01-15 17:19:18.677   335  1931 I CCodec  : setting up 'default' as default (vendor) store
01-15 17:19:18.678   343   701 V C2Store : in init
01-15 17:19:18.679   343   701 V C2Store : loading dll
01-15 17:19:18.691   335  1931 I CCodec  : Created component [c2.android.aac.decoder]
01-15 17:19:18.697   335  1931 D CCodecConfig: read media type: audio/mp4a-latm
01-15 17:19:18.668   335   335 I CodecLooper: type=1400 audit(0.0:153): avc: denied { transfer } for scontext=u:r:mediaserver:s0 tcontext=u:r:init:s0 tclass=binder permissive=1
01-15 17:19:18.706   335  1932 I CCodec  : Created component [c2.ffmpeg.h264.decoder]
01-15 17:19:18.707   335  1931 D ReflectedParamUpdater: extent() != 1 for single value type: algo.buffers.max-count.values
01-15 17:19:18.707   335  1931 D ReflectedParamUpdater: extent() != 1 for single value type: output.subscribed-indices.values
01-15 17:19:18.707   335  1932 D CCodecConfig: read media type: video/avc
01-15 17:19:18.708   335  1931 D ReflectedParamUpdater: extent() != 1 for single value type: input.buffers.allocator-ids.values
01-15 17:19:18.708   335  1931 D ReflectedParamUpdater: extent() != 1 for single value type: output.buffers.allocator-ids.values
01-15 17:19:18.709   335  1931 D ReflectedParamUpdater: extent() != 1 for single value type: algo.buffers.allocator-ids.values
01-15 17:19:18.709   335  1931 D ReflectedParamUpdater: extent() != 1 for single value type: output.buffers.pool-ids.values
01-15 17:19:18.709   335  1931 D ReflectedParamUpdater: extent() != 1 for single value type: algo.buffers.pool-ids.values
01-15 17:19:18.711   335  1932 D ReflectedParamUpdater: extent() != 1 for single value type: algo.buffers.max-count.values
01-15 17:19:18.712   335  1932 D ReflectedParamUpdater: extent() != 1 for single value type: output.subscribed-indices.values
01-15 17:19:18.712   335  1932 D ReflectedParamUpdater: extent() != 1 for single value type: input.buffers.allocator-ids.values
01-15 17:19:18.713   335  1932 D ReflectedParamUpdater: extent() != 1 for single value type: output.buffers.allocator-ids.values
01-15 17:19:18.713   335  1932 D ReflectedParamUpdater: extent() != 1 for single value type: algo.buffers.allocator-ids.values
01-15 17:19:18.713   335  1932 D ReflectedParamUpdater: extent() != 1 for single value type: output.buffers.pool-ids.values
01-15 17:19:18.713   335  1932 D ReflectedParamUpdater: extent() != 1 for single value type: algo.buffers.pool-ids.values
01-15 17:19:18.716   335  1932 D ReflectedParamUpdater: ignored struct field coded.color-format.locations
01-15 17:19:18.716   335  1932 D CCodecConfig: ignoring local param raw.size (0xd2001800) as it is already supported
01-15 17:19:18.717   335  1932 D ReflectedParamUpdater: ignored struct field raw.hdr-static-info.mastering
01-15 17:19:18.719   335  1932 I CCodecConfig: query failed after returning 8 values (BAD_INDEX)
01-15 17:19:18.719   335  1932 D CCodecConfig: c2 config diff is Dict {
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::u32 coded.pl.level = 20491
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::u32 coded.pl.profile = 20482
01-15 17:19:18.719   335  1932 D CCodecConfig:   Buffer coded.raw-codec-data.value = {
01-15 17:19:18.719   335  1932 D CCodecConfig:   }
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::u32 default.color.matrix = 0
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::u32 default.color.primaries = 0
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::u32 default.color.range = 0
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::u32 default.color.transfer = 0
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::u32 input.delay.value = 0
01-15 17:19:18.719   335  1932 D CCodecConfig:   string input.media-type.value = "video/avc"
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::u32 output.delay.value = 8
01-15 17:19:18.719   335  1932 D CCodecConfig:   string output.media-type.value = "video/raw"
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::u32 raw.pixel-format.value = 256
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::i32 raw.rotation.flip = 0
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::i32 raw.rotation.value = 0
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::u32 raw.sar.height = 1
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::u32 raw.sar.width = 1
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::u32 raw.size.height = 240
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::u32 raw.size.width = 320
01-15 17:19:18.719   335  1932 D CCodecConfig:   c2::u32 raw.surface-scaling.value = 1
[... cut ...]
Pi's stateless H.265 hw decoder seems to be entirely different story as it's not even supported on mainline FFmpeg but requires ton of external patches (https://github.com/jc-kynesim/rpi-ffmpeg) and apparently produces buffers in SAND format(?).

H.264 hardware video decoding and encoding is now somewhat working on my LineageOS 20 builds and Raspberry Vanilla sources (https://github.com/raspberry-vanilla) with AOSP v4l2_codec2 anyway (decoding still produces random glitches/green tint on occasions).

acrolink, 2023-01-15:
@KonstaT 

Great news, thank you very much. Should I just go ahead, download and test? Is it ready?

goffioul, 2023-01-15:
@KonstaT From the logs, it seems you're not using the latest version if my repo (or patch set), so you may to update that. Regarding your question, I don't have enough knowledge of v4l2 codec integration in ffmpeg to answer it. The code now uses the generci HW accel API from FFMPEG, so if the v4l2_m2m codec adheres to it, it should work, in theory...

KonstaT, 2023-01-15:
@goffioul
Thanks. Yeah, I had some strategic reverts for testing purposes. Didn't seem to make any difference, though. Wouldn't bee too surprising if I have some configuration issue on FFmpeg side either.

KonstaT, 2023-01-19:
Managed to get both hevc_v4l2request hwaccel for stateless H.265 hw decoding and h264_v4l2m2m for stateful H.264 hw decoding working after merging jc-kynesim's FFmpeg branch!
https://twitter.com/KonstaKANG/status/1616054277629218819

Only hevc_v4l2request is actually listed as hwaccel on FFmpeg. h264_v4l2m2m is just a decoder. H.265 hw decoding seems fine at 1080p30 (haven't found test clips with higher frame rate or 4k). H.264 hw decoding seems fine at 1080p30 as well but can't quite keep up with 1080p60. h264_v4l2m2m actually performs worse than FFmpeg's software decoder that can also do 1080p60. :o FFmpeg's h264 software decoder also outperforms Android's own h264 software decoder as that can't keep up with 1080p60 either.

@goffioul
Couple of questions about the libstagefright integration.

I see you have have the plugin in vendor partition. I had to move ffmpeg and the plugin to system partition as libstagefright is framework only. Would making libstagefright vendor available allow moving ffmpeg/plugin to vendor as well or how you've managed that? It looked like libstagefright has other dependencies as well so I didn't even bother trying yet.

What patches to libstagefright at frameworks/av are relevant if I just want to use the codec2 plugin and none of the old omx stuff?

Also somewhat unrelated, but what's currently the place to keep up with latest developments with Android on x86? Android-x86, BlissOS, Waydroid, or something else?

goffioul, 2023-01-19:
I you only want the codec2 stuff, you might as well try without any of the /framework/av patches and see where this gets you (that also assumes you're not interested in the FFMPEG based extractor as well).

The Android on x86 community is unfortunately quite fragmented and small. The original developer hasn't been active lately and the mailing lists are more or less in limbo. There's an android-x86-dev channel on BlissLabs slack channel, where some of us are active. I know there's also the discord and telegram community, but I don't use that personally.

rsglobal, 2023-11-30:
Support for HW video codecs added:

https://github.com/GloDroidCommunity/pine64-pinephone/commit/1fce8f8fd3e8d089b2dddb27910314fc946bebd8

BTW, if somebody wants to see updates for OPI3, OPI-Plus2E, and other Sunxi devices that were supported previously, open an issue for that.

closed 2023-11-30 by rsglobal