GloDroid/glodroid_manifest #166: RPI4: Libcamera integration CSI+ISP

rsglobal, 2021-11-11:
We've made a good progress towards libcamera integration, preview and shots are working, still it require more testing / bug fixing.

### Some of links: 
#### Discussions with the maintainers:
https://github.com/kbingham/libcamera/issues/31
https://github.com/kbingham/libcamera/pull/26

#### Local branches
1. LibCamera: 
https://github.com/GloDroid/glodroid_forks/tree/libcamera-v0.7.0
2. Libraries: 
https://github.com/GloDroid/glodroid_forks/tree/libyaml-v0.7.0
https://github.com/GloDroid/glodroid_forks/tree/libyuv_chromium-v0.7.0

#### Configuration:
1. https://github.com/GloDroid/glodroid_manifest/commit/0a54e5a909029056ab0fd6267310c2a3a7b1e604
2. https://github.com/GloDroid/glodroid_device/commit/5c258a304994e69e7257341e889953f1e728126b
3. https://github.com/GloDroid/glodroid_device/commit/035c402a8988710e5d56ba63f1a1152c64e60bb1

rsglobal, 2021-11-11:
@KonstaT,

Hi,

I've posted it here in case you would like to try it out in your builds.

KonstaT, 2021-11-11:
Thanks. I'll surely give it a test and report back. Probably won't be until next week as I'm away from the keyboard for the rest of the week.

KonstaT, 2021-11-16:
I tested this on Raspberry Pi 4 with 5.10.78 kernel and Pi camera modules v1 (OV5647) and v2 (IMX219). It seems to work well for camera preview and photos with the stock camera app with both camera hw. Didn't find it to be too unstable or crash (might have missed some though as the logging is highly verbose currently). Camcorder preview shows garbled image and capturing videos doesn't work. Only third party camera app I tested was OpenCamera and camera preview and photos don't work with that (shows same garbled image as stock app with camcorder).

Some notes.

With recent firmware you can set `camera_auto_detect=1` in config.txt which is handy if you want to support more than one camera hardware. Also for some reason just setting e.g. `dtoverlay=imx219` didn't work so I wonder what else the firmware enables with auto detect.

My camera_hal.yaml looks like this to support camera modules v1, v2 & HD.
```
cameras:
  "/base/soc/i2c0mux/i2c@1/imx219@10":
    location: back
    rotation: 0
  "/base/soc/i2c0mux/i2c@1/imx477@1a":
    location: back
    rotation: 0
  "/base/soc/i2c0mux/i2c@1/ov5647@36":
    location: back
    rotation: 0
```

If you move /vendor/etc/libcamera/ipa_rpi.so.sign to /vendor/lib(64)/ (same place as ipa_rpi.so) you get from
`D libcamera: DEBUG IPAModule ipa_module.cpp:320 ipa_rpi.so: IPA module /vendor/lib64/libcamera//ipa_rpi.so is not signed`
to
`D libcamera: DEBUG IPAModule ipa_module.cpp:329 ipa_rpi.so: IPA module /vendor/lib64/libcamera//ipa_rpi.so is signed`
Untested if this resolves the issue you had with IPA signature.

If you wasn't aware I have Android fork of v4l-utils here (https://github.com/lineage-rpi/android_external_v4l-utils/tree/lineage-19.0). Especially `v4l2-ctl` comes handy when working with cameras and hw dec/enc. E.g. to check which dev nodes belong to which driver (to set permissions).
```
console:/ # v4l2-ctl --list-devices                                            
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/media0

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/media1

unicam (platform:fe801000.csi):
        /dev/video0
        /dev/video1
        /dev/media2
```

Overall IMO this looks very promising so great job as always! :)

On somewhat related matters, there's was a report from an user with external USB camera with 
`[11-09 22:33:57.428 263:263 E/[minigbm:CrosGralloc4Allocator.cc(58)]] Unsupported combination -- pixel format: PixelFormat::YV12, drm format:DRM_FOURCC_9997, usage: BufferUsage::CAMERA_OUTPUT|BufferUsage::COMPOSER_OVERLAY|BufferUsage::CPU_WRITE_OFTEN|BufferUsage::GPU_TEXTURE`
I assume some DRM_FORMAT_YVU420_ANDROID handling in minigbm's gbm_mesa platform would sort this? I currently don't have any USB webcams that are supported with the AOSP external camera HAL that I can use to test.

Also would https://android.googlesource.com/platform/external/minigbm/+/3a171db9be34ef1e9ff62e3ee17bc0632cf4cdc7 and https://android.googlesource.com/platform/external/minigbm/+/71bc665179f9e8a94c4b2293f6b3406887f96d34 be something that we'd want for gbm_mesa?

rsglobal, 2021-11-16:
Thanks for all your comments. It was very useful. I'll update GloDroid with more cams support.

> I assume some DRM_FORMAT_YVU420_ANDROID handling in minigbm's gbm_mesa platform would sort this?

I'll test it soon.

> Also would https://android.googlesource.com/platform/external/minigbm/+/3a171db9be34ef1e9ff62e3ee17bc0632cf4cdc7 and https://android.googlesource.com/platform/external/minigbm/+/71bc665179f9e8a94c4b2293f6b3406887f96d34 be something that we'd want for gbm_mesa?

Currently I'm running GD through all those CTS test and will add all necessary handling. Also I plan to continuously rebase onto upstream minigbm, so all new patches will enter into my fork sooner or later.

Just in case you also want to play CTS&VTS game, I started to accumulate some experience here:
https://github.com/GloDroid/glodroid_manifest/wiki/VTS-&-CTS

rsglobal, 2021-11-16:
I redesigned minigbm to support size alignment. Setting size_alignment to 4096 (page) fixed camera crashes. (available in minigbm-next branch).
I assume it may also help you with codecs issues.

rsglobal, 2021-11-20:
Preview and shots are working. Will be available in v0.7.0.

closed 2021-11-20 by rsglobal

rsglobal, 2021-11-25:
@KonstaT ,

I used info form dts/overlays/* to extract nodes into .yaml file for remaining cameras. Not tested as I do not have them:

https://github.com/GloDroid/glodroid_device/commit/be6ba9a457c28a70f1307c69639a7caeecad0b22#diff-b34d0c03657ef9f5b48562515ea1c69544921cb8e1beb112f793781d899da5d6

KonstaT, 2021-11-25:
Thanks. Yep, that's where I got them for the Pi camera modules as well. I'm sticking with the official modules for now but I'll probably add the configs/kernel drivers for the other hw supported with libcamera later as well. I don't have any other hardware to test myself either.

I shipped libcamera with my LineageOS 19 build today so I'm sure people will complain if it's utterly broken. I was very clever again and reverted your logging change which apparently killed all logs for libcamera so there won't be any logs to help, though. :P