GloDroid/glodroid_manifest #167: Default Password?

Csanderson03, 2021-11-12:
Hey, What is the default password that its asking for to get log in on the phone?

Thank You so much
Chris

rsglobal, 2021-11-12:
Hi,

GloDroid doesn't have any passwords

Csanderson0313, 2021-11-12:
Oh wait, I know why. It's not actually loading on my SD card. Why does it say not a bootable image when loading?
Thank you

rsglobal, 2021-11-13:
Could you provide more info on your system, what is happening and what is expected to happen. Attach some logs if possible.

Csanderson0313, 2021-11-13:
I will go and download GloDroid 6.1. I will use balena etcher to Flash it on my SD card. When I select the file to flat, a pop up saying "missing partition table" (see attached picture).
[image]

[image]
 
Thank you

rsglobal, 2021-11-13:
You have to unpack it first.

Csanderson0313, 2021-11-13:
I did unzip the file and doing the same thing. What unzip program are you using? Maybe it's the unzip software that I'm using that causing issues.

rsglobal, 2021-11-20:
Please refer to README.md file for instructions or search for installation guidelines on youtube.

closed 2021-11-20 by rsglobal