GloDroid/glodroid_manifest #168: need help with installing for RPI4

kiai72, 2021-11-12:
so there isn't a binary that i can flash directly with rpi imager or balena or something similar ?

if i understood right i have to flash deploy-sd.img what acts as kind of bootloader ?
then i have to connect the RPI to my PC that flashes the OS by fastboot, right ?

But what kind of cable do i need for that ? i only know OTG from smartphones (micro USB or USB C) - but how do i connect a RPI via OTG to a pc ?

rsglobal, 2021-11-12:
Thanks for reporting, i'll check it.
Could you update issue description accordingly?

SimoneFrau, 2021-11-18:
For me it works, just do not use the open camera app already installed. That's OK for the CSI camera, but for the USB camera install some other apk it's possible to find for free on web

kiai72, 2021-11-18:
can you please check if it works for you in a browser (chrome dev or firefox) ? https://webcamtests.com/

SimoneFrau, 2021-11-18:
I've just tried v0.6.1 and in the browser does not work

kiai72, 2021-11-18:
ok, 
my main-usage for android is: i want to do  jitsi, zoom and skype.
jitsi works fine in raspberryOS - in the browser.
But zoom hardly works - and skype not at all. Not even in the browser -> that's why i wanted to have the camera running in browser.
But if it works in zoom app and skype app - that would be fine as well. (do you know ?)

SimoneFrau, 2021-11-18:
No idea if Zoom and Skype are available, I do not see them in Aurora Store but I feel they are quite heavy as apps...

kiai72, 2021-11-20:
i tried skype lite (install apk). I can use skype - but my camera picture is not captured and shown to the other participants.

SimoneFrau, 2021-12-03:
Quick update about this issue using the new version based on Android 12. Skype is now available as apk but USB camera is not detected by it. On the opposite CSI Rpi Camera yes, but the image gets completely messed up.

kiai72, 2021-12-04:
thank you for keeping me updated. I hope USB-cameras will be supported in future.

KonstaT, 2021-12-16:
I got myself a HDMI-USB video capture device as I didn't need another USB webcam (one I have doesn't support MJPG). Someone recommended it because it uses MJPG and it's still basically an UVC webcam so it can be used with the external camera HAL.

I had two errors with minigbm gralloc.

One with starting the camera preview:
```
E [minigbm:CrosGralloc4Allocator.cc(52)]: Unsupported combination -- pixel format: YV12, drm format:DRM_FOURCC_9997, usage: CPU_READ_NEVER | CPU_WRITE_NEVER | CPU_WRITE_RARELY | CP
U_WRITE_OFTEN | GPU_TEXTURE | CAMERA_OUTPUT (0x20130)
```
And another one when capturing an image:
```
E [minigbm:CrosGralloc4Allocator.cc(52)]: Unsupported combination -- pixel format: BLOB, drm format:DRM_FOURCC_R8  , usage: CPU_READ_NEVER | CPU_READ_RARELY | CPU_READ_OFTEN | CPU_
WRITE_NEVER | CPU_WRITE_RARELY | CPU_WRITE_OFTEN | CAMERA_OUTPUT (0x20033)
```

Added this: https://github.com/lineage-rpi/android_external_minigbm/commit/f9d7925d81c7b40efb56586188f34952f29dc7e9
Preview, photos, and recording videos works fine after that and doesn't seem it broke anything with Pi camera modules or hw video dec. Only "issue" is that camera preview is vertically mirrored (selfie mirror effect?) but only on the camcorder preview. :o

@rsglobal 
Does that look sane to you?

rsglobal, 2021-12-16:
@KonstaT 

> Does that look sane to you?

More than. Thank you. Will add this to the my minigbm for as well.

kiai72, 2021-12-16:
nice! will this be implemented in GloDroid and/or konstagang's rom soon ?

rsglobal, 2021-12-16:
Unfortunately both of my cameras don't expose V4L2_CAP_VIDEO_CAPTURE , therefore can't be used with external camera HAL (at least without modification of the HAL).

KonstaT, 2021-12-16:
> nice! will this be implemented in GloDroid and/or konstagang's rom soon ?

It's already in my latest LineageOS 19 build and I see Roman added a similar thing on GloDroid. But as you can see from the comments it's likely that there's more UVC webcams that won't work for various reasons than ones that do.

> Unfortunately both of my cameras don't expose V4L2_CAP_VIDEO_CAPTURE 

I guess getting that HDMI-USB video capture card was a good suggestion that was made to me.

KonstaT, 2021-12-16:
> ? can you please explain: what is that card ? you connect it via HDMI to the PI - or via USB ?

If you were really cheeky, why not both. :P

It has a USB in one and HDMI in another end. It's used to make screen recordings i.e. capture HDMI signal from some device and it exposes it as an USB camera to another device to record (Pi with the camera app in this case).

kiai72, 2021-12-16:
> > ? can you please explain: what is that card ? you connect it via HDMI to the PI - or via USB ?
> 
> If you were really cheeky, why not both. :P
> 
> It has an USB in one end and HDMI in another end. It's used to make screen recordings i.e. capture HDMI signal from some device and it exposes it as an USB camera to another device to record (Pi with the camera app in this case). Or you could also use it capture the screen of your Pi as e.g. many YouTubers do (but that's not the case here).

got it. Thank you.
By the way: in the meanwhile i checked your konstakang-release of today. Camera app says "cannot connect to camera". It's a USB UVC cam with MJPG.
Should i try to get a log ? Should i post it here - or at konstakang.com ?

SimoneFrau, 2021-12-20:
And any idea why with Skype the image from the CSI Rpi Camera is completely messed up? It works very well with the included Camera app

kiai72, 2021-12-21:
Besides "Skype" there is also "Skype Lite"!  Maybe you want to try if this works. https://www.apkmirror.com/apk/skype/skype-lite/

SimoneFrau, 2021-12-21:
With Skype Lite same behavior + Android Reboot  :)

kiai72, 2021-12-21:
there's also a browser based skype version (skype web). It works with webrtc.
Microsoft tries to block all browsers beside chrome and edge. 
I had it running on fydeOS/chromeOS on the raspberry before, so there blocker is not 100%. Maybe it's worth trying. Maybe with "chrome dev", as far as i know even firefox is blocked.

SimoneFrau, 2021-12-22:
I was able to use Skype Web on Firefox, CSI Rpi Camera works fine only on Preview. I do not know why it is not available during normal calls. Anyway it can be a limitation on Firefox or HW porting, I am using an Rpi4 and you?

kiai72, 2021-12-22:
i remember that i had similar effects: see my camera picture - but it will not be transfered to the other participants. Maybe that's the effect you have ?
I'm also using a rpi4 4b. But honestly: since the last version of glodroid i tried did not support USB cam at all - i concentrated more on konstas at the moment.
With latest konstas i made progress but still have a lot of problems:
1. my webcam does not work - but logitech c310 (borrowed) works!
2. the audio of the RPI will not be transfered to the other participants

So i keep watching all options and roms at the moment. The only one that works for me is fydeOs.
But i would rather have a "real" android, that's why i try glodroid and konsta.

SimoneFrau, 2021-12-22:
I've understood, unfortunately for me the version from KonstaT is not an option because I am changing things everywhere for my real activity which is this one https://github.com/GloDroid/glodroid_manifest/issues/177

I am using GloDroid with the updated master code + I install Aurora Store to install the app Skype version which you can find from there but USB cams are not seen. the CSI Rpi Camera yes and I am able to call using an USB Audio Set and I hear others and others can hear me without problem, the only thing is the image of the camera which is like a really low resolution but anyway me and others can see it.

Using bluetooth Audio Set there is a further issue because it is seen as a Speaker and not as a HeadSet or HandFree Bluetooth device, which means you ear everything but the mic does not work. This has to be something with the porting of a Client Profile for these two kinds of BL devices which is not in place at the moment. Anyway consider this is an issue in Linux too!