GloDroid/glodroid_manifest #170: Building error after sync of the next branch

SimoneFrau, 2021-11-16:
Hi,

I've just got an error on the next branch after a repo sync for rpi4 -> option 77 in lunch. I did not have this problem until the sync, but I was one or two weeks behind. I've also cleaned all my changes so it can not be my stuff.
This is the error:

build out/target/product/rpi4/obj/ATF_OBJ/rpi4/debug/bl31.bin
FAILED: out/target/product/rpi4/obj/ATF_OBJ/rpi4/debug/bl31.bin
/bin/bash -c "M0_CROSS_COMPILE=/home/simo/Projects/Distro/GloDroid/next/prebuilts/gcc/linux-x86/arm/gcc-linaro-arm-eabi/bin/arm-eabi- PATH=/u
sr/bin:/bin:/sbin:\$PATH ARCH=arm64 CROSS_COMPILE=/home/simo/Projects/Distro/GloDroid/next/prebuilts/gcc/linux-x86/aarch64/gcc-linaro-aarch64
-linux-gnu/bin/aarch64-linux-gnu- make -j4 CC=/home/simo/Projects/Distro/GloDroid/next/prebuilts/clang/host/linux-x86/clang-r416183b1/bin/cla
ng HOSTCC=/home/simo/Projects/Distro/GloDroid/next/prebuilts/clang/host/linux-x86/clang-r416183b1/bin/clang LD=/home/simo/Projects/Distro/Glo
Droid/next/prebuilts/clang/host/linux-x86/clang-r416183b1/bin/ld.lld -C external/arm-trusted-firmware BUILD_BASE=/home/simo/Projects/Distro/G
loDroid/next/out/target/product/rpi4/obj/ATF_OBJ PLAT=rpi4 DEBUG=1 bl31"
make: Entering directory '/home/simo/Projects/Distro/GloDroid/next/external/arm-trusted-firmware'
  CC      plat/rpi/common/rpi3_pm.c
  CC      lib/xlat_tables_v2/xlat_tables_context.c
  CC      lib/xlat_tables_v2/xlat_tables_core.c
  CC      lib/xlat_tables_v2/xlat_tables_utils.c
plat/rpi/common/rpi3_pm.c:268:29: error: initializer overrides prior initialization of this subobject [-Werror,-Winitializer-overrides]
        .pwr_domain_pwr_down_wfi = rpi3_pwr_down_wfi,
                                   ^~~~~~~~~~~~~~~~~
plat/rpi/common/rpi3_pm.c:265:29: note: previous initialization is here
        .pwr_domain_pwr_down_wfi = rpi3_pwr_domain_pwr_down_wfi,
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make: *** [Makefile:1201: /home/simo/Projects/Distro/GloDroid/next/out/target/product/rpi4/obj/ATF_OBJ/rpi4/debug/bl31/rpi3_pm.o] Error 1
make: *** Waiting for unfinished jobs....
make: Leaving directory '/home/simo/Projects/Distro/GloDroid/next/external/arm-trusted-firmware'
17:30:08 ninja failed with: exit status 1

Thank you!
Simone

SimoneFrau, 2021-11-17:
I found a workaround commenting line 268 in file external/arm-trusted-firmware/plat/rpi/common/rpi3_pm.c

//	.pwr_domain_pwr_down_wfi = rpi3_pwr_down_wfi,

and also all the static function rpi3_pwr_down_wfi() defined at line 177 and not called anywhere else.

Actually I can not be sure this is the right function to delete and not rpi3_pwr_domain_pwr_down_wfi() called at line 265.

BTW this issue was already present in master branch because the file is the same, it's just not compiled

rsglobal, 2021-11-18:
Thank you for reporting this issue. I created a ticket in ATF bug-tracker:

https://developer.trustedfirmware.org/T977

SimoneFrau, 2021-11-18:
Thanks to you for this work! Camera is working and also the stability of the platform is much better with the last commits.

Any chance to have a look to this?

https://github.com/GloDroid/glodroid_manifest/issues/87

I'm working trying to have Bluetooth working but nothing...

rsglobal, 2021-11-18:
Sure, but ATM I am busy with other activities.

rsglobal, 2021-11-20:
Next branch is our internal work in progress brach and may not work sometimes.
Please try to update in a week or so.

closed 2021-11-20 by rsglobal