GloDroid/glodroid_manifest #172: Rpi4: Upstream Android for v3dv support

rsglobal, 2021-11-20:
Some initial  work was done and shared by @KonstaT, in https://github.com/GloDroid/glodroid_manifest/issues/160#issuecomment-940197543

TODO:
Rebase on top of mainline mesa, test and send to upstream

rsglobal, 2021-11-23:
`device/glodroid` changes required:

```diff
diff --git a/common/sepolicy/vendor/file_contexts b/common/sepolicy/vendor/file_contexts
index 228305b..5c6e628 100644
--- a/common/sepolicy/vendor/file_contexts
+++ b/common/sepolicy/vendor/file_contexts
@@ -13,2 +13,3 @@
 /vendor/lib(64)?/libdrm.so                                                      u:object_r:same_process_hal_file:s0
+/vendor/lib(64)?/hw/vulkan\.broadcom\.so                                        u:object_r:same_process_hal_file:s0
 
diff --git a/rpi4/BoardConfig.mk b/rpi4/BoardConfig.mk
index 29abb3b..d7b519d 100644
--- a/rpi4/BoardConfig.mk
+++ b/rpi4/BoardConfig.mk
@@ -23,2 +23,3 @@ TARGET_BOARD_INFO_FILE := device/glodroid/rpi4/board-info.txt
 BOARD_MESA3D_GALLIUM_DRIVERS := vc4 v3d
+BOARD_MESA3D_VULKAN_DRIVERS := broadcom
 
diff --git a/rpi4/device.mk b/rpi4/device.mk
index 64f2b0e..f4fb9bb 100644
--- a/rpi4/device.mk
+++ b/rpi4/device.mk
@@ -37 +37,13 @@ PRODUCT_COPY_FILES += \
     device/glodroid/rpi4/camera_hal.yaml:$(TARGET_COPY_OUT_VENDOR)/etc/libcamera/camera_hal.yaml \
+
+# Vulkan
+PRODUCT_PACKAGES += \
+    vulkan.broadcom
+
+PRODUCT_COPY_FILES += \
+    frameworks/native/data/etc/android.hardware.vulkan.level-1.xml:$(TARGET_COPY_OUT_VENDOR)/etc/permissions/android.hardware.vulkan.level.xml \
+    frameworks/native/data/etc/android.hardware.vulkan.version-1_0_3.xml:$(TARGET_COPY_OUT_VENDOR)/etc/permissions/android.hardware.vulkan.version.xml \
+    frameworks/native/data/etc/android.software.vulkan.deqp.level-2021-03-01.xml:$(TARGET_COPY_OUT_VENDOR)/etc/permissions/android.software.vulkan.deqp.level.xml \
+
+PRODUCT_VENDOR_PROPERTIES += \
+    ro.hardware.vulkan=broadcom
```

KonstaT, 2021-11-23:
I now have v3dv Android rebased on top of Mesa 21.3 here (https://github.com/lineage-rpi/android_external_mesa3d/commit/7b48d7a35e3ba445c1c909f0b2066ff7192dc95a).
There was some conflicting changes from before and also this (https://github.com/lineage-rpi/android_external_mesa3d/commit/09ee5dbdb636d9cc4bf756b9e92f931a93e1c493) is now needed but I'm not exactly sure why.

I haven't looked into rebasing this on top of main branch if there's even more conflicting changes. It would be great to have this upstreamed hoping it might prevent some future breakage as well.

BTW Mesa main gives Vulkan 1.1 on Pi 4 now.

rsglobal, 2021-12-02:
MR created: https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/14016

dEQP for Vulkan only can be executed using the following line:

```
atest  CtsDeqpTestCases -- --module-arg 'CtsDeqpTestCases:include-filter:dEQP-VK.*'
```

closed 2021-12-02 by rsglobal