GloDroid/glodroid_manifest #177: Virtual camera support

SimoneFrau, 2021-12-16:
Hi, using last master code I am trying to develop a virtual camera to be able to see initially a fake camera in all the apps, including Skype for example. Only later I will try to fill the data of this camera with some stream from other origin. Anyway I am still stuck in the first step, what I tried to do was to enable the Emulated Camera already present in the GloDroid environment as part of the generic/goldfish emulator but I obtain errors of compatibility with the pre-existing camera HALs.

I was wandering if maybe you know some better approach for this kind of activity or you feel mine can be a good way to do it.

Thank you!
Simone

rsglobal, 2021-12-16:
Hello,

I doubt it will be easy, but try using this one https://github.com/umlaeute/v4l2loopback and libcamera which is already a part of glodroid.

SimoneFrau, 2021-12-16:
Thank you for your quick response. I will let you know how it goes because I think can be a useful feature to add

SimoneFrau, 2022-05-11:
No more relevant.

closed 2022-05-11 by SimoneFrau