GloDroid/glodroid_manifest #178: Please add support for PinePhone Pro

andrew712-1, 2021-12-22:


PsychoGame, 2021-12-26:
Does anyone in the RSGlobal Team have the PinePhone Pro? I've got the Developer Edition, which standard comes with a fully functional Android BSP Rom from Rockchip itself. I'm willing to share the imagedump with RSGlobal, maybe could help with bringing up the PPP, but maybe also key components on the original PP (like modem for example). Let me know if there's any interest.

rsglobal, 2021-12-26:
Hi, we do not have PPP at the moment.
It would be really nice to have ppp stock Android image available somewhere on pine64 website.

Porting vendor stuff in proprietary form to newer Android system is something lineageos is doing. Glodroid aims to have vendor partition in opensource form.

PsychoGame, 2021-12-26:
I know that RS Global aims to be fully opensourced, but closed-sourced implementations sometimes might reveal how certain components might have been enabled by the vendor.
In which it might be easier to mimic certain behavior in opensource form.
I'll upload the image later on and share a link for anybody to fiddle with.

rsglobal, 2021-12-26:
Hm, we don't have any relationship with rsglobal.pl just in case it matters for you.

rsglobal, 2022-04-23:
Initial PPP support implemented in v0.7.4 release.

closed 2022-04-23 by rsglobal