GloDroid/glodroid_manifest #182: [Raspberry Pi4] flash-sd.sh is hung on MAC OSX

dhoomakethu, 2022-01-31:
Hello I am trying to flash https://github.com/GloDroid/glodroid_manifest/releases/tag/v0.7.0 on to rasberry pi4 . Here are the steps I followed.

* Flash `deploy-sd.img` on to sd card using balena etcher
* Insert `sd card` to raspberry pi, connect USB-C to MAC to power on.
* Ensure `fastboot` shows the pi 
```
fastboot devices
10000000a0bb32ca  fastboot
```
* Execute `./flash-sd.sh`

The script is struck at the flashing gpt .
```
./flash-sd.sh
+ fastboot flash gpt deploy-gpt.img
```
Fastboot version
```
fastboot --version
fastboot version 31.0.3-7562133
```

How long it generally takes to flash the image ? And is there a limitation of doing fastboot from Mac osx ?

dhoomakethu, 2022-02-01:
Looks like an issue with Mac. I could get this working after using a linux machine.

powenn, 2022-02-04:
> Looks like an issue with Mac. I could get this working after using a linux machine.

https://github.com/powenn/glodroid_manifest/tree/master/MacOS-ShellScript

rsglobal, 2023-05-02:
Newer releases are shipped with a sdcard image. Also, fastboot is no longer formatting partitions, which reduces the native tools required and makes it possible to flash on every OS.

closed 2023-05-02 by rsglobal