GloDroid/glodroid_manifest #183: make dist fails

dhoomakethu, 2022-02-01:
Trying to run `make dist` fails with the below error
```
++++ userdata  ++++

2022-02-01 13:19:03 - add_img_to_target_files.py - INFO    : creating userdata.img...
2022-02-01 13:19:03 - common.py - INFO    :   userdata size (135) is 0.00% of limit (33554432)
2022-02-01 13:19:03 - add_img_to_target_files.py - INFO    :

++++ cache  ++++

2022-02-01 13:19:03 - add_img_to_target_files.py - INFO    : creating cache.img...
Traceback (most recent call last):
  File "/home/ubuntu/GloDroid/out/host/linux-x86/bin/add_img_to_target_files/internal/stdlib/runpy.py", line 174, in _run_module_as_main
  File "/home/ubuntu/GloDroid/out/host/linux-x86/bin/add_img_to_target_files/internal/stdlib/runpy.py", line 72, in _run_code
  File "/home/ubuntu/GloDroid/out/host/linux-x86/bin/add_img_to_target_files/__main__.py", line 12, in <module>
  File "/home/ubuntu/GloDroid/out/host/linux-x86/bin/add_img_to_target_files/internal/stdlib/runpy.py", line 174, in _run_module_as_main
  File "/home/ubuntu/GloDroid/out/host/linux-x86/bin/add_img_to_target_files/internal/stdlib/runpy.py", line 72, in _run_code
  File "/home/ubuntu/GloDroid/out/host/linux-x86/bin/add_img_to_target_files/add_img_to_target_files.py", line 1030, in <module>
  File "/home/ubuntu/GloDroid/out/host/linux-x86/bin/add_img_to_target_files/add_img_to_target_files.py", line 1024, in main
  File "/home/ubuntu/GloDroid/out/host/linux-x86/bin/add_img_to_target_files/add_img_to_target_files.py", line 897, in AddImagesToTargetFiles
  File "/home/ubuntu/GloDroid/out/host/linux-x86/bin/add_img_to_target_files/add_img_to_target_files.py", line 573, in AddCache
  File "/home/ubuntu/GloDroid/out/host/linux-x86/bin/add_img_to_target_files/common.py", line 2320, in CheckSize
KeyError: '/cache'
13:19:04 ninja failed with: exit status 1

#### failed to build some targets (03:19 (mm:ss)) ####

```

Is this known or am I missing something?

rsglobal, 2022-02-01:
Please run `make images` instead

dhoomakethu, 2022-02-01:
Thanks for the reply @rsglobal . I have already run `make images` and was able to successfully flash it on rpi4. I am now trying to build a OTA package and that's where I am facing the above error. Apologies if the description was not clear.

rsglobal, 2023-04-27:
That was fixed some time ago as well as seamless Virtual A/B OTA. New versions are all here: https://github.com/orgs/GloDroidCommunity/repositories

closed 2023-04-27 by rsglobal