GloDroid/glodroid_manifest #185: [Raspberry Pi4] USB bus KO on PC after flashing

SimoneFrau, 2022-02-22:
Hi, I notice this behavior both in master and next branches.
With an Rpi4 connected through USB to a laptop with Ubuntu 20.04.3 LTS, immediately after flashing with flash-sd.sh, during the startup of Android the USB bus and all the USB devices connected to the laptop do not work anymore. I do not use any USB bridge, I just have an USB keyboard and mouse connected to other separate USB connectors. I always need to restart the laptop after a new flash. I do not know if this is connected to this https://github.com/GloDroid/glodroid_manifest/issues/72 because I was on the next branch thinking it was a known issue, but when I came back to the master I noticed there too.

I saw this behavior using lineage-rpi too, but it was fixed in the last release.

Thank you for your help!

SimoneFrau, 2022-02-22:
[  102.912797] usb 1-5: new high-speed USB device number 7 using xhci_hcd
[  108.080712] usb 1-5: device descriptor read/64, error -110
[  123.696455] usb 1-5: device descriptor read/64, error -110
[  123.932384] usb 1-5: new high-speed USB device number 8 using xhci_hcd
[  129.072241] usb 1-5: device descriptor read/64, error -110
[  130.628350] usb 1-5: device descriptor read/64, error -71
[  130.736351] usb usb1-port5: attempt power cycle
[  131.388272] usb 1-5: new high-speed USB device number 9 using xhci_hcd
[  131.414093] usb 1-5: New USB device found, idVendor=18d1, idProduct=4ee7, bcdDevice= 2.23
[  131.414106] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  131.414111] usb 1-5: Product: ADB Gadget
[  131.414115] usb 1-5: Manufacturer: GloDroid
[  131.414118] usb 1-5: SerialNumber: 10000000a5bf7b1e
[  173.060491] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[  173.980007] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[  699.764073] usb 1-5: USB disconnect, device number 9
[  737.328398] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[  758.647106] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[  763.178757] usb 1-5: new high-speed USB device number 10 using xhci_hcd
[  768.562645] usb 1-5: device descriptor read/64, error -110
[  784.174712] usb 1-5: device descriptor read/64, error -110
[  784.410668] usb 1-5: new high-speed USB device number 11 using xhci_hcd
[  789.550675] usb 1-5: device descriptor read/64, error -110
[  790.502689] usb 1-5: device descriptor read/64, error -71
[  790.610617] usb usb1-port5: attempt power cycle
[  791.262638] usb 1-5: new high-speed USB device number 12 using xhci_hcd
[  791.293709] usb 1-5: New USB device found, idVendor=18d1, idProduct=4ee7, bcdDevice= 2.23
[  791.293723] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  791.293728] usb 1-5: Product: ADB Gadget
[  791.293732] usb 1-5: Manufacturer: GloDroid
[  791.293735] usb 1-5: SerialNumber: 10000000a5bf7b1e
[  968.408289] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[  981.568859] usb 1-5: USB disconnect, device number 12
[ 1148.210970] mmc0: new high speed SDHC card at address 59b4
[ 1148.221871] mmcblk0: mmc0:59b4 USD   7.28 GiB 
[ 1148.243454]  mmcblk0: p1 p2 p3 p4 p5 p6 p7 p8 p9 p10
[ 1148.747588] EXT4-fs (mmcblk0p9): recovery complete
[ 1148.752075] EXT4-fs (mmcblk0p9): mounted filesystem with ordered data mode. Opts: (null). Quota mode: none.
[ 1148.752087] ext4 filesystem being mounted at /media/simo/c017e9fc-1b52-47b7-9e72-d19c53943208 supports timestamps until 2038 (0x7fffffff)
[ 1226.806240] GPT:Primary header thinks Alt. header is not at the end of the disk.
[ 1226.806252] GPT:524287 != 15267839
[ 1226.806259] GPT:Alternate GPT header not at the end of the disk.
[ 1226.806263] GPT:524287 != 15267839
[ 1226.806267] GPT: Use GNU Parted to correct GPT errors.
[ 1226.806299]  mmcblk0: p1 p2 p3
[ 1243.415060] mmc0: card 59b4 removed
[ 1262.942542] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[ 1273.599625] usb 1-5: new high-speed USB device number 13 using xhci_hcd
[ 1274.701558] usb 1-5: New USB device found, idVendor=18d1, idProduct=4ee0, bcdDevice=99.99
[ 1274.701574] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1274.701580] usb 1-5: Product: USB download gadget
[ 1274.701585] usb 1-5: Manufacturer: FSL
[ 1274.701589] usb 1-5: SerialNumber: 10000000a5bf7b1e
[ 1278.912036] audit: type=1400 audit(1645528367.620:73): apparmor="DENIED" operation="open" profile="snap.skype.skype" name="/etc/issue" pid=7134 comm="skypeforlinux" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[ 1308.226102] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[ 1309.339303] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[ 1381.160304] usb 1-5: USB disconnect, device number 13
[ 1394.279003] usb 1-5: new high-speed USB device number 14 using xhci_hcd
[ 1395.381273] usb 1-5: New USB device found, idVendor=18d1, idProduct=4ee0, bcdDevice=99.99
[ 1395.381286] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1395.381291] usb 1-5: Product: USB download gadget
[ 1395.381296] usb 1-5: Manufacturer: FSL
[ 1395.381299] usb 1-5: SerialNumber: 10000000a5bf7b1e
[ 1759.195255] usb 1-5: USB disconnect, device number 14
[ 1775.740817] usb 1-5: new high-speed USB device number 15 using xhci_hcd
[ 1781.032861] usb 1-5: device descriptor read/64, error -110
[ 1781.476744] usb 1-5: device descriptor read/64, error -71
[ 1781.712765] usb 1-5: new high-speed USB device number 16 using xhci_hcd
[ 1781.865327] usb 1-5: New USB device found, idVendor=18d1, idProduct=4ee0, bcdDevice= 5.10
[ 1781.865341] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1781.865346] usb 1-5: Product: rpi4
[ 1781.865349] usb 1-5: Manufacturer: RaspberryPiFoundation
[ 1781.865353] usb 1-5: SerialNumber: 10000000a5bf7b1e
[ 2021.271507] usb 1-5: USB disconnect, device number 16
[ 2037.875596] usb 1-5: new high-speed USB device number 17 using xhci_hcd
[ 2043.175580] usb 1-5: device descriptor read/64, error -110
[ 2058.791480] usb 1-5: device descriptor read/64, error -110
[ 2059.027120] usb 1-5: new high-speed USB device number 18 using xhci_hcd
[... cut ...]
[ 2091.215243] usb 1-5: device not accepting address 19, error -108
[ 2091.215330] usb usb1-port5: couldn't allocate usb_device
[ 2091.215360] usb 1-1: USB disconnect, device number 2
[ 2091.328115] usb 1-2: USB disconnect, device number 3
[ 2091.392085] usb 1-4: USB disconnect, device number 4
[ 2091.423998] usb 1-7: USB disconnect, device number 5
[ 2091.424570] usb 1-10: USB disconnect, device number 6
[ 2098.609391] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7

SimoneFrau, 2022-02-22:
That was the result of a dmesg during the issue

SimoneFrau, 2022-02-22:
This is on the opposite the dmesg after a reboot of the laptop, plug in of the Rpi4 and no issue:

[   75.783544] usb 1-5: new high-speed USB device number 7 using xhci_hcd
[   80.920588] usb 1-5: device descriptor read/64, error -110
[   96.538194] usb 1-5: device descriptor read/64, error -110
[   96.774185] usb 1-5: new high-speed USB device number 8 using xhci_hcd
[  101.914489] usb 1-5: device descriptor read/64, error -110
[  103.702571] usb 1-5: device descriptor read/64, error -71
[  103.810566] usb usb1-port5: attempt power cycle
[  104.462521] usb 1-5: new high-speed USB device number 9 using xhci_hcd
[  104.494106] usb 1-5: New USB device found, idVendor=18d1, idProduct=4ee7, bcdDevice= 2.23
[  104.494118] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  104.494123] usb 1-5: Product: ADB Gadget
[  104.494127] usb 1-5: Manufacturer: GloDroid
[  104.494130] usb 1-5: SerialNumber: 10000000a5bf7b1e
[  326.403659] loop19: detected capacity change from 0 to 126752

rsglobal, 2022-03-30:
@SimoneFrau ,

Yes, this issue is very annoying. Upgrading to latest firmware/kernel/u-boot haven't solved it. 
On my HP laptop I can do `sudo rmmod xhci_pci && sudo modprobe xhci_pci` to restore USB functionality, but only single time.

To create an issue for RPI guys I need a way to reproduce it on raspbian, otherwise they will not look into it.
So all can I do is to wait for someone fixes it and then upgrade to the latest firmware/kernel.

SimoneFrau, 2022-03-30:
Hi,

ok thank you for your work!

Simone

rsglobal, 2022-04-23:
Thanks.
Looks like the issue has gone. At least I can't see it on v0.7.4.
I'll close this issue for now. Feel free to reopen in case the issue still relevant.

closed 2022-04-23 by rsglobal