GloDroid/glodroid_manifest #186: Raspberry Pi CM4 with eM

NumberOneGit, 2022-02-27:
I'd like to test out GloDroid on the Raspbery Pi CM4. The installer for Pi 4B only comes with an installer for SD cards but not the eMMC related files. Is this something that I can take from another device or are there contents that are hardware specific?

rsglobal, 2022-03-30:
Hi,

You have to edit `device/glodroid/rpi4/rpi4.mk` file.
1. Add `PRODUCT_HAS_EMMC := true` line
2. Add `SYSFS_MMC1_PATH := emmc2bus/f???????.???` line. Instead of `???` you have to put real sysfs path to rpicm4 emmc controller.

Probably you also have to copy appropriate device-tree into boot partition.

danascape, 2025-04-08:
> Hi,
>
> You have to edit `device/glodroid/rpi4/rpi4.mk` file.
>
> 1. Add `PRODUCT_HAS_EMMC := true` line
> 2. Add `SYSFS_MMC1_PATH := emmc2bus/f???????.???` line. Instead of `???` you have to put real sysfs [...]

Is there a way to find the emmc2bus/ path??

danascape, 2025-04-08:
SYSFS_MMC0_PATH := emmc2bus/fe340000.mmc

This is on pi4, may I ask how can I find this for some other SoC? like rpi5

danascape, 2025-04-08:
sudo hwinfo --disk --only /dev/mmcblk0

I think I found it? Is this correct