GloDroid/glodroid_manifest #189: [Raspberry Pi4] Couldn't boot to home screen

nguyenanhgiau, 2022-04-01:
Hi GloDroid team,
I really love your project.
I'm using the image [gd-v0.7.1-rpi4-userdebug.tar.gz](https://github.com/GloDroid/glodroid_manifest/releases/download/v0.7.1/gd-v0.7.1-rpi4-userdebug.tar.gz)
After preparing recovery sdcard and flashing successfully the image with cmd "./flash-sd.sh", rpi4 could not boot to home screen. I got the error message as below:
log from uart:
```bash
[ 1237.474676]  vc4_hdmi_runtime_resume+0x44/0x4b0 [vc4]
[ 1300.498015] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
[ 1300.504199] rcu: 	2-...0: (0 ticks this GP) idle=77b/1/0x4000000000000000 softirq=788/791 fqs=160149 
[ 1300.513560] 	(detected by 1, t=320548 jiffies, g=597, q=84)
[ 1300.519213] Task dump for CPU 2:
[ 1300.522483] task:ueventd         state:R  running task     stack:    0 pid:  143 ppid:   139 flags:0x0400000a
[ 1300.532551] Call trace:
[ 1300.535027]  __switch_to+0x164/0x210
```
log from dmesg:
```bash
[ 3276.426573] usb 3-1.4: new full-speed USB device number 28 using xhci_hcd
[ 3276.551316] usb 3-1.4: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[ 3276.551319] usb 3-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3276.551320] usb 3-1.4: Product: CP2102 USB to UART Bridge Controller
[ 3276.551321] usb 3-1.4: Manufacturer: Silicon Labs
[ 3276.551322] usb 3-1.4: SerialNumber: 0001
[ 3276.555304] cp210x 3-1.4:1.0: cp210x converter detected
[ 3276.558598] usb 3-1.4: cp210x converter now attached to ttyUSB0
[ 3281.564400] cp210x ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[ 3355.806701] usb 3-1.2: USB disconnect, device number 27
[ 3374.441866] usb 3-1.2: new high-speed USB device number 29 using xhci_hcd
[ 3379.613967] usb 3-1.2: device descriptor read/64, error -110
[ 3395.230194] usb 3-1.2: device descriptor read/64, error -110
[ 3395.430168] usb 3-1.2: new high-speed USB device number 30 using xhci_hcd
[ 3400.606150] usb 3-1.2: device descriptor read/64, error -110
[ 3416.221710] usb 3-1.2: device descriptor read/64, error -110
[ 3416.330007] usb 3-1-port2: attempt power cycle
[ 3416.962016] usb 3-1.2: new high-speed USB device number 31 using xhci_hcd
[ 3422.013657] xhci_hcd 0000:28:00.3: Timeout while waiting for setup device command
[ 3427.389959] xhci_hcd 0000:28:00.3: Timeout while waiting for setup device command
[ 3427.597939] usb 3-1.2: device not accepting address 31, error -62
```
Any thing missing? Could you please help me fix this issue?
Note: My screen is connected to rpi4 by HDMI cable.
Thanks!

rsglobal, 2022-04-02:
Hi,

Can you check it with some big monitor?
RPI4 has some limitations with my 1024x600 monitor, but instead it falls-back to 1920x1080 mode which my display support by EDID blob.

I may guess that you have the same situation but EDID has only 1 mode in the list.

nguyenanhgiau, 2022-04-03:
Hi @rsglobal 
Thank you for your help. It worked on the big monitor.
Maybe, we need to support more monitors in the future.

closed 2022-04-03 by nguyenanhgiau