GloDroid/glodroid_manifest #191: OrangePi 3 LTS videos from youtube crashing webview (firefox and other browsers)

Sekilsgs2, 2022-04-05:
crashes when watching a youtube video

logcat - 
https://pastebin.com/ziqp0R0J

rsglobal, 2022-04-05:
Hi,

I can't see anything suspicious in logcat , is it complete?
Please try updating external/mesa3d to v0.7.2 and check again

Sekilsgs2, 2022-04-05:
> Hi,
> 
> I can't see anything suspicious in logcat , is it complete? Please try updating external/mesa3d to v0.7.2 and check again

Full logcat - https://www.filemail.com/d/kgkgasuqbtjdrfd

Ok  -i'm trying latest 0.7.2 variants..

What about panvk (vulkan panfrost) - this have working state? maybe using vulkan for ui? or this never working for us?

rsglobal, 2022-04-05:
Vulkan require adding some features to mesa3d. It doesn't work now for Android. UI on vulkan will require adding even more features to mesa3d.

rsglobal, 2022-04-05:
I've been watching youtube on opi3 @ v0.7.2 for an hour. I suspect it's an issue related to opi-lts.

Sekilsgs2, 2022-04-05:
Yes on 0.7.2 - working thank.

And one moment - 

I'm have lcd hdmi 10 inch 1280x800 - in edid have two modes - with 60hz and with 15 hz - why abdroid by default change to 15hz mode and i'm have very bad fps for all system. To fix this i'm edit system display service and refuse modes which have < 25 hz refresh rate.. But maybe android have settings for hdmi modes and need activate this ?

rsglobal, 2022-04-06:
I'll look into it. Thanks.

closed 2022-04-06 by Sekilsgs2