GloDroid/glodroid_manifest #192: GloDroid 0.7.2 won't boot on PinePhone

MoTLD, 2022-04-15:
Hello, and thanks for this awesome project!

I've installed a few previous versions of GloDroid on my PinePhone (3GB/32GB version) and they've worked fine, so I think I know how to install it correctly, but I just flashed 0.7.2 to the eMMC and it can't seem to get much past the initial android screen.

When I turn it on the green light comes on for a few seconds and then goes out, then "android" shows in the center of the screen and throbs for a minute or two, then the whole phone goes dark for a good few seconds, then the green light comes back on for a few seconds, then the phone goes dark again for a few seconds, then the backlight comes on but nothing shows on the screen and it will sit like this, screen black but backlight on, until I give up and hold the power button down to shut it off. I've left it sitting for at least 10 or 15 minutes once and it never did anything else.

I usually keep the modem and front cameras turned off with the dip switches, but I tried it with everything turned on and it made no difference. The previous version of GloDroid that I had installed was working as well as can be expected, I think it was 0.7.0.

Just now as I was writing this I booted it again, and this time it's just sitting on the android throbber screen.

If there are any logs I can share or adb stuff I can do to help with throubleshooting, just let me know and I'll do whatever I can. I might try reflashing the eMMC and/or installing it to an SD card instead to see if that makes a difference.

Thanks again for this project!

-Mo
 
ps - the 0.7.1 release note suggests y'all are in Ukraine; my heart goes out to you, please be safe!

MaksymPrymierov, 2022-04-16:
Hello! Do you have usb-jack uart cable? Using this cable you can save kernel boot logs via pinephone's serial port. For example: https://www.amazon.com/3-5mm-Audio-Jack-Programming-Cable/dp/B088LS5HNQ

rsglobal, 2022-04-16:
I checked v0.7.2 image and it works on my PP.

rsglobal, 2022-07-18:
@MoTLD ,

Hi,
Please check on v0.7.5 - v0.7.6 and reopen the issue if it still relevant.

closed 2022-07-18 by rsglobal