GloDroid/glodroid_manifest #193: [Raspberry Pi4] doesn't load device tree overlay

nguyenanhgiau, 2022-04-16:
Hi @rsglobal 
I'm trying to add 3 buttons for rpi4 (power, vol- and vol+).
Firstly, I create a device tree overlay in kernel: arch/arm/boot/dts/overlays/arpi-gpio-key-overlay.dts
```patch
@@ -0,0 +1,36 @@
/dts-v1/;
/plugin/;
/ {
    compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

    fragment@0 {
        target-path = "/";
        __overlay__ {
            keypad: arpi_keys {
                compatible = "gpio-keys";
                #address-cells = <1>;
                #size-cells = <0>;
                gpio-key,wakeup;
                autorepeat;

                button@pow {
                    label = "POWER";
                    linux,code = <116>;
                    gpios = <&gpio 21 1>;
                };

                button@vlu {
                    label = "VOLUME_UP";
                    linux,code = <115>;
                    gpios = <&gpio 20 1>;
                };

                button@vld {
                    label = "VOLUME_DOWN";
                    linux,code = <114>;
                    gpios = <&gpio 16 1>;
                };
            };
        };
    };
}; 
```
Then, I add this dtoverlay into arch/arm/boot/dts/overlays/Makefile
```patch
@@ -29,6 +29,7 @@ dtbo-$(CONFIG_ARCH_BCM2835) += \
	audioinjector-wm8731-audio.dtbo \
	audiosense-pi.dtbo \
	audremap.dtbo \
	arpi-gpio-key.dtbo \
	balena-fin.dtbo \
	cma.dtbo \
	dht11.dtbo \
```
To load this overlay, I add the following line in config.txt
```patch
dtoverlay=dwc2
dtoverlay=vc4-kms-v3d-pi4

# Adding gpio-key
gpio=16,20,21=ip,pu
dtoverlay=arpi-gpio-key
```
Finally, I build with command: "make images -j$(nproc)"
I'm sure this overlay was built because there is arpi-gpio-key.dtbo file in output. But not sure It was copied to boot image.

When the device boot to home screen. I didn't see my buttons in /sys/class/input
I make sure this overlay worked well on raspbian OS.

I'm curious why dwc2 was loaded, but arpi-gpio-key was not.
Anything missing?
Thanks a lot!

rsglobal, 2022-04-16:
Hi, you also have to ship compiled dts (.dtb) file.
e.g.:
https://github.com/GloDroid/glodroid_device/blob/cb5423bf00057a788790442569bcfaef5819a1d0/platform/uboot/uboot.mk#L184

rsglobal, 2022-04-16:
Something like this:

```diff
ifeq ($(PRODUCT_BOARD_PLATFORM),broadcom)
BOOT_FILES := \
    $(RPI_FIRMWARE_DIR)/boot/bootcode.bin \
    $(RPI_FIRMWARE_DIR)/boot/start_x.elf \
    $(RPI_FIRMWARE_DIR)/boot/start4x.elf \
    $(RPI_FIRMWARE_DIR)/boot/fixup_x.dat \
    $(RPI_FIRMWARE_DIR)/boot/fixup4x.dat \
    $(PRODUCT_OUT)/obj/KERNEL_OBJ/arch/$(TARGET_ARCH)/boot/dts/broadcom/bcm2711-rpi-4-b.dtb \
    $(PRODUCT_OUT)/obj/KERNEL_OBJ/arch/$(TARGET_ARCH)/boot/dts/broadcom/bcm2711-rpi-400.dtb \
    $(PRODUCT_OUT)/obj/KERNEL_OBJ/arch/$(TARGET_ARCH)/boot/dts/broadcom/bcm2711-rpi-cm4.dtb \

+CUSTOM_RPI_OVERLAY_FILES := $(PRODUCT_OUT)/obj/KERNEL_OBJ/arch/$(TARGET_ARCH)/boot/dts/broadcom/overlays/arpi-gpio-key-overlay.dtb

OVERLAY_FILES := $(sort $(shell find -L $(RPI_FIRMWARE_DIR)/boot/overlays))

$(PRODUCT_OUT)/bootloader-sd.img: $(UBOOT_BINARY) $(OVERLAY_FILES) $(ATF_BINARY) $(RPI_CONFIG) $(KERNEL_BINARY)
	dd if=/dev/null of=$@ bs=1 count=1 seek=$$(( 128 * 1024 * 1024 - 256 * 512 ))
	/sbin/mkfs.vfat -F 32 -n boot $@
	/usr/bin/mcopy -i $@ $(UBOOT_BINARY) ::$(notdir $(UBOOT_BINARY))
	/usr/bin/mcopy -i $@ $(ATF_BINARY) ::$(notdir $(ATF_BINARY))
	/usr/bin/mcopy -i $@ $(RPI_CONFIG) ::$(notdir $(RPI_CONFIG))
	/usr/bin/mcopy -i $@ $(BOOT_FILES) ::
	/usr/bin/mmd -i $@ ::overlays
	/usr/bin/mcopy -i $@ $(OVERLAY_FILES) ::overlays/
+	/usr/bin/mcopy -i $@ $(CUSTOM_RPI_OVERLAY_FILES) ::overlays/
endif
```

nguyenanhgiau, 2022-04-16:
> 

Hi @rsglobal 
It worked well.
Thank you for your suggestion
This is my patch:
```patch
--- a/platform/uboot/uboot.mk
+++ b/platform/uboot/uboot.mk
@@ -183,6 +183,11 @@ BOOT_FILES := \
     $(RPI_FIRMWARE_DIR)/boot/bcm2710-rpi-3-b-plus.dtb \
     $(PRODUCT_OUT)/obj/KERNEL_OBJ/arch/$(TARGET_ARCH)/boot/dts/$(KERNEL_DTB_FILE) \
 
+CUSTOM_RPI_OVERLAY_DIR := $(PRODUCT_OUT)/obj/KERNEL_OBJ/arch/$(TARGET_ARCH)/boot/dts/overlays
+
+CUSTOM_RPI_OVERLAY_FILES := \
+       $(CUSTOM_RPI_OVERLAY_DIR)/arpi-gpio-key.dtbo
+
 OVERLAY_FILES := $(sort $(shell find -L $(RPI_FIRMWARE_DIR)/boot/overlays))
 
 $(PRODUCT_OUT)/bootloader-sd.img: $(UBOOT_BINARY) $(OVERLAY_FILES) $(ATF_BINARY) $(RPI_CONFIG) $(KERNEL_BINARY)
@@ -194,6 +199,7 @@ $(PRODUCT_OUT)/bootloader-sd.img: $(UBOOT_BINARY) $(OVERLAY_FILES) $(ATF_BINARY)
        /usr/bin/mcopy -i $@ $(BOOT_FILES) ::
        /usr/bin/mmd -i $@ ::overlays
        /usr/bin/mcopy -i $@ $(OVERLAY_FILES) ::overlays/
+       /usr/bin/mcopy -i $@ $(CUSTOM_RPI_OVERLAY_FILES) ::overlays/
 endif
```

closed 2022-04-16 by nguyenanhgiau