GloDroid/glodroid_manifest #194: [Raspberry Pi4] Not found I2C driver

nguyenanhgiau, 2022-04-17:
Hi @rsglobal 
Have you ever tried to use the I2C interface for rpi4 on GloDroid?
I'm trying to implement an EEPROM that uses the I2C protocol. But I realize that I2C driver was not initialized? (not found /dev/i2c-*)
The following configuration was added in config.txt
```bash
dtparam=i2c_arm=on
dtparam=spi=on
```
This configuration worked well on raspbian OS.
This issue is blocking my journey to understand AOSP.
Could you please tell me what was wrong?
Thanks!

rsglobal, 2022-04-17:
You have to `modprobe` i2c kernel module.
E.g.
```
adb root
adb shell modprobe -d /vendor/lib/modules i2c_dev
```

nguyenanhgiau, 2022-04-17:
Thanks @rsglobal 
I forgot this step. I also added the following lines to init.common.rc for automatically initialization.
```patch
diff --git a/common/init.common.rc b/common/init.common.rc
index 6fcb99b..08151ed 100644
--- a/common/init.common.rc
+++ b/common/init.common.rc
@@ -25,6 +25,10 @@ on early-init
     chown bluetooth net_bt /sys/class/rfkill/rfkill0/device/extldo
     chmod 0660 /sys/class/rfkill/rfkill0/device/extldo
 
+    # I2C interface
+    exec u:r:vendor_modprobe:s0 -- /vendor/bin/modprobe -a -d \
+        /vendor/lib/modules i2c-dev
+
 on fs
     mount_all /vendor/etc/fstab.${ro.hardware} --early
     setprop ro.crypto.fuse_sdcard false
```

closed 2022-04-17 by nguyenanhgiau